raspberrypi/picamera2 #771: [BUG] HQ Camera recording at 2028x1520 does result in crash

nmaas87, 2023-08-08:
**Describe the bug**
I try to record video with the RPi HQ Camera at 2028x1520 while taking JPEG pictures at a lower resolution at the same time (and having preview via DRM active). The program crashs as seen below. Trying to use a lower resolution for the video (e.g. 2028x1080) works. I guess I am doing something really, really wrong :)

**To Reproduce**
````
#!/usr/bin/python3

import time
from picamera2 import Picamera2, Preview
from picamera2.encoders import H264Encoder

picam2 = Picamera2()
picam2.start_preview(Preview.DRM, width=1920, height=1200)

#main_stream = {"size": (2028, 1080)}
main_stream = {"size": (2028, 1520)}
#lores_stream = {"size": (1014, 760)}
lores_stream = {"size": (2028, 1080)}
#lores_stream = {"size": (2028, 1520)}

video_config = picam2.create_video_configuration(main_stream, lores_stream, encode="main")
picam2.configure(video_config)

encoder = H264Encoder(10000000)

picam2.start_recording(encoder, 'app_test.h264')
time.sleep(5)

request = picam2.capture_request()
request.save("main", "app_test.jpg")
request.release()
print("Still image captured!")

time.sleep(5)
picam2.stop_recording()
````

**Expected behaviour**
Record the video and take the snapshot.

**Console Output, Screenshots**
````
app@system:/app# python3 app.py
[0:07:41.395761910] [557]  INFO Camera camera_manager.cpp:284 libcamera v0.1.0+dirty (2023-08-06T16:35:29+00:00)
[0:07:41.451402947] [558]  INFO RPI vc4.cpp:390 Registered camera /base/soc/i2c0mux/i2c@1/imx477@1a to Unicam device /dev/media0 and ISP device /dev/media2
[0:07:41.487673206] [557]  INFO Camera camera.cpp:1033 configuring streams: (0) 2028x1520-XBGR8888 (1) 2028x1080-YUV420 (2) 2028x1520-SBGGR12_CSI2P
[0:07:41.488981243] [558]  INFO RPI vc4.cpp:512 Sensor: /base/soc/i2c0mux/i2c@1/imx477@1a - Selected sensor format: 2028x1520-SBGGR12_1X12 - Selected unicam format: 2028x1520-pBCC
Traceback (most recent call last):
  File "/app/app.py", line 66, in <module>
    picam2.start_recording(encoder, 'app_test.h264')
  File "/usr/lib/python3.11/site-packages/picamera2/picamera2.py", line 1597, in start_recording
    self.start_encoder(encoder, output, pts=pts, quality=quality, name=name)
  File "/usr/lib/python3.11/site-packages/picamera2/picamera2.py", line 1537, in start_encoder
    _encoder.start()
  File "/usr/lib/python3.11/site-packages/picamera2/encoders/encoder.py", line 234, in start
    self._start()
  File "/usr/lib/python3.11/site-packages/picamera2/encoders/h264_encoder.py", line 69, in _start
    super()._start()
  File "/usr/lib/python3.11/site-packages/picamera2/encoders/v4l2_encoder.py", line 137, in _start
    fcntl.ioctl(self.vd, VIDIOC_STREAMON, typev)
ProcessLookupError: [Errno 3] No such process
````

**Hardware :**
CM4, RPi HQ Camera

**Additional context**
libcamera 0.1.0
picamera2 0.3.12

Thanks a lot for your help! :)

davidplowman, 2023-08-09:
h.264 encode is only supported up to 1920x1080. In practice you can go slightly beyond this, but as you have discovered, not much.

nmaas87, 2023-08-15:
Thank you very much. If I want to reach higher resolutions, I guess I need to use MJPEG encoder or raw video? :)

davidplowman, 2023-08-16:
Yes. With raw video, getting stuff onto the disk fast enough may be more of an issue but otherwise that's correct.

nmaas87, 2023-08-16:
Yeah I see and using software compression for H.264 will most likely not be sufficient enough to hit ~30 FPS. So this answers my question, thank you very much for your time 👍🏻

closed 2023-08-16 by nmaas87