raspberrypi/picamera2 #772: [SUGGESTION] record_sequence

nmaas87, 2023-08-08:
**Is your feature request related to a problem? Please describe.**
Picamera (1) had a feature called record_sequence which allows to continiously record h264 video without interruption in small, configurable lengths and save them in different file chunks. This is extremely useful for e.g. action cameras/ "GoPro" implementations to allow for secure video recordings even if the power source is interupted during recording. Recording the video to a single file usually means that such an interupt will mean the complete file will be corrupted, implementing it this way usually only destroys the last one or two short files. I want to convert an old picamera (1) application I made to picamera2 - and thats the missing link :)

**Describe the solution you'd like**
As seen on picamera1 - [record_sequence](https://picamera.readthedocs.io/en/release-1.13/api_camera.html#picamera.PiCamera.record_sequence)

**Describe alternatives you've considered**
-

**Additional context**
-

nmaas87, 2023-08-15:
Thanks for your help, sorry for the late response, I wanted to test it first.
My requirements are basically:
- live preview via DRM/KMS
- recording the video stream in smaller pieces with no frame losses
- saving a picture snapshot from time to time on request

I modified your script to record 10 seconds in 1 second pieces for testing the 2nd requirement:

````
import time
from picamera2 import Picamera2
from  picamera2.encoders import H264Encoder
from picamera2.outputs import FileOutput

with Picamera2() as camera:
    camera.start(camera.create_video_configuration())
    for filename in ['clip%02d.h264' % i for i in range(10)]:
        encoder = H264Encoder()
        camera.start_encoder(encoder, FileOutput(filename))
        time.sleep(1)
        camera.stop_encoder(encoder)
````

I recorded a running Android Stopwatch to get a feeling for the frametiming - however, piecing together the recordings via ffmpeg I got a 7 second video as result, instead of a 10 second one - and within the video I can see the frame jumps between the recordings.
So I guess the start/stop encoder eats up too much time.

https://github.com/raspberrypi/picamera2/assets/1331457/feb9b0e6-63df-4bd6-b966-605cab95bf96

As a second experiment, I moved the encoder init before the for loop:

````
import time
from picamera2 import Picamera2
from  picamera2.encoders import H264Encoder
from picamera2.outputs import FileOutput

with Picamera2() as camera:
    camera.start(camera.create_video_configuration())
    encoder = H264Encoder()
    for filename in ['clip%02d.h264' % i for i in range(10)]:
        camera.start_encoder(encoder, FileOutput(filename))
        time.sleep(1)
        camera.stop_encoder(encoder)
````

... that got stuff even more interesting. For reasons I am not aware of, the recording got even shorter. Adding the videos together, the new video was now 4 seconds and one could see a lot of frame drops.

https://github.com/raspberrypi/picamera2/assets/1331457/0b1d23d2-dd6e-43c4-8e13-f68e64283c38

Is there any way to get more reliable results as wanted to my use-case?

Thanks so much :)

nmaas87, 2023-08-17:
Thanks, that helps. Reducing it to the more horrible test pattern (1 second video, 10 takes - which will not be the final product, but a good benchmark ;)) it takes 18 seconds of video, but these look very smooth! :) 

````
import threading
import time

from picamera2 import Picamera2
from picamera2.encoders import H264Encoder
from picamera2.outputs import Output

class MyOutput(Output):
    def __init__(self, filenames, duration):
        super().__init__()
        self._filenames = filenames
        self._duration = duration
        self.event = threading.Event()
        self._start_time = time.monotonic()
        self._file = open(self._filenames.pop(0), 'wb')

    def outputframe(self, frame, keyframe=True, timestamp=None):
        if self._file is not None and keyframe and time.monotonic() - self._start_time > self._duration:
            self._start_time = time.monotonic()
            self._file.close()
            self._file = None
            if self._filenames:
                self._file = open(self._filenames.pop(0), 'wb')
            else:
                self.event.set()
        if self._file is not None:
            self._file.write(frame)
        
with Picamera2() as camera:
    camera.start(camera.create_video_configuration())
    encoder = H264Encoder(iperiod=30)
    filenames = list(['clip%02d.h264' % i for i in range(10)])
    output = MyOutput(filenames, 1)
    camera.start_encoder(encoder, output)
    output.event.wait()
    camera.stop_encoder()
````

Video: 
https://github.com/raspberrypi/picamera2/assets/1331457/253dab73-1f0b-4868-b60a-eec4b47e3904

I tried to put it together (loading tuning file, starting preview, adding configuration for video and snapshot mode) - but it ended up start starting the first recording file and then hanging. It also did not output any preview.

````
#!/usr/bin/python3

import threading
import time
from picamera2 import Picamera2, Preview
from picamera2.encoders import H264Encoder
from picamera2.outputs import Output

tuning = Picamera2.load_tuning_file("imx477.json")

class MyOutput(Output):
    def __init__(self, filenames, duration):
        super().__init__()
        self._filenames = filenames
        self._duration = duration
        self.event = threading.Event()
        self._start_time = time.monotonic()
        self._file = open(self._filenames.pop(0), 'wb')

    def outputframe(self, frame, keyframe=True, timestamp=None):
        if self._file is not None and keyframe and time.monotonic() - self._start_time > self._duration:
            self._start_time = time.monotonic()
            self._file.close()
[... 37 lines cut ...]
````
cameraThread:
    # housekeeping on boot up
    while True:
        # additional stuff, pre_compute e.g. correct fileNumber starter
        fileNumber = 4711
        with picamera.PiCamera() as camera:    
            # configuration to the camera
            # start preview here
            filename_pre = '/app/clip'  
            # semaphore here to only start recording if button pressed
            for filename in camera.record_sequence(filename_pre + '%d.h264' % i for i in range(fileNumber,999999)):
                print('Recording to %s' % filename)
                camera.wait_recording(10)
                if (imageRequested):
                    camera.snap(otherFilename)
                if (done_recording):
                    break
````

Going into the video, I had a fileNumber, already prepared so that subsequent videos will not overwrite old ones. Then, as I have no idea how long the video will be needed to record for, I need to input a dynamically computed list of filenames with running range. Precomputing all the possible length (recording could run for days(!)) would eat up memory, so the for list here is an important choice). As the filename is also important for internal bookkeeping, it is shown (this is just an example here :)). This would basically then record for 2 second pieces, if after that a picture request was sent, it would also save a picture - or if we recorded enough, it would break the for loop and exit.

Obviously I don't like the camera.wait_recording option from the first picamera, as it means I need (in the worst case) to wait a whole record cycle until operations finish or it can take a still image. With Picamera2, it would make sense to just wait for the end of the iFrame, so that the video could be ended early. I would also like to be able to take a still image anytime during recording without interupting it - and I see a big possibilty to achieve this with picamera2 due to its video_configuration main/lores_stream config.

I think this paints a better picture of what I am trying to achieve here. The camera subsystem is usually running in its own thread and then just doing its own housekeeping before starting the always running preview and then getting it held by a semaphore before falling into the mentionied record_sequene cycle.

davidplowman, 2023-08-17:
Thanks for the update. I'm not sure why that would be hanging for you, I'm afraid you might have to do some investigating and put a few print statements in the code, for example to check that the output is receiving frames, switching files at the right moment, and so on. If you're using the DRM preview, presumably you're running in console mode? Does that work without doing the video recording (I can't see how it would make a difference)? I think you can also set the `camera.pre_callback = print` which (while it will rather swamp your console) will make it clear if the camera is actually still running.

The code I posted is also just a starting point. For example, you could easily add a `start_new_file(filename)` function to the `MyOutput` class so that you could make on-the-fly decisions about when to start a new file (rather than a fixed list of files), waiting for I frames to make the switch if you want each file to be fully self-contained.

nmaas87, 2023-08-23:
Thanks a lot @davidplowman for your help already. I will do some additional tests later next week, currently another deadline came up and I put an reminder for tuesday to when the other project should be out :). Thanks for your help and patience, this will be continued soon! :)

nmaas87, 2023-09-24:
Thanks David and sorry to keep you waiting. The error I perceived (no output) was a stupid error on my site (I actually forgot camera.start() :/). I think I will need to dive a bit deeper to generate new file names on the fly and also be able to take snapshots at any time during the video recording process. Not really sure yet, but I will take a look. Thanks already for your help :)