raspberrypi/picamera2 #776: [BUG] Can't preview and record using pygame

elblogbruno, 2023-08-13:
**Please only report one bug per issue!**

**Describe the bug**
I want to show a preview of the camera in a pygame window and have a background thread that keeps recording videos (the sample code provided here does not implement threads to make it simpler)
I saw #527  and it changes the preview format to 'BGR888' in order to work with pygame. However, when changing that format and try to record (with any of the record functions) the code crashes

Here is the sample code:

```
from picamera2 import Picamera2
import pygame

pygame.init()
res = (1080, 720)
screen = pygame.display.set_mode(res)

camera = Picamera2()
camera.preview_configuration.main.size = res
camera.preview_configuration.main.format = 'BGR888'
camera.configure("preview")

camera.start()

while True:
    array = camera.capture_array()
    img = pygame.image.frombuffer(array.data, res, 'RGB')
    screen.blit(img, (0, 0))
    pygame.display.update()

    camera.start_and_record_video("test.mp4", duration=1) # I know is wrong to block the main thread recording but its just to simplify the code. My main code uses threads to record on the background and still crashes this way.
```

**To Reproduce**
Steps to reproduce the behaviour. Wherever possible, please include a small self-contained Python script that shows the bug.

**Expected behaviour**
A clear and concise description of what you expected to happen.

**Console Output, Screenshots**
If applicable, any console output or screenshots that show the problem and associated error messages.

```
Traceback (most recent call last):
  File "/home/pi/DashMediaPi/test_game.py", line 22, in <module>
    camera.start_and_record_video("test.mp4", duration=1)
  File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 1758, in start_and_record_video
    self.start_encoder(encoder, output, quality)
  File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 1537, in start_encoder
    _encoder.start()
  File "/usr/lib/python3/dist-packages/picamera2/encoders/encoder.py", line 234, in start
    self._start()
  File "/usr/lib/python3/dist-packages/picamera2/encoders/h264_encoder.py", line 69, in _start
    super()._start()
  File "/usr/lib/python3/dist-packages/picamera2/encoders/v4l2_encoder.py", line 59, in _start
    fmt.fmt.pix_mp.pixelformat = self.format
TypeError: an integer is required (got type str)
pipe:: Invalid data found when processing input

```

**Hardware :**
 Please state what kind of Pi you are using, and what camera is attached.

Raspberry PI 4 8GB ram with camera module v3

**Additional context**
Add any other context about the problem here.

davidplowman, 2023-08-15:
Thanks for the report. It seems the 'BGR888' format simply got forgotten about and is not properly supported. I'll get that fixed up. In the meantime, are you able to use 'RGB888' instead? Or even one of the 32-bit formats? It seems to be only that particular one that is missing.

I wouldn't particularly recommend the use of that API (`start_and_record_video`) for anyone other than those who want to "record a 10 second video, do nothing else, and quit", because it stops, reconfigures everything and restarts the camera which is all quite slow, but I appreciate this was only test code. Please ask if you have any further questions, and thank you again for the report!

elblogbruno, 2023-08-15:
Thanks for answering!
I actually switched to using PySimpleGUI instead of pygame, but I came back to try and  it seems setting RGB888 fixes it!
Now I am curious about the API performance for what I want to do, and I will ask something despite is not related to the original issue.
What is the correct way of recording short videos in a loop (ex: 5 seconds short videos) so it acts like a dashcam?
```

 print("Recording clip: " + clip_name)

            # self.api.start_recording(encoder,  clip_path_h264)

            self.api.start_encoder(encoder, clip_path_h264, name="main")

            sleep(5)

            # self.api.stop_recording()
            self.api.stop_encoder()

            pool = ThreadPool(processes=1)
            async_result = pool.apply_async(self.convert_h264_to_mp4, (clip_path_h264, clip_path_mp4))

            print("Finished recording clip: " + clip_name)
```

I did this that records .h264 and then a thread that goes over them and converts them to .mp4 using ffmpeg.
Is it corrrect?

Thanks again for all, and congratulations for this piece of software!

davidplowman, 2023-08-16:
Hi again, yes I think `start_encoder` and `stop_encoder` is the right way to start and stop a recording. You might want to try using the `FfmpegOutput` which would pipe the encoded frames directly to FFmpeg as it goes (for example, [here](https://github.com/raspberrypi/picamera2/blob/main/examples/mp4_capture.py)).

elblogbruno, 2023-08-23:
Thanks a lot!

closed 2023-08-23 by elblogbruno