raspberrypi/picamera2 #778: Using libcamera - Videos(h264) have different lengths although recording time is always 29:50

medem23, 2023-08-16:
Hi Guys, basically I'm using lib camera to take recordings of 30 minutes (fps=30, size=(1640, 1232)):
`
bitrate 10 Mbps
encoder = H264Encoder(5000000). # 5 Mbps

Resolution definition
lores_size = (640, 480)
main_size = (1640, 1232)

Config file for picamera
config = picam2.create_video_configuration(
main={"size": main_size},
lores={"size": lores_size},
controls={"FrameDurationLimits": (33333, 33333)},
display="lores",
encode="main",
buffer_count=8
)

Configure the camera
picam2.configure(config)
`

I'm using PyQt5 timer to switch files every 30 minutes:

`
rec_timer = QtCore.QTimer()

recording_minutes = 30

rec_timer.setInterval(recording_minutes * 60 * 1000)

rec_timer.timeout.connect(switch_filename)
`

Now, my problem is that the video ends up being 29:48 or 29:27 or 29:18 or 29:xx and even sometime 28:xx but they should be 29:50 (I add 10 seconds sleep) but there is no indication of error etc. Are some frames dropped sometimes without warning? I tried increasing the buffer to 12 but that did not help.

davidplowman, 2023-08-16:
Hi, thank you for the question. It would be helpful to have a _very minimal_ code example so that we can see exactly what's going on. Please also include the code within _triple_ backquotes, so that the formatting does not get mangled.

Perhaps you could also clarify how you determine that the clips are short? Is the problem also visible in shorter 1 or 2 minute clips (with a proportionally similar error)? Does the problem go away if you don't display the preview (if indeed you are)? Does the problem go away if you don't save the output? Thanks!

medem23, 2023-08-16:
Yeah sure, here:
```
# global variable definition
picam2 = None
file_counter = 0
encoder = H264Encoder(5000000)                                  # Bit-rate is 5 Mbps

def switch_filename():
    """
    This function will stop and restart recording with a new filename every 30min
    :return: none
    """
    # Pythonic way of dealing with globals
    global picam2
    global encoder
    global file_counter

    # Close everything in order to be able to instantiate again
    if isinstance(picam2, Picamera2):
        if encoder.running and picam2:
            picam2.stop_encoder()
            picam2.stop()
            picam2.close()
            time.sleep(10)

    # Instantiate the picamera class for recordings
    picam2 = Picamera2()

    # Resolution definition
    lores_size = (640, 480)
    main_size = (1640, 1232)

    # Config file for picamera
    config = picam2.create_video_configuration(
        main={"size": main_size},                             # What we encode to disc
        lores={"size": lores_size},                           # What we put into the preview
        controls={"FrameDurationLimits": (33333, 33333)},     # 30fps fixed
        display="lores",                                      # So we show only the small stream
        encode="main",                                        # While encoding in HQ
        buffer_count=8                                        # Let's give it some more buffers due to larger frame size
    )

    # Configure the camera
    picam2.configure(config)

    # Control configs
    awb_dict = {
        "AwbEnable": False,                                    # Manual WB with the set ColourGains
        "AwbMode": 0,                                          # 0 is auto
        "ColourGains": [1.0, 1.0]                              # 1, 1 seem to do fine for now
    }

    # Set the controls
    picam2.set_controls(awb_dict)

    # Filename to write to and tell the picamera that
    filename = XXXXXX # your choice 
    encoder.output = FileOutput(filename + ".h264", pts=filename + ".txt")

    # Set success of starting picamera to False until it succeeded
    success = False
    # Loop until picamera successfully started
    while not success:
        try:
            success = True
            picam2.start()
        except:
            success = False
            time.sleep(10)

            # Start encoder if not already running, important here in order to be able to close it and instantiate again
            if not encoder.running:
                picam2.start_encoder(encoder)

            # Close everything in order to be able to instantiate again
            if isinstance(picam2, Picamera2):
                if encoder.running and picam2:
                    picam2.stop_encoder()
                    picam2.stop()
                    picam2.close()

            # Instantiate the picamera class for recordings
            picam2 = Picamera2()

            # Configure the camera
            picam2.configure(config)

            # Control configs
            picam2.set_controls(awb_dict)

            # Filename to write to and tell the picamera that
            encoder.output = FileOutput(filename + ".h264", pts=filename + ".txt")

    # Start encoder if not already running
    if not encoder.running:
        picam2.start_encoder(encoder)

    # Increase file counter after file switch as we have an additional file
    file_counter += 1

```

Now you can actually run this function switch file on its own (only need to define your filename). But I also implemented a timer to open the function every 30 minutes:

```
# Timer init and some parameter settings
rec_timer = QtCore.QTimer()
recording_minutes = 30
rec_timer.setInterval(recording_minutes * 60 * 1000)                    # 30 minutes hardcoded interval in ms
rec_timer.timeout.connect(switch_filename)
rec_timer.stop()

# App generation
app = QApplication([])
```
But I do not think that the timer has anything to do with this as I display every time this function is called and that seems to work. For completeness, I do not display preview so far. I check the length of clips by transforming them to mp4 and opening them using quickplayer.

davidplowman, 2023-08-16:
Thanks for posting the code, that gives a better idea what's happening.

So there's no preview here, but you are collecting a timestamp file. I assume that you're treating it as if it's a fixed 30fps stream when you convert it to mp4? It's probably worth checking the timestamp file - what's the largest value at the end of the file? I'm guessing that this does correspond to 30 minutes (minus a few seconds). I would also guess that there are a number of frame drops happening - you should be able to determine this from the timestamp file.

medem23, 2023-08-17:
I did the analysis right now and found out that sometimes up to 10 frames (consecutive) are dropped, any idea how to fix this?

davidplowman, 2023-08-17:
10 consecutive frames is presumably about 1/3 of a second, so you're seeing many occurrences of frame drops like this during a several minute recording? I'm struggling to think if there's anything in the camera system that could do that. If you were failing to keep up I would have expected drops of one or two frames, probably more frequently. Is there anything else running on your system?

It might be worth experimenting with a smaller resolution, or see what happens if you drop the output completely (set `encoder.output = Output()`), maybe the filesystem is occasionally stalling?

medem23, 2023-08-17:
Actually, I made a small mistake. It is either 2 ,3 consecutive frames that are missing or 23-28 consecutive frames that are missing. But that being said, it is not happening too often in 30 minutes recording, my worst result was 27.20, so about 2 minutes and 30 seconds were missing.

Sorry I also forget to mention that I get around 10-20 times this error 'imx219 10-0010': Unable to set control 0x00980911: Input-/Outputerror but that is only for one frame and would not explain this big of a gap.

More info: dropping of frames occurred 203 times for a video recording of 30 minutes, there is also nothing else running, not even preview, I do save the data in external HDD.

davidplowman, 2023-08-17:
That "unable to set control" message seems strange, are you using a short ribbon cable? I see that you're using an imx219, can you also say what kind of a Pi this is?

Beyond that, I would recommend trying some smaller resolutions, and try dropping the output, to find out what seems to be affecting it.

medem23, 2023-08-17:
Pi 4, the ribbon is not that long

davidplowman, 2023-08-17:
Can you say how long that is? Please let me know if those other tests reveal anything. I'll also try and find some time to run a simplified version of your code.

davidplowman, 2023-08-17:
Here are a couple of simple test scripts to try. Run the first one and pass it the number of seconds for which you want it to record video.
```
import sys
import time

from picamera2 import Picamera2
from picamera2.encoders import H264Encoder

picam2 = Picamera2()
config = picam2.create_video_configuration({'size': (1640, 1232)})
picam2.configure(config)
encoder = H264Encoder(5000000)
picam2.start_recording(encoder, "test.h264", pts="test.txt")

for i in range(int(sys.argv[1])):
    print(i, end=" ", flush=True)
    time.sleep(1)
print()

picam2.stop_recording()
```
Then I checked the timestamps with this script (pass it the timestamp filename "test.txt").
```
import sys

with open(sys.argv[1], 'r') as f:
    timestamps = [float(line.rstrip()) for line in f.readlines()]

diffs = [round((next - now) / 33.33) - 1 for now, next in zip(timestamps, timestamps[1:])]

print("Total frames:", len(timestamps), " lost:", int(sum(diffs)))
```
I only ran it for 5 minutes but it reported no frame drops. Could you try this?

medem23, 2023-08-17:
This is great! I will do it tomorrow and let you know. Thank very much for fast replies and help!:)

medem23, 2023-08-21:
So, after some experiments, I found out that the main issue comes from the warning imx219 10-0010': Unable to set control 0x00980911: Input-/Outputerror. For every warning I get around 25 frame drops and not only one as I thought.

davidplowman, 2023-08-21:
That's interesting. I would certainly be suspicious of transmission problems there. I'd check the length of those cables, power supplies, any other peripherals that you've connected, that kind of thing. Of course it may turn out to be something else too...

closed 2023-08-28 by medem23