raspberrypi/picamera2 #783: [BUG] HQ Camera External Syncronization Timing

sam598, 2023-08-27:
I have been successfully generating an external synchronization signal from a Raspberry Pi Pico to trigger an HQ Camera at 10hz.

I use a voltage divided resistors to go from 3.3v to 1.8v. The signal goes from high to low for about 40 nanoseconds, and then remains high for 100 milliseconds. Nothing else is happening on the Pico, so this creates a fairly accurate 10hz signal.

In version 3.7 of picamera2 this works great. When measuring the timestamp metadata between frames, the delta time is just over 100 milliseconds. I have verified that it is indeed the sync signal triggering the exposure, as frames are not being received if the signal is stopped, and a 5hz signal has a delta time just over 200 milliseconds.

However in the latest version of picamera2 that ships with raspberry pi os and python it does not work properly. Instead when giving a 10hz sync signal it ignores every other pulse, and the delta time between frames is just over 200 milliseconds.

This has been tested on multiple pis, HQ cameras, and picos. With the same hardware an SD card with 3.7 works, and swapping it for another SD card with an up to date os and picamera2 does not.

Is there anything that has changed in how libcamera, picamera2, or the operating system handles receiving camera frames that could be the cause for it? Is there a shorter window for when the sync signal can be received, or is there additional processing happening that could be causing frame drops?

```
picam2 = Picamera2()
picam2.configure(picam2.create_still_configuration())
picam2.set_controls({"FrameDurationLimits": (100000, 100000)})
picam2.start()

last_sensorTimestamp = 0

while True :
    request = picam2.capture_request()
    metadata = request.get_metadata()
    delta_time = int(metadata["SensorTimestamp"]) - last_sensorTimestamp
    last_sensorTimestamp = int(metadata["SensorTimestamp"])
    request.release()
    print(delta_time)
```

naushir, 2023-09-07:
Could it possibly be the internal libcamera pipeline handler timeout that triggers every other frame to restart and causes every other frame to timeout?  The timeout code is a fairly recent additions, but I could not tell you exactly which library/image version included this change.

You can try manually overriding the timeout value by adding the following:
```yml
"camera_timeout_value_ms": 999999
```
to ``/usr/local/share/libcamera/pipeline/rpi/vc4/rpi_apps.xml`` (for local install) or ``/usr/share/libcamera/pipeline/rpi/vc4/rpi_apps.xml`` (for packaged install), and see if that solves your issue.

sam598, 2023-09-08:
I don't believe that's the issue.

If we had a long time between synchronization pulses, and that time exceeded camera_timeout_value_ms, then the camera would report an error.

In this case we are sending synchronization signals constantly at the camera's expected framerate (every 100 milliseconds).

The issues is that on the latest software it is missing every other frame, where previously it would trigger and return each frame every 100 milliseconds (10fps).