raspberrypi/picamera2 #785: [HOW-TO] define H264 Profile "Baseline"

LeMajors, 2023-08-28:
I need to define the h264 profile "Baseline" for streaming h264 via websocket to the browser as described here:
https://www.codeinsideout.com/blog/pi/stream-picamera-h264/#websocket-server

The scripts are written for picamera not picamera2. For the old version the code would be e.g.:
`camera.start_recording(frame_buffer, format='h264', profile="baseline")`

How is that done with picamera2?

davidplowman, 2023-09-07:
There is a V4L2 profile control, but as far as I can see it isn't implemented in the python-v4l2 library. We'll have to put that on the list of things to do, and then incorporate it into Picamera2.

puterboy, 2024-01-26:
So is there any way to mimic `profile="baseline"`

davidplowman, 2024-01-29:
I'm sorry that I haven't got round to adding the profile control to python-v4l2. I don't know how else you could force baseline profile, I'm afraid.

LeMajors, 2024-01-29:
The main thing is that it doesn't get thrown out of the todo list. ;-)

BlueWolfProduction, 2024-05-10:
I too just got hit with the exact same issue, following the same guide. Seems this is the only feasible way to stream from picamera2 without ffmpeg (which I am not sure is even possible on a Pi Zero). So count me in for waiting for an option to configure the profile.

@puterboy Did you happen to find an alternative since then or are you still stuck on this?

LeMajors, 2024-05-11:
@BlueWolfProduction: No, I still use the prehistoric MJPEG streaming...

davidplowman, 2024-05-20:
Not sure if this is still relevant, but I believe the latest Picamera2 (and python-v4l2 package) support baseline profile now, so it might be worth re-trying this after an update. Thanks!

LeMajors, 2024-05-20:
First of all, thank you very much, it works well and smoothly!
However, the latency is worse than with MJPEG streaming. Maybe this has something to do with this:

Picam2 Manual:
"On Pi 4 and earlier devices there is dedicated hardware for H264 and MJPEG encoding. On a Pi 5, however, these codecs are
implemented in software using FFmpeg libraries. The performance on a Pi 5 is similar or better, and the images are encoded with
better quality. The JPEG encoder runs in software on all platforms."

Anyway, the real problem is solved. I now work with a Pi 5 and if latency wasn't important, I would use H264. For my robot, if low latency is important, I'll stick with MJPEG streaming.

Greetings, Marcus

davidplowman, 2024-05-20:
Thanks for the update. Yes, H264 encoding is done with FFmpeg on a Pi 5 which incurs more several frames of latency compared to (for example) a Pi 4. JPEG/MJPEG have no such latency. If you're using a Pi 5 you may find the JpegEncoder (which uses libjpeg) is more efficient than the MJPEGEncoder (which is FFmpeg), you'd have to try it and see.

LeMajors, 2024-05-20:
Sorry!
I probably said nonsense half-knowingly. I already _use_ the JpegEncoder and only mentioned MJEPG because the streamed file name is called "stream.mjpg"... :-/