raspberrypi/picamera2 #788: [BUG]  Cannot stream MJPEG

stuartofmt, 2023-09-04:
**Describe the bug**
Testing streaming of USB camera.  Works with Pi camera but not USB.
I realize that full support for USB may not be available, but it seems this is a straightforward use case that should work.

USB camera displays stills in preview with **picam.create_preview_configuration({"format": "MJPEG"})**

Throws an error when attempting to capture video with: 
**picam2.configure(picam2.create_video_configuration({"format": "MJPEG"}))**
Have tried multiple variations of video_configuration (including default) without success.

**To Reproduce**
This following sample code is taken from one of the examples with some added code to provide info about the cameras.
All changes are at the end of the code.

`
#!/usr/bin/python3

# Mostly copied from https://picamera.readthedocs.io/en/release-1.13/recipes2.html
# Run this script, then point a web browser at http:<this-ip-address>:8000
# Note: needs simplejpeg to be installed (pip3 install simplejpeg).

import io
import logging
import socketserver
from http import server
from threading import Condition

from picamera2 import Picamera2
from picamera2.encoders import JpegEncoder
from picamera2.outputs import FileOutput

PAGE = """\
<html>
<head>
<title>picamera2 MJPEG streaming demo</title>
</head>
<body>
<h1>Picamera2 MJPEG Streaming Demo</h1>
[image]
</body>
</html>
"""

class StreamingOutput(io.BufferedIOBase):
    def __init__(self):
        self.frame = None
        self.condition = Condition()

    def write(self, buf):
        with self.condition:
            self.frame = buf
            self.condition.notify_all()

class StreamingHandler(server.BaseHTTPRequestHandler):
    def do_GET(self):
        if self.path == '/':
            self.send_response(301)
            self.send_header('Location', '/index.html')
            self.end_headers()
        elif self.path == '/index.html':
            content = PAGE.encode('utf-8')
            self.send_response(200)
            self.send_header('Content-Type', 'text/html')
            self.send_header('Content-Length', len(content))
            self.end_headers()
            self.wfile.write(content)
        elif self.path == '/stream.mjpg':
            self.send_response(200)
            self.send_header('Age', 0)
            self.send_header('Cache-Control', 'no-cache, private')
[... 109 lines cut ...]

{'use_case': 'video', 'transform': <libcamera.Transform 'identity'>, 'colour_space': <libcamera.ColorSpace 'Rec709'>, 'buffer_count': 6, 'queue': True, 'main': {'format': 'XBGR8888', 'size': (1280, 720)}, 'lores': None, 'raw': None, 'controls': {}, 'display': 'main', 'encode': 'main'}

[43:40:03.056194782] [7540]  INFO Camera camera_manager.cpp:297 libcamera v0.0.5+83-bde9b04f
[43:40:03.130713790] [7552]  WARN RPI vc4.cpp:383 Mismatch between Unicam and CamHelper for embedded data usage!
[43:40:03.132523094] [7552]  INFO RPI vc4.cpp:437 Registered camera /base/soc/i2c0mux/i2c@1/imx219@10 to Unicam device /dev/media3 and ISP device /dev/media0
[43:40:03.132719551] [7552]  INFO RPI pipeline_base.cpp:1101 Using configuration file '/usr/share/libcamera/pipeline/rpi/vc4/rpi_apps.yaml'
[43:40:03.159512706] [7540]  INFO Camera camera.cpp:1033 configuring streams: (0) 1280x720-MJPEG
Traceback (most recent call last):
  File "/home/pi/videostream/./videostream-lite.py", line 103, in <module>
    picam2.start_recording(JpegEncoder(), FileOutput(output))
  File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 1597, in start_recording
    self.start_encoder(encoder, output, pts=pts, quality=quality, name=name)
  File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 1518, in start_encoder
    _encoder.format = streams[name]['format']
  File "/usr/lib/python3/dist-packages/picamera2/encoders/encoder.py", line 156, in format
    formats.assert_format_valid(value)
  File "/usr/lib/python3/dist-packages/picamera2/formats.py", line 39, in assert_format_valid
    raise ValueError(f"Invalid format: {fmt}. Valid formats are: {ALL_FORMATS}")
ValueError: Invalid format: MJPEG. Valid formats are: {'NV21', 'YUYV', 'R12', 'RGB888', 'R10', 'R8_CSI2P', 'SRGGB10', 'SGBRG12', 'SGRBG8', 'SGBRG12_CSI2P', 'SGBRG10_CSI2P', 'SGRBG12_CSI2P', 'R12_CSI2P', 'SRGGB12_CSI2P', 'NV12', 'BGR888', 'SGRBG12', 'SGBRG8', 'SRGGB12', 'SGBRG10', 'SRGGB10_CSI2P', 'UYVY', 'VYUY', 'YVYU', 'SBGGR12_CSI2P', 'XRGB8888', 'R8', 'YVU420', 'XBGR8888', 'SGRBG10', 'SBGGR10', 'SBGGR10_CSI2P', 'SGRBG10_CSI2P', 'SBGGR12', 'SRGGB8', 'SBGGR8', 'R10_CSI2P', 'YUV420'}

`

**Hardware :**
Pi 3B+ Glamor enabled Bullseye 64

davidplowman, 2023-09-07:
Hi, I'm afraid support for USB cameras is very limited, and there are no plans to extend support for them.

stuartofmt, 2023-09-07:
Understood.  To what extent is there support for USB cameras? 

I'm asking only so I don't go down a rabbit hole of investigation.  I can (and have) handled USB cameras with openCV  but was hoping to avoid that due to the various problems getting openCV to install on pi's of different stripes. 

In other words - even experienced folks may have to try several techniques to get openCV installed.  My program is aimed at inexperienced folks :-)

Probably what I'll do is create a "lite" version just for dedicated pi cameras.

davidplowman, 2023-09-07:
I think the support extends to getting frames out of USB cameras, and displaying them in the slow (non-EGL accelerated) preview. I think you can save it as a JPEG as well (which obviously doesn't need to do an extra JPEG encode now). I use USB cams mostly for testing multi-camera functionality without requiring a compute module!

You can't do "video encode" so far as I know, though there's a chance you could make something work by tossing the JPEG frames at the base Encoder class, for which encode is a no-op. I suppose that might even allow you to stream them, though I don't think USB cams are sufficiently important to us to devote resources to that.

I recommend installing OpenCV from apt (sudo apt install python3-opencv, I think). I've had no end of trouble installing from other sources, where it goes off and spends a day compiling stuff and then failing at the final hurdle. I would install Qt from apt as well, otherwise it can assume a different version of OpenCV from what you have (if you didn't get OpenCV from apt). It's all a bit of a minefield.