raspberrypi/picamera2 #791: Picamera2 with raspberry pi global shutter camera returns one color background image instead of the real image

aymeric75, 2023-09-06:
Hello,

I have a **raspberry pi global shutter camera**  on a  raspberry pi 4 model B

I ran this code:
```
#!//usr/bin/python3

from picamera2 import Picamera2

picam2 = Picamera2()
config = picam2.create_still_configuration()
picam2.configure(config)

picam2.start()

np_array = picam2.capture_array()
print(np_array)
picam2.capture_file("demo.jpg")
picam2.stop()

```

And it produces this image:

[image]

aymeric75, 2023-09-07:
yes it does, I am wondering, is having a lens mandatory ? because I don't have one

davidplowman, 2023-09-07:
Yes, you need to get a lens of some sort, maybe like this https://thepihut.com/products/raspberry-pi-high-quality-camera-lens (other lenses and suppliers are available, this is just one example)

aymeric75, 2023-09-07:
so, the problem comes from the fact that I don't have a lens ? thanks

sam598, 2023-09-08:
Yes. All cameras need lenses.

An image sensor by itself will just capture all of the light that hits it. Lenses focus the light into a coherent image.

https://en.wikipedia.org/wiki/Camera_lens

The one exception would be if you create a pinhole camera, but even then the pinhole is considered a lens.

https://en.wikipedia.org/wiki/Pinhole_camera

aymeric75, 2023-09-12:
Hello, I have put the lens, but it now returns a black image

davidplowman, 2023-09-12:
Hi, could you please double-check that you've opened the lens aperture. Might it be possible to attach a small image of your Pi and the connected camera? Thanks.

aymeric75, 2023-09-12:
Ok, I found the solution, it was a matter of setting the lens (more light, more focus)

closed 2023-09-12 by aymeric75