raspberrypi/picamera2 #792: [HOW-TO] Rotate on 90?

IgorChuz, 2023-09-08:
Rotate on 90?

davidplowman, 2023-09-08:
I'm afraid the camera system doesn't support this. It's often easiest to rotate files after capture by setting the file metadata, or rotating them upon playback. Might this be possible for you?

jvoss11, 2023-10-31:
@davidplowman as i understand you.  you are stating that a software solutions is not possible to rotate the camera image 0-359 degrees and that it is only possible to rotate the image 180?

is this how i understand this?  if so then what is this.

 * Copyright (C) 2020, Raspberry Pi Ltd
 *
 * transform.cpp - 2D plane transforms.

 it clearly states that this is possible with the following information.
 	static const char *strings[] = {
		"identity",
		"hflip",
		"vflip",
		"hvflip",
		"transpose",
		"rot270",
		"rot90",
		"rot180transpose"  
		
so how do we interact with is....

thank you.

davidplowman, 2023-11-06:
The Raspberry Pi doesn't support rotating by 90 degrees. The transform class which you've found forms part of a general API in libcamera that would allow it if the underlying platform did, but as I've said, the Pi doesn't. Sorry about that. Often there are other ways to handle 90 degree rotations - can you say anything more about your use case?

jvoss11, 2023-11-06:
you can read here https://github.com/Gordon999/Pi_LIbCamera_GUI/issues/15#issuecomment-1795491710