raspberrypi/picamera2 #793: [OTHER] ImportError: cannot import name 'QGlPicamera2' from 'picamera2.previews.qt'

ajkelsey, 2023-09-09:
I'm receiving this error when attempting to access the camera. `ImportError: cannot import name 'QGlPicamera2' from 'picamera2.previews.qt'`

I tried using this example code straight from the Picamera2 docs and I received the same error as well.
```
from picamera2 import Picamera2, Preview
import time
picam2 = Picamera2()
camera_config = picam2.create_preview_configuration()
picam2.configure(camera_config)
picam2.start_preview(Preview.QTGL)
picam2.start()
time.sleep(2)
picam2.capture_file("test.jpg")
```

Any help on how I can resolve this would be appreciated.

ajkelsey, 2023-09-09:
It seem I have found the issue. Installing python3-opengl resolved it. However, now I'm getting `Exception: EGL_EXT_image_dma_buf_import missing`.

davidplowman, 2023-09-11:
Can you say what image or OS you are using? Normally I would recommend starting with a fresh install of the latest Raspberry Pi OS image, and everything should "just work". Once something has happened that breaks this, it's often a bit tricky trying to unpick it again.

ajkelsey, 2023-09-11:
I was running the lite version and decided to install the desktop version for the same reason you suggested. I have since determined that the source of my problem is that I am connected via VNC and can't display the camera stream at the same time. I mainly wanted to be able to do that to check the code of a project I am working. For the meantime, I will just save images to file and view them that way. I'm thinking that if I can output the camera to an http stream, I could view it that way.