raspberrypi/picamera2 #794: [BUG] capture_array() hangs after random amount of time

willemkempers, 2023-09-11:
**Please only report one bug per issue!**

**Describe the bug**
A clear and concise description of what the bug is.

`capture_array()` hangs after a random amount of time - I've seen this range from 10 seconds to ~20 minutes.

**To Reproduce**
Steps to reproduce the behaviour. Wherever possible, please include a small self-contained Python script that shows the bug.

Below is a mildly stripped down version of the code I'm using. The code I've omitted has to do with sending data over OSC, this issue played up before that was added. I haven't tried stripping out the opencv blob detection code, but right now I do not have access to the Raspberry Pi that runs this code to check if it has anything to do with hang caused by `capture_array()`. I tracked the issue down by adding print statements, the last print I would see is the one just before `capture_array()`.

```python
#!/usr/bin/env python3

# System imports
import os
import sys
import time
import math

# Third-party imports
import cv2
from libcamera import controls
import numpy as np
from picamera2 import Picamera2

def main():
    # Create opencv window thread
    cv2.startWindowThread()

    # Size of capture
    radius = 256
    diameter = radius * 2
    size = (diameter, diameter)

    # Create picamera 2 object
    picam2 = Picamera2()
    config = picam2.create_preview_configuration({"size": size}, raw=picam2.sensor_modes[1])
    picam2.align_configuration(config)
    picam2.configure(config)
    picam2.set_controls({"FrameRate": 30, "AfMode": controls.AfModeEnum.Manual, "LensPosition": 0.0})
    picam2.start()

    # --- Setup Blobdetector ---

    # Setup SimpleBlobDetector parameters.
    params = cv2.SimpleBlobDetector_Params()

    # Change thresholds
    params.minThreshold = 4
    params.maxThreshold = 64
    params.thresholdStep = 32

    # Filter by Area.
    params.filterByArea = True
    params.minArea = 4
    params.maxArea = 200

    # Filter by Circularity
    params.filterByCircularity = True
    params.minCircularity = 0.3

    # Filter by Convexity
    params.filterByConvexity = False
    params.minConvexity = 0.87

    # Filter by Inertia
    params.filterByInertia = True
    params.minInertiaRatio = 0.25

    # Create a detector with the parameters
    detector = cv2.SimpleBlobDetector_create(params)

    # used to record the time when we processed last frame
    prev_frame_time = 0

    # used to record the time at which we processed current frame
    new_frame_time = 0

    # font which we will be using to display FPS
    font = cv2.FONT_HERSHEY_SIMPLEX

    # --- Mask ---

    # create circular mask
    circular_mask = np.zeros(size, dtype="uint8")
    cv2.circle(circular_mask, (radius, radius), radius, diameter, -1)

    # --- Points of interest

    x = radius
    y = radius

    # --- Execution loop ---

    while True:
        # Raw capture from camera
        capture = picam2.capture_array()

        # mask image to focus on area we want
        image = cv2.bitwise_and(capture, capture, mask=circular_mask)

        grey = cv2.cvtColor(image, cv2.COLOR_BGR2GRAY)

        invert = cv2.bitwise_not(grey)

        keypoints = detector.detect(invert)

        # get largest keypoint
        size = 0
        for keypoint in keypoints:
            if keypoint.size > size:
                size = keypoint.size
                x = keypoint.pt[0]
                y = keypoint.pt[1]

        im_with_keypoints = cv2.drawKeypoints(invert, keypoints, np.array([]), (0,0,255), cv2.DRAW_MATCHES_FLAGS_DRAW_RICH_KEYPOINTS)

        # time when we finish processing for this frame
        new_frame_time = time.time()

        # Calculating the fps

        # fps will be number of frame processed in given time frame
        # since their will be most of time error of 0.001 second
        # we will be subtracting it to get more accurate result
        fps = 1 / (new_frame_time - prev_frame_time)
        prev_frame_time = new_frame_time

        # converting the fps into integer
        fps = int(fps)

        # converting the fps to string so that we can display it on frame
        # by using putText function
        fps = str(fps)

        # putting the FPS count on the frame
        cv2.putText(im_with_keypoints, fps, (7, 70), font, 3, (100, 255, 0), 3, cv2.LINE_AA)

        # Show on screen
        cv2.imshow("keypoints", im_with_keypoints)
        cv2.waitKey(1)

def quit():
    # Clean-up

    """ --- Exit --- """
    try:
        sys.exit(0)
    except SystemExit:
        os._exit(0)

# Main body
if __name__ == '__main__':
    # Arguments

    """ --- Main script --- """
    try:
        main()
    except KeyboardInterrupt:
        quit()
```

**Expected behaviour**
A clear and concise description of what you expected to happen.

Continuous execution

**Console Output, Screenshots**
If applicable, any console output or screenshots that show the problem and associated error messages.

**Hardware :**
 Please state what kind of Pi you are using, and what camera is attached.

Raspberry Pi Camera Module 3 NoIR wide-angle
Raspberry Pi 4 Model B 2GB running Raspberry Pi OS (64-bit)

**Additional context**
Add any other context about the problem here.

I saw similar reports on GitHub, but none seemed close enough so I thought I'd create this report. Please forgive me if there's an obvious mistake, I'm working towards a show and under some pressure :-)

davidplowman, 2023-09-11:
Thank you for the report. I don't see anything obviously wrong with what you've got, and the symptom is very much as if the camera has stopped sending you frames. We probably need to strip the code back further and do some more testing to see if we can find the cause of the problem. Here are some things to try.

1. Are there any console messages being printed out that might suggest a problem?
2. Can you strip out all the OpenCV code? Just run the loop with nothing but `picam2.capture_array()` in it. Does that lock up?
3. Does it lock up eventually if you simply run `libcamera-hello -t 0`?
4. How long is the ribbon cable that you are using? Do you have another ribbon cable to try?
5. Do you have another camera module to try?

Thanks!

davidplowman, 2023-09-11:
I have to say it doesn't "smell" like a memory problem to me. I'm thinking cables, power supplies, hardware all feel more likely, though you never can tell. But please post again when you know more.

willemkempers, 2023-09-12:
Received the new hardware earlier today - it's been running without errors for a bit over 2hrs now. No auto-focus related error messages either. Not 100% sure what was causing the hang, but it seems to have been fixed by a fresh OS + new hardware..!

willemkempers, 2023-09-15:
OK so I'm currently on site and have swapped out the old hardware for the new hardware. The new hardware worked fine in the studio (had it run for multiple hours), but now on site I'm experiencing the same hang again. I am seeing the error that I mentioned before as well - the one related to autofocus - I didn't see this error when I was testing in the studio. The error is `ERROR IPARPI cam_helper_imx708.cpp:273 PDAF data in unsupported format`.

davidplowman, 2023-09-15:
Hi again, yes power supplies are a depressingly common cause of intermittent and otherwise hard-to-explain problems. There are many out there that don't supply the correct (or even claimed) current and voltages reliably. It would certainly be worth trying different power supplies, perhaps an official Raspberry Pi one if you have one. I don't recall if you said how long the ribbon cable is, but if it's longer than 50cm I would consider trying a shorter one as well.

willemkempers, 2023-09-15:
Thanks for your replies, very helpful! The ribbon cable we're using is the default one, eyeballing that it's roughly 10cm so doubt that's the issue. We'll try swapping the power supply!

willemkempers, 2023-10-15:
I'll close this issue with this comment. After some experimentation, it seemed like either the Pi or the camera was overheating, I currently don't think it was a power supply issue as we're using a pretty decent Meanwell power supply and it didn't seem to make a difference once we started swapping psu's.
The RPi and camera live in a box, I've had it run for 12+ hours while the lid was off. Once the lid was on, it would take about 20 minutes before the hangs started.

Thanks for your help!

closed 2023-10-15 by willemkempers