raspberrypi/picamera2 #799: [HOW-TO] Poor performance for picamera2 and ROS2

BurmazDaniel, 2023-09-18:
**Describe what it is that you want to accomplish**
I am using the IMX219-160 on a Raspberry Pi 4 running ROS 2 Humble. I am tring to make a simple publisher node for my camera using picamera2 but unfortunately the fps is pretty low. At 1640 x 1232 i get around 0.1 fps. At 480p i get around 7 fps. 

`
#!/usr/bin/python3
import rclpy
from rclpy.node import Node
import cv2
from sensor_msgs.msg import Image
from picamera2 import Picamera2
from cv_bridge import CvBridge, CvBridgeError

class ImagePublisher(Node):

    def __init__(self):
        super().__init__("image_publisher")
        self.pub = self.create_publisher(Image, "/image", 10)
        timer_period = 0
        self.timer = self.create_timer(timer_period, self.timer_callback)
        self.bridge = CvBridge()
        self.cap = Picamera2()
        video_config = self.cap.create_video_configuration(main={"size": (1640, 1232), "format": "RGB888"})
        self.cap.configure(video_config)
        self.cap.start()

    def timer_callback(self):
        img=self.cap.capture_array()

        scale_percent = 20 # percent of original size
        width = int(img.shape[1] * scale_percent / 100)
        height = int(img.shape[0] * scale_percent / 100)
        dim = (width, height)
        img = cv2.resize(img, dim, interpolation = cv2.INTER_AREA)
        
        self.pub.publish(self.bridge.cv2_to_imgmsg(img, "rgb8"))
        self.get_logger().info('Publishing...')

def main(args=None):
    rclpy.init(args=args)
    image_publisher = ImagePublisher()
    rclpy.spin(image_publisher)
    image_publisher.destroy_node()
    rclpy.shutdown()

if __name__ == '__main__':
    main()
`
Not a single core reaches 100% at 480p. Not even 50%.

**Describe alternatives you've considered**
I tried different resolutions, even 164x123 i could not get past 15fps.
I tried to send a video source insted of a picamera2 capture and i have a much better performance.
Maybe it has to do with the array from Picamera2 capture_array()?

**Additional context**
Thank you!

davidplowman, 2023-09-19:
Hi, have you profiled the code to see where it spends its time? A video configuration with the imx219 2x2 binned camera mode should be able to give you 30fps (it might be worth checking that, by running the camera with no other processing going on). Is it perhaps the resize that is taking time? There's certainly no reason not to configure the camera to give you the resolution you really want, which would give you the resize completely free.

BurmazDaniel, 2023-09-20:
> Hi, have you profiled the code to see where it spends its time? A video configuration with the imx219 2x2 binned camera mode should be able to give you 30fps (it might be worth checking that, by [...]

Thank you so much for the tip! This is my first time profiling python code. I used `python -m cProfile -o my_program.prof cameratotopic.py` and then snakeviz to view the file.
[my_program.txt](https://github.com/raspberrypi/picamera2/files/12667824/my_program.txt)
If i am correct my problem is not capture_array() because it takes 1.62s but cv2_to_imgmsg that takes way more time to run. 
Am i right?
Thank you so much!

davidplowman, 2023-09-20:
Hi, I'm afraid I didn't know how to interpret that file - it didn't actually look like a text file to me. But nonetheless, I'm suspicious that it's the image processing that's taking the time. Maybe just remove the image processing from your code and check that you get 30fps as you expect. As I suggested previously, can you ask the camera system for the image size that you actually want?

popcornmix, 2023-09-20:
@davidplowman `pip install snakeviz; snakeviz my_program.txt` should open profiling info in a web browser.

davidplowman, 2023-09-20:
Thanks. I don't know if I'm reading it right, but is it spending a long time in cv2_to_imgmsg? That's probably something to look into. Could you get by with asking for 'passthrough' rather than 'rgb8'? But I'm afraid I don't really know anything about this software....

BurmazDaniel, 2023-09-20:
> Thanks. I don't know if I'm reading it right, but is it spending a long time in cv2_to_imgmsg? That's probably something to look into. Could you get by with asking for 'passthrough' rather than 'rgb8'? But I'm afraid I don't really know anything about this software....

That's all! Thank you so much for helping me! Have a great day!

closed 2023-09-20 by BurmazDaniel