raspberrypi/picamera2 #801: [HOW-TO] First Example in Picamera2 Library Documentation Fails

baqwas, 2023-09-23:
**Please only ask one question per issue!**

What are the additional prerequisite packages to run the First Example in the Picamera2 Library user guide.

**Describe what it is that you want to accomplish**
I want to run [A first example](https://datasheets.raspberrypi.com/camera/picamera2-manual.pdf) as documented on page 6.

**Describe alternatives you've considered**

- from picamera2 import Picamera2, Preview
- Glamor is enabled
- Validated that v4l2-python3 is current:

```
$ python3 -m pip install v4l2-python3
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: v4l2-python3 in /home/chowkidar/.local/lib/python3.9/site-packages (0.3.1)
$ 
```

**Additional context**

```
Traceback (most recent call last):
  File "/home/chowkidar/projects/picamera2/example1.py", line 27, in <module>
    from picamera2 import Picamera2, Preview
  File "/usr/lib/python3/dist-packages/picamera2/__init__.py", line 9, in <module>
    from .picamera2 import Picamera2, Preview
  File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 23, in <module>
    from picamera2.encoders import Encoder, H264Encoder, MJPEGEncoder, Quality
  File "/usr/lib/python3/dist-packages/picamera2/encoders/__init__.py", line 2, in <module>
    from .h264_encoder import H264Encoder
  File "/usr/lib/python3/dist-packages/picamera2/encoders/h264_encoder.py", line 5, in <module>
    from v4l2 import (V4L2_CID_MPEG_VIDEO_H264_I_PERIOD,
ImportError: cannot import name 'V4L2_CID_MPEG_VIDEO_H264_MAX_QP' from 'v4l2' (/home/chowkidar/.local/lib/python3.9/site-packages/v4l2.py)

------------------
(program exited with code: 1)
Press return to continue
```

Platform: Raspberry Pi 3 Model B Plus Rev 1.3
OS: Raspbian GNU/Linux 11 (bullseye)
Linux 14 6.1.32-v7+ #1656 SMP Wed Jun  7 11:31:19 BST 2023 armv7l GNU/Linux

davidplowman, 2023-09-25:
Hi, thanks for the report. Could you try fetching python-v4l2 from apt instead of pip? Or alternatively try a pip install from the git repo (using pip install git+https). I'm wondering if there's a problem with the maintenance of the pip package (this package was being maintained by the Foundation and the person who used to do it has since left, so maybe there's something that needs sorting out here). Thanks.

baqwas, 2023-09-25:
Will do as advised (fetch _python3-v4l2_ using apt instead of pip). Will report back here as a comment in due course. Thanks.

Regards.

closed 2023-09-25 by baqwas

baqwas, 2023-09-25:
Interim update:
Here is the console terminal window for basic detection of the **v4l2** package:

```
>>> import v4l2
>>> import fcntl
>>> cp = v4l2.v4l2_capability()
>>> vd=open('/dev/video0', 'r')
>>> fcntl.ioctl(vd, v4l2.VIDIOC_QUERYCAP, cp)
0
>>> cp.driver
b'unicam'
>>> cp.card
b'unicam'

```

Will work on _uninstall/reinstall_ shortly. Thanks.

Regards.

baqwas, 2023-09-25:
Hello @davidplowman,

Your suggestion worked flawlessly. I may have unwittingly done something that v4l2 was not comfortable in doing but now with:
```
Setting up python3-v4l2 (0.3.2-1) ...
Setting up python3-picamera2 (0.3.12-2) ...

```
I am back working with yet another deployment of picamera2 with slightly different hardware thanks again to your patience and understanding:

```
[0:24:02.885220866] [4411]  INFO Camera camera_manager.cpp:297 libcamera v0.0.5+83-bde9b04f
[0:24:02.937900392] [4418]  WARN RPI vc4.cpp:383 Mismatch between Unicam and CamHelper for embedded data usage!
[0:24:02.939100067] [4418]  INFO RPI vc4.cpp:437 Registered camera /base/soc/i2c0mux/i2c@1/imx219@10 to Unicam device /dev/media2 and ISP device /dev/media0
[0:24:02.939186473] [4418]  INFO RPI pipeline_base.cpp:1101 Using configuration file '/usr/share/libcamera/pipeline/rpi/vc4/rpi_apps.yaml'
[0:24:02.959456427] [4411]  INFO Camera camera.cpp:1033 configuring streams: (0) 640x480-XBGR8888 (1) 640x480-SBGGR10_CSI2P
[0:24:02.960474437] [4418]  INFO RPI vc4.cpp:565 Sensor: /base/soc/i2c0mux/i2c@1/imx219@10 - Selected sensor format: 640x480-SBGGR10_1X10 - Selected unicam format: 640x480-pBAA

------------------
(program exited with code: 0)
Press return to continue
```

You may have mentioned elsewhere something about the WARN message in this output. I don't know what "Mismatch between Unicam and CamHelper for embedded data usage" means but it is definitely above my job grade.

Many, many thanks for helping me to move along albeit at a slower pace (and some outdated hardware) but it helps me to incrementally refine solutions with diverse uses of picamera2.

Regards.