raspberrypi/picamera2 #802: [OTHER] start_encoder() needs a parameter?

baqwas, 2023-09-25:
**Please only include one item/question/problem per issue!**
In **capture_circular.py**, line 20:
The following statement did not work for me:
`picam2.start_encoder()`
However, the following change worked:
`picam2.start_encoder(encoder)`

Regards.

P.S.
Apologies for not submitting a pull request. I'm still learning how to use GitHub productively.

davidplowman, 2023-09-26:
Hi, thanks for the report. Yes, it seems we aren't running some of those as part of our CI test suite because it runs forever! There are a few others to fix up as well, so thanks again for letting us know.

baqwas, 2023-09-26:
Will continue to step through the examples one script at a time (alphabetical order, of course). Hopefully, I'll learn a little bit about Picamera2 too in the process.

Regards.

closed 2023-09-26 by baqwas