raspberrypi/picamera2 #807: [BUG] Audio/Video can't stop encoder/record

RyanABahr, 2023-09-30:
**Please only report one bug per issue!**

**Describe the bug**
Stopping an audio video capture, such as the audio_video_capture.py example script, seems to hang. 

**To Reproduce**
Run audio_video_capture.py. May be hardware dependent.

**Expected behaviour**
In the past, the code would stop immediately. 

**Console Output, Screenshots**
If applicable, any console output or screenshots that show the problem and associated error messages.

**Hardware :**
Pi Zero W2. I've tried multiple power supplies based off comments in different thread, no dice. 
Intel USB Plug and Play microphone. 
Native camera sensor, forgot exact version but can provide if necessary.

**Additional context**
I had code working, fried a microsd card, did a fresh install. The udpated version of picamera2 required me to add the encoder to the start_encoder() function, which was my first fix. After that, I realized all my code that had audio=true would not stop the encoder and hang. I then decided to start from the other direction, and started from the examples to determine if I was missing additional updates/changes required for my code to run. From what I can tell, the audio_video_capture.py hangs. Using old code that had stop_encoder() hangs. Trying stop_recording() seems to still have the issue as well.

Let me know if you believe I need to provide additional relevant information. Thanks so much for your work and effort into this package.

cambridgeBlueMan, 2023-10-17:
I too can confirm this behavour. I'm also pretty sure that this code (audio_video_capture.py ) has worked for me previously.

Camera is standard HQ camera, and OS is bullseye, pi is a 4b 8Gb 

The expected output file is created, but is not accessible, until the program is interrupted (see below). Once interrupted the output file can be opened and played.

Having had a brief look at the code inffmpegoutput.py, it seems that the program hangs waiting for the subprocess to indicate that it has completed as at line 80:

self.ffmpeg.wait()

I'm not quite sure where to go from there, but hopefully this helps! :)

-----------------------------------------------------------------------------------------

if I interrupt using Ctrl-C then I get the following error message:

^CTraceback (most recent call last):
  File "/home/lea/Code/picamera2/examples/audio_video_capture.py", line 19, in <module>
    picam2.stop_recording()
  File "/home/lea/Code/picamera2/picamera2/picamera2.py", line 1603, in stop_recording
    self.stop_encoder()
  File "/home/lea/Code/picamera2/picamera2/picamera2.py", line 1546, in stop_encoder
    encoder.stop()
  File "/home/lea/Code/picamera2/picamera2/encoders/encoder.py", line 245, in stop
    out.stop()
  File "/home/lea/Code/picamera2/picamera2/outputs/ffmpegoutput.py", line 81, in stop
    self.ffmpeg.wait()
  File "/usr/lib/python3.9/subprocess.py", line 1189, in wait
    return self._wait(timeout=timeout)
  File "/usr/lib/python3.9/subprocess.py", line 1919, in _wait
    (pid, sts) = self._try_wait(0)
  File "/usr/lib/python3.9/subprocess.py", line 1877, in _try_wait
    (pid, sts) = os.waitpid(self.pid, wait_flags)
KeyboardInterrupt

cambridgeBlueMan, 2023-10-17:
I can give some time to this over the next couple of days. I'll keep you posted, though I may have the odd question :)

davidplowman, 2023-10-17:
Thanks, that would be very helpful!

davidplowman, 2023-10-18:
Hi, thanks for that, that's very helpful. I wonder if sending a SIGINT might be better the terminate(), I'd be a bit worried that all the video might not get flushed. Though I don't know if FFmpeg treats a SIGINT like a SIGTERM or not. Maybe one could close the input pipe, wait() with a short-ish timeout, and then send a SIGINT if wait() doesn't finish. Not sure!!

davidplowman, 2023-10-19:
I've done some experimentation.

Recording a 5 second stream with no audio, using ffmpeg.wait() results in a file with 150 frames. Using ffmpeg.terminate() results in a file with between 140 and 149 frames. Sending a SIGINT seems identical to using terminate().

When I add audio, using terminate() gives me a file also with 140 to 149 frames, so audio makes no difference in this regard. So what I might do is use wait() with a short timeout (maybe 1 second). If it times out (as it will with an audio stream) then I'll terminate() it.

So I guess it's a choice of (possibly, slightly randomly) truncating the video stream, or letting the audio run on a little. I'll leave it so that the application can set the output.timeout field back to zero if it wants the video truncation behaviour.

Do that sound reasonable? https://github.com/raspberrypi/picamera2/pull/828

cambridgeBlueMan, 2023-10-19:
That seems pretty reasonable at this stage, and I suspect from my own local
experiments would close this :)

Could I make a further point? And again this may be crazy :)

I assume from your figures that you are running at 30fps (or (33333,33333).
so a loss of 0 to 9 frames is anything up to a third of a second. A third
of a second is the default audio_sync argument, and you must have decided
on that for some reason?

Could it be that the video is not truncated, but rather that there are
frames missing at the beginning. Thus because it is taking a short while
for ffmpeg to get up and running when an audio stream is present, the
earliest video frames are lost.

I say this because from my own experience running a picamera capture
alongside an audio capture with a program such as arec there were no sync
issues to speak of.

I'll try and run some tests on this by capturing a video of a program that
has both audio and video output representation of a clock

I note your comment in the program file:

"Because we are prepared to accept whatever parameters and values that
FFmpeg supports,
    there is generally no checking up at this level. That may change over
time as we
    develop better expectations as to what can and cannot work."

I believe this will leave you hostage to fortune over time and there will
probably be a need for exception checking around the subprocess which runs
ffmpeg?

I realise that final point is a separate issue, so I am going to post it as
a [suggestion]