raspberrypi/picamera2 #813: [BUG] Python cannot find the module after manually changing python version

Skillnoob, 2023-10-03:
**Describe the bug**
After changing the python version to 3.11.6 (Due to the shipped version being quite old and 3.11.6 having performance improvements) using this [guide](https://hub.tcno.co/pi/software/python-update/) my python script which uses the picamera2 module python cannot find the module anymore.

**To Reproduce**
1. Install Python 3.11.6 using this [guide](https://hub.tcno.co/pi/software/python-update/)
2. Install the picamera2 module using the apt method in the README
3. Run a python script that uses the picamera2 module

**Expected behaviour**
The script would execute without any errors and work as it would with the Python version that is installed by default

**Console Output, Screenshots**
`pi@raspberrypi:~/Desktop/Overengineering-squared/new Programm/Python $ python cam.py
Traceback (most recent call last):
  File "/home/pi/Desktop/Overengineering-squared/new Programm/Python/cam.py", line 6, in <module>
    from picamera2 import Picamera2
ModuleNotFoundError: No module named 'picamera2'`

**Hardware :**
Raspberry Pi 4b with 4GB of ram
Camera Module V3

davidplowman, 2023-10-04:
Hi, unfortunately the libcamera and kms++ packages (these are not Raspberry Pi packages) do not have Python bindings that are installable with different versions of Python, so this is not possible. Actually we are looking at the problem but the main issue is that the libcamera library does not have sufficient ABI stability for it to be easy to create Python bindings for different versions of Python, and different versions of libcamera, that will actually work.

Skillnoob, 2023-10-04:
Thanks for the answer.
Guess il have to use python 3.9.x until the Pi 5 releases which hopefully ships with a newer version of python as its based on a newer debian where the issues are hopefully fixed.

tvoverbeek, 2023-10-04:
FYI Debian Bookworm uses Python 3.11.x

closed 2023-10-04 by Skillnoob