raspberrypi/picamera2 #815: [HOW-TO] Set CircularBuffer for a specific size in seconds ...

lmitcham, 2023-10-05:
Platform - Pi Zero 2W
OS - Bullseye, version 11
Camera - Pi HQ

I have a project that records to a CircularBuffer and then stores it on command.  The desired size of the CircularBuffer is 10 seconds, but the buffer seems to store more, for example 15 seconds instead of only the last 10 seconds of video.  Is there something that I am missing?

The following is the pertinent code section ...

RECORD_FOR_SECONDS     = 10    # Video clip length in seconds
RECORD_FPS                      = 15    # Camera's frame per second setting

CAMERA = Picamera2()
config = CAMERA.create_video_configuration(main={"size":(2028, 1080)}, raw={"size":(4056, 3040)})
config["transform"] = libcamera.Transform(hflip=0, vflip=1)
CAMERA.configure(config)
CAMERA.controls.FrameDurationLimits = (66667, 66667)    # Set to 15 fps
encoder = H264Encoder(bitrate=10000000)
OUTPUT = CircularOutput(buffersize=int( RECORD_FOR_SECONDS * RECORD_FPS ), outputtofile=False)

CAMERA.start_recording(encoder, OUTPUT)
#    record for some length of time ...
OUTPUT.fileoutput = RAM_DISK + VIDEO_CLIP_FILENAME + FILE_TYPE_H264
time.sleep(0.1)
OUTPUT.stop()
CAMERA.stop_recording()

davidplowman, 2023-10-05:
Thanks for the question.

Looking at the code I don't really see how the circular queue could be anything other than the value you have set  (15 * 10 = 150), so that seems correct. I guess you could check the size of the queue just before you stop the the output.

Maybe we need to look at the framerate? The way you're setting it 15fps looks fine, but again, you could capture some metadata and check it is really reporting the value you have set. You could even count frames for a few seconds and check how many you've seen.

That also leaves the question of how you play the clip back. Remember that plain h.264 streams don't have any proper timestamp information in them, so things can definitely go wrong here. Can you say exactly how you are determining the length of the video? You can count frames in an h.264 file with a command like this `ffmpeg -i file.h264 -c copy -f null /dev/null` and look for `frame=` near the end.

lmitcham, 2023-10-05:
I use MP4 to convert it to mp4

cmd = "sudo MP4Box -add " + fromFile + ":fps=15 " + toFile

Perhaps the conversion adds the extra time

davidplowman, 2023-10-05:
By "record time was 28 seconds" do you mean that in your code you set RECORD_FOR_SECONDS to 28? An by "stored mp4" time do you mean the duration reported by the mp4 file after conversion using MP4Box?

lmitcham, 2023-10-05:
No.  My RECORD_FOR_SECONDS = 10 and my "record for some length of time ..." was 28 seconds

lmitcham, 2023-10-05:
Here is the MP4box output for the 32 second record time that produced a 15 second MP4

FILE CONVERSION_THREAD: BEGIN CONVERTING FILE TO MP4: 10100_20231005_083517
FILE CONVERSION THREAD: Store copy of SRC FILE ON RAM DISK as video.h264
2023-10-05 08:35:50:528911
FILE CONVERSION THREAD: FROM SRC FILE ON RAM DISK: /var/tmp/10100_20231005_083517.h264
FILE CONVERSION THREAD: sudo rm '/media/usb0/10100_20231005_083517.mp4'
FILE CONVERSION THREAD: TO DEST FILE ON USB: /media/usb0/10100_20231005_083517.mp4
FILE CONVERSION THREAD: sudo MP4Box -add /var/tmp/10100_20231005_083517.h264:fps=15 /media/usb0/10100_20231005_083517.mp4
Track Importing MPEG-4 AVC - Width 1920 Height 1080 FPS 15000/1000 SAR 0/0
AVC|H264 Import results: 142 samples (150 NALUs) - Slices: 5 I 139 P 0 B - 0 SEI - 3 IDR
0.500 secs Interleaving
FILE CONVERSION THREAD: CONVERSION TIME (IN SECONDS): 3.717623
Deleting /var/tmp/10100_20231005_083517.h264 from RAM location
Removed 10100_20231005_083517 from list
Clips left to convert: 0
********************* Conversion Complete Tone
FILE CONVERSION THREAD: NO additional video files are waiting to move to USB

davidplowman, 2023-10-05:
I wonder if we're getting bitten by the I-frame period. When it saves the file it has to skip forward to the next I frame and by default these are spaced quite widely.  When you create the encoder, use `iperiod=15`, this should limit the variability here to no more than 15 frames (1 second).

Secondly, can you find an independent way of checking the clip length? I don't see how 142 frames can be converted at 15fps and Windows Media Player thinks the result is 15 seconds long. I expect `ffprobe` might tell you.

lmitcham, 2023-10-05:
ffprobe -v error -select_streams v:0 -show_entries stream=width,height,duration,bit_rate -of default=noprint_wrappers=1 /media/usb0/10100_20231005_083517.mp4

width=1920
height=1080
duration=9.466667
bit_rate=4990889

davidplowman, 2023-10-05:
Is that the 142 frame stream?

lmitcham, 2023-10-05:
Yes, it is the 142 stream

---------------------------------------------------------

I changed the encoder line to be 
       encoder = H264Encoder(bitrate=10000000, iperiod=15)

Recorded another video and it still produced a 14 second MP4 as played on windows player and here is the ffmpeg result ...
      frame=   98 fps=0.0 q=-1.0 Lsize=N/A time=00:00:03.23 bitrate=N/A speed=2.1e+03x
video:4017kB

and the ffprobe ...
      ffprobe -v error -select_streams v:0 -show_entries stream=width,height,duration,bit_rate -of default=noprint_wrappers=1 /media/usb0/10100_20231005_092729.mp4

width=1920
height=1080
duration=9.200000
bit_rate=5046892

davidplowman, 2023-10-05:
Ah, another thing I notice is that you're using the full resolution raw mode. This is limited to 10fps. You probably want to set the raw size to 2028x1080 as well? I tried your code with the revised raw size and iperiod, and seem to get just short of 150 frames every time.

I don't understand why Windows Media Player thinks the clips are longer than they are. The joke is that it seems to interpret them as being at 10fps, which was actually true! Yet I can't see how on earth it would know that. I'm inclined to believe what ffprobe says.

lmitcham, 2023-10-05:
That seems to be the culprit.  Changing the raw size is now giving me a 10 second video at 15 fps.  I tested this on multiple capture sequences.

Thank you so much for your patience and your help.