raspberrypi/picamera2 #816: python script containing picamera2 when added in crontab not working

sravich1, 2023-10-05:
python script containing picamera2 when added in crontab not working

sudo crontab -e
@reboot /bin/sleep 60;sudo /usr/bin/python3 /home/pi/Work/test_camera.py &

test_camera.py

 1 from picamera2 import Picamera2, Preview
 2 import time
 3 picam2 = Picamera2()
 4 camera_config = picam2.create_preview_configuration()
 5 picam2.configure(camera_config)
 6 picam2.start_preview(Preview.QTGL)
 7 picam2.start()
 8 time.sleep(2)
 9 picam2.capture_file("test.jpg")

if I run the python script individually this is working
but when run using crontab the python script does not get executed

sravich1, 2023-10-05:
Hi,
I even tried giving SUDO ..since I saw similar question thought of putting this in the forum.

cgobat, 2023-10-11:
Running `sudo python ...` will execute Python as the root user, so if any packages are installed specifically in your (non-root) user space, it may not be able to find them. You can do `sudo -u <username> python ...` to run it with elevation but as "yourself".

Also note that by doing `sudo crontab -e` you are opening the root user's crontab already; there is no need to specify `sudo` in the command to run it with elevation, as everything there will be executed by the root user anyway. However, this also means that the issue I described above will apply regardless of whether `sudo` is present in your Python invocation.

As David mentioned, including some error output would be helpful in diagnosing the issue.

sravich1, 2023-10-12:
Hi,
I got the error solved ..since I am using 2Gb RAM and 8GB EmmC , there was problem in the memory and corrpution of files.once I fixed this, as you mentioned I tried crontab -e and everything worked fine.
Thanks for the response.

ski-pow, 2024-06-13:
I'm having the same problem exactly (same code)!  Any help would be appreciated! 

8BG ram on bookworm.

Thanks!

sravich1, 2024-06-14:
The problem I had was to do with memory issue where the compute module
memory was 2GB RAM and 8GB MMC...once I did df -h to find the memory usage
it was already more than 90%, so the
code to autostart was failing.
I changed to 8GB 32 GB Emmc and everything worked fine for me.
Thanks