raspberrypi/picamera2 #818: [QUESTION] ExposureTime vs. FrameDuration?

cgobat, 2023-10-09:
What is the difference between the metadata attributes `"ExposureTime"` and `"FrameDuration"`? The documentation PDF describes FrameDuration as

> The amount of time (in microseconds) since the previous camera frame. 

and ExposureTime as

>Exposure time for the sensor to use, measured in microseconds.

But the practical difference in what these two things mean (specifically, in the context of a still image capture) is not clear to me.

Consider the following setup:
```python
from picamera2 import Picamera2()

picam2 = Picamera2()

config = picam2.create_still_configuration(raw={"size": picam2.sensor_resolution, "format": "SRGGB12"},
                                           controls={"ExposureTime": 1000000, "AnalogueGain": 1.0, "AwbEnable": False})
picam2.configure(config)

picam2.start()

capture = picam2.capture_request()
metadata = capture.get_metadata()
capture.release()

print(metadata)
```

which outputs
```
Metadata: {..., 'FrameDuration': 1000628, ..., 'ExposureTime': 1000000}
```

How should one interpret the fact that these two values differ (by 628 &mu;s)? Which one represents the actual integration time of the image associated with `capture`?

davidplowman, 2023-10-10:
The frame duration determines the framerate of the camera. It's quite common to set it to a fixed value so that you have a fixed framerate, but you don't have to - framerates can be variable too.

The exposure time is the amount of time for which pixels are exposed. This is bounded above by the frame duration. For example, in full resolution mode the imx477 can't do more than 10fps, so the frame duration can never be lower than 0.1s. However, it's perfectly fine to set the exposure time much lower than this - less than a millisecond if you want.

In practice there are a few extra rows of "blanking", or kind of "pretend pixels" at the bottom of the frame which the sensor needs to reset everything properly for the next frame. So usually, the exposure time can't quite reach the advertised frame duration.

davidplowman, 2023-10-10:
Yes, normally the frame duration will get pushed up if necessary for the exposure time you request (unless you have set actual limits that are lower).

cgobat, 2023-10-10:
Great, thank you!

closed 2023-10-10 by cgobat