raspberrypi/picamera2 #820: [OTHER] Reconciling `sensor_format` with `camera_properties["ColorFilterArrangement"]` value

cgobat, 2023-10-10:
The documentation (p. 21) indicates that the `Picamera2.sensor_format` attribute is indicative of the camera's Bayer matrix layout:

> For a raw stream, the format normally begins with an S, followed by four characters that indicate the Bayer order of the sensor

This information can also theoretically be obtained via the `Picamera2.camera_properties["ColorFilterArrangement"]`, which is described as (p. 72):

> A number representing the native Bayer order of sensor (before any rotation is taken into account).

With 0 being RGGB, 1 GRBG, 2 GBRG, 3 BGGR, and 4 meaning monochrome.

However, I am seeing some pieces of information that seemingly contradict one another. This code:

```python
from picamera2 import Picamera2
from pprint import pformat

picam = Picamera2()

print(f"\nCamera proprerties:\n{pformat(picam.camera_properties)}")
print(f"\nSensor format: {picam.sensor_format!r}")
```

Produces the output:
```

Camera proprerties:
{'ColorFilterArrangement': 0,
 'Location': 2,
 'Model': 'imx477',
 'PixelArrayActiveAreas': [(8, 16, 4056, 3040)],
 'PixelArraySize': (4056, 3040),
 'Rotation': 0,
 'ScalerCropMaximum': (0, 0, 0, 0),
 'SystemDevices': (20749, 20750, 20737, 20738, 20739),
 'UnitCellSize': (1550, 1550)}

Sensor format: 'SBGGR12_CSI2P'
```

The `"ColorFilterArrangement"` keyword's value being 0 indicates that the Bayer filter layout should be RGGB, yet the characters after the 'S' in the sensor format are 'BGGR'. Which is correct? Is this a bug, or expected behavior that I am just misunderstanding?

davidplowman, 2023-10-11:
Hi, the problem has its roots in the fact that all the Raspberry Pi sensors are actually mounted upside down, and correcting the orientation of the images actually rotates the Bayer pattern.

Unfortunately libcamera has been through some different iterations in its implementation of all this, and actually I think it will change (be corrected, in my view) in the next update. The sensor_format property should change (back) to "SRGGB12_CSI2P" and match the colour filter arrangement.

But note that this _doesn't_ describe the Bayer order of any raw stream that you request. For that you should always look at `picam2.camera_configuration()` after configuring the sensor. The format of the `raw` stream in there has always been and will always be correct. If you don't explicit demand a transform in your configuration, the sensor mounting rotation will be transparently corrected, and the Bayer order will actually be BGGR.

Hope that makes sense. All this rotation/transform stuff is actually highly confusing, as you can tell from the trouble libcamera has had with it!

cgobat, 2023-10-11:
Okay, thanks! I think I follow. So the sensor's "native" Bayer matrix is RGGB, but the 180&deg; rotation of the sensor on the camera board makes it BGGR in practice? This would make the most "upper left" pixel of a captured image with no explicitly-specified transform a B&mdash;is that right?

closed 2023-10-11 by cgobat

davidplowman, 2023-10-12:
That's right. The top left pixel will be B. If you ask explicitly for a 180 degree rotation it will go back to R (undoing the transparent fixing of the mounting rotation).