raspberrypi/picamera2 #821: [DOCS] Clarification of `"SensorTimestamp"` reported in capture metadata

cgobat, 2023-10-10:
Page 71 of the documentation describes the "SensorTimestamp" keyword as

> The time this frame was produced by the sensor, measured in nanoseconds since the system booted.

However, this could mean lots of things: the time of request, the time at the start of the exposure, the exposure mid-time, the time at the end of the exposure, the time the camera board finished reading out the sensor, the time that the frame was received from the camera, etc.

The docs should be updated to clarify this metadata field and be more explicit about its meaning.

cgobat, 2023-10-11:
Do the interrupt signal and first pixel get sent (at least nominally) at the beginning or the end of the exposure time window?

davidplowman, 2023-10-12:
Pixels have to be sent at the end of the exposure window, they're obviously not "ready" till they've finished being exposed. So as I remarked earlier, the timestamp you get is later than the time at which it starts to expose the (first pixel of the) frame, by an amount that should be the exposure time. For rolling shutter sensors the exposure window moves down the frame at the same rate that pixels are read out.

cgobat, 2023-10-12:
Understood. Thank you for all your help!

closed 2023-10-12 by cgobat

cgobat, 2023-10-12:
Do you happen to know the approximate readout time of the sensor? How much time are we talking about between the first and last pixel?

davidplowman, 2023-10-13:
That will more or less correspond to the fastest framerate you can have in any particular camera mode. When you do `libcamera-hello --list-cameras` it will list out all the available sensor modes. The max framerate that it gives is obviously the time from the first pixel in one frame to the first pixel in the next, but it starts over at the top as soon as it can after finishing at the bottom, so apart from a few lines of minimal blanking (just the odd microsecond) that will be the answer.

Rolling shutter cameras seem a bit confusing until you realise that they're just like old DSLRs where the shutter has a "front" and "rear" curtain. Only the shutter is electronic and travels through the pixel array at a fixed rate. First the "front curtain" starts scanning through the pixels, turning each one on (the start of the exposure). Then at a later time (which is the exposure time) the "rear curtain" sets off, following the front curtain, again at exactly the same rate. The rear curtain reads each pixel value and outputs it.

Note that two types of scenario are quite possible. For a very short exposure, the rear curtain follows just a few pixel rows behind the front curtain - that seems obvious enough. But when the exposure is maxed out it looks as though the front curtain is following just behind the read curtain. That's because as soon as you've read the pixel value out, you need to start exposing it again, so the rear curtain is really an entire frame length behind the front curtain. This gives you slightly head-scratchy cases where the next frame is already being exposed (in part) before it's even finished exposing and reading out the previous one. Hope that makes sense!

cgobat, 2023-10-13:
Yes it does! Thanks for the thorough explanation.