raspberrypi/picamera2 #823: [BUG]

dechuansun, 2023-10-11:
Hi,

I found a strange issue. I'm using an ov9281 sensor and a pi3 board. When I run the app_full.py, I got the error below. However, all other capture and recording .py files worked. Everything worked well previously when I used an earlier version with pi4. Any ideas?
Traceback (most recent call last):
  File "/home/dechuan/Downloads/picamera2-main/apps/app_full.py", line 1191, in <module>
    img_tab = IMGTab()
              ^^^^^^^^
  File "/home/dechuan/Downloads/picamera2-main/apps/app_full.py", line 772, in __init__
    self.reset()
  File "/home/dechuan/Downloads/picamera2-main/apps/app_full.py", line 796, in reset
    self.saturation.setValue(picam2.camera_controls["Saturation"][2], emit=True)
                             ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^
KeyError: 'Saturation'

Many thanks,
Dechuan

davidplowman, 2023-10-11:
Can you just confirm:

* Are you running Raspberry Pi's version of libcamera and Picamera2? (installed via apt, not a third party version)
* Is this a mono sensor?

Thanks!

dechuansun, 2023-10-12:
Yes, I'm running Raspberry Pi's version of libcamera and Picamera2.
This is a mono sensor.
After I commented the error line, I saw another error saying "Control Saturation is not advertised by libcamera"

davidplowman, 2023-10-12:
The apps that we provide are only examples, and we test them against various official Raspberry Pi cameras, none of which are mono. You're very welcome to edit the application to work with a 3rd party mono sensor. If you can modify the app so that it works with _both_ colour and mono sensors, then we'd be very happy to receive a pull request to our repository!