raspberrypi/picamera2 #824: [BUG] HQ camera: metadata `"Lux"` is always 400.0

cgobat, 2023-10-11:
**Describe the bug**
The Lux value reported in my capture metadata is always 400.0, regardless of exposure time, image content, etc.

**To Reproduce**
Demo script:
```python
import numpy as np
from picamera2 import Picamera2
from libcamera import controls

picam2 = Picamera2()

picam2.configure(picam2.create_still_configuration(raw={"size": picam2.sensor_resolution,
                                                        "format": picam2.sensor_format},
                                                   controls={"AnalogueGain": 1.0, "ColourGains": (1., 1.), "AwbEnable": False,
                                                             "ExposureTime": 1_000_000, # = 1 s
                                                             "NoiseReductionMode": controls.draft.NoiseReductionModeEnum.Off}))
picam2.start()
request = picam2.capture_request()
metdata = request.get_metadata()
request.release()
picam2.stop()

print(f"Metadata:\n{metdata}")

picam2.close()
```

**Expected behaviour**
I would expect the output from the ambient light sensor's measured illuminance to vary based on scene and imaging conditions.

**Console Output, Screenshots**
(edited) output from demo script shown above:
```
Metadata:
{'SensorTimestamp': ..., 'ScalerCrop': (0, 0, 4056, 3040),
 'DigitalGain': 1.0, 'ColourGains': (1.0, 1.0), 'SensorBlackLevels': (4096, 4096, 4096, 4096),
 'AeLocked': False, 'SensorTemperature': 26.0, 'Lux': 400.0, 'FrameDuration': 1000628,
 'ColourCorrectionMatrix': (...),
 'AnalogueGain': 1.0, 'ColourTemperature': 5932, 'ExposureTime': 1000000}
```

Regardless of what I point the camera at or how long my exposure time is, the Lux value always comes back as 400.0.

**Hardware :**
Zero 1 W + HQ camera module

**Additional context**
I am using the 32-bit Bullseye version of Raspbian Lite with Linux kernel 6.1.55+. I ran `rpi-update` as suggested in #768 in order to restore `"SensorTemperature"` functionality, which worked. I don't recall if "Lux" was functioning before I ran that update. If that is a candidate cause for this problem, I can reflash, but if there's no reason that would have broken this, I'd prefer not to.

Additionally, here are my installed versions of relevant packages: (from `apt list --installed`)
```
python3-picamera2/stable,now 0.3.12-2 all [installed]
libcamera-apps-lite/stable,now 1.2.1-1 armhf [installed]
libcamera0/stable,now 0~git20230720+bde9b04f-1 armhf [installed,automatic]
python3-libcamera/stable,now 0~git20230720+bde9b04f-1 armhf [installed]
libv4l2rds0/oldstable,now 1.20.0-2 armhf [installed,automatic]
python3-v4l2/stable,now 0.3.2-1 all [installed,automatic]
```

cgobat, 2023-10-12:
> (there is no separate ambient light sensor)

Hmm okay, I thought it did based on &sect;6-9-11 of the IMX477 datasheet from Sony.

Regardless, I think your diagnosis is correct. The first frame is the only one with Lux of 400.0, then it takes on more reasonable values for subsequent iterations. So, problem solved! However, I do think it would be good to call out this quirk (of the first frame(s) being untrustworthy/bad) more explicitly in the documentation.

Thanks!

davidplowman, 2023-10-13:
No, unfortunately the Linux kernel drivers don't expose any sensor features like that.

As I think I've said, we do indeed have a really big job ahead of us updating all our guides and documentation once the Bookworm and in particular Pi 5 dust has settled. So much stuff has been improved, and of course there are some completely new features on the Pi 5 too.

closed 2023-10-13 by cgobat