raspberrypi/picamera2 #825: [HOW-TO]

i2rlabs, 2023-10-16:
I have a Raspberry camera with which I take a lot of photos. Although I have manually set all the parameters like the code below, the photos taken have different colors. Some photos are cold (blue) and some are warm (red). The camera is located in an isolated device where no outside light enters and the lighting inside the device is completely the same and unchanged. Note that even the metadata of the photos that are taken are completely the same. But when you see the picture, you can see the difference in color.

        self.cam.set_controls({"AeEnable": 0})
        self.cam.set_controls({"AwbEnable": 0})
        self.cam.set_controls({"ExposureTime": 120000})
        self.cam.set_controls({"ExposureValue": 1.0})
        self.cam.set_controls({"AnalogueGain": 1.0})
        self.cam.set_controls({"ColourGains": (1.88, 0.941)})

davidplowman, 2023-10-17:
Hi, can you try running the following:
```
libcamera-hello -t 0 --shutter 500000 --gain 1 --awbgains 1.8,0.9, --info-text "%exp %ag %dg %rg %bg"
```
and report back what happens. The window title bar will tell us the values of the exposure time, analogue gain, digital gain, red colour gain and blue colour gain. Do you get the values you asked for, or do they change?

Can you also say what camera you're using? Perhaps supply the console long when you run that command, it might be helpful.

i2rlabs, 2023-10-17:
we used Arducam 5MP OV5647 Motorized Focus Camera B0176

i run 5 number test and outPut is same: ( gain changed to 1.88,0.941)

pi@raspberrypi:~/Desktop/Bala/ORG_CODE $ sudo libcamera-still -o Test2.jpg --autofocus-mode manual --lens-position 5 --awbgains 1.88,0.941 --analoggain 1 --shutter 500000 --gain 1 , --info-text "%exp %ag %dg %rg %bg"
Preview window unavailable
[0:31:12.213384634] [1489]  INFO Camera camera_manager.cpp:297 libcamera v0.0.0+4367-ad9428b4
[0:31:12.311679485] [1491]  INFO RPI vc4.cpp:444 Registered camera /base/soc/i2c0mux/i2c@1/ov5647@36 to Unicam device /dev/media3 and ISP device /dev/media0
[0:31:12.311801517] [1491]  INFO RPI pipeline_base.cpp:1101 Using configuration file '/usr/share/libcamera/pipeline/rpi/vc4/rpi_apps.yaml'
[0:31:12.313287563] [1489]  INFO Camera camera.cpp:1033 configuring streams: (0) 1296x972-YUV420
[0:31:12.314003137] [1491]  INFO RPI vc4.cpp:572 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected sensor format: 1296x972-SGBRG10_1X10 - Selected unicam format: 1296x972-pGAA
33222.00 1.00 4.00 1.88 0.94
33222.00 1.00 1.06 1.88 0.94
33222.00 1.00 1.06 1.88 0.94
33222.00 1.00 1.06 1.88 0.94

and for all test loge not different and every time output is same (33222.00 1.00 4.00 1.88 0.94)
and i get metadata from python for two image :

{'SensorTimestamp': 381310930000, 'ScalerCrop': (0, 0, 2592, 1944), 'DigitalGain                                                                                                                     ': 1.0629823207855225, 'ColourGains': (1.8799999952316284, 0.9409999847412109),                                                                                                                      'SensorBlackLevels': (1024, 1024, 1024, 1024), 'AeLocked': True, 'FrameDuration'                                                                                                                     : 120098, 'ColourCorrectionMatrix': (2.2165400981903076, -0.5678570866584778, -0                                                                                                                     .6486829519271851, -0.4107871353626251, 1.9673433303833008, -0.5565561652183533,                                                                                                                      -0.02404981292784214, -0.8331319093704224, 1.8571916818618774), 'AnalogueGain':                                                                                                                      1.0, 'Lux': 24.05396842956543, 'FocusFoM': 1514, 'ColourTemperature': 8625, 'Ex                                                                                                                     posureTime': 119968}

{'SensorTimestamp': 422000859000, 'ScalerCrop': (0, 0, 2592, 1944), 'DigitalGain                                                                                                                     ': 1.0629823207855225, 'ColourGains': (1.8799999952316284, 0.9409999847412109),                                                                                                                      'SensorBlackLevels': (1024, 1024, 1024, 1024), 'AeLocked': True, 'FrameDuration'                                                                                                                     : 120098, 'ColourCorrectionMatrix': (2.2165400981903076, -0.5678570866584778, -0                                                                                                                     .6486829519271851, -0.4107871353626251, 1.9673433303833008, -0.5565561652183533,                                                                                                                      -0.02404981292784214, -0.8331319093704224, 1.8571916818618774), 'AnalogueGain':                                                                                                                      1.0, 'Lux': 24.088321685791016, 'FocusFoM': 1513, 'ColourTemperature': 8625, 'E                                                                                                                     xposureTime': 119968}

and metadata is same but image have different color(only very little different in lux, we even had wider lux variation like 23 and 24 but still image is same)

[image]
[image]

davidplowman, 2023-10-17:
If the metadata for the two images are the same then I think we can be pretty certain that it's using the same parameters. However, there are a couple of adaptive algorithms that _could_ make a difference even when all those other parameters are fixed. To disable them, find the ov5647.json tuning file and do the following:

* Find the line that says `"n_iter": 100,` and change it to `"n_iter": 0,` (this disables adaptive colour shading correction)
* Find the line that says `"ce_enable": 1,` and change it to `"ce_enable": 0,` (this disables adaptive contrast enhancement)

I don't believe there's anything else adaptive going on under the hood, but let us know if this helps.

i2rlabs, 2023-10-17:
we change this two parameter in json tuning file but still  same problem and with same metadata we have color different. we get multiple images and noted that some of the images have color different.we also dropping some initial  images.

i2rlabs, 2023-10-17:
this is my python Code: (Focuser.py only python code for set i2c motor focus camera):

import time
import threading
from picamera2 import Picamera2, Preview
import picamera2
from libcamera import controls
from Focuser import Focuser

class Camera(object):
    # frame_reader = None
    cam = None
    _value_lock = None
    config = None

    def __init__(self, gain,width=640, height=360):
        self.open_cameraRaw(gain,width, height)

    def getRawFrame(self, Name, SaveImage, a_wait: bool = True):
        r = self.cam.capture_request(self.config)
        if (SaveImage == 1):
            r.save("main", Name)
        Metadata = r.get_metadata()
        print(Metadata)
        r.release()
        return Metadata

    def open_cameraRaw(self,gain, width=640, height=360, framerate=30):
        self.cam = Picamera2()
        self.config = self.cam.create_still_configuration(main={"size": (width, height)}, raw={"format": "SRGGB10"},
                                                          buffer_count=1)
        self.cam.configure(self.config)
        ###### COntrol
        self.cam.set_controls({"AeEnable": 0})
        self.cam.set_controls({"AwbEnable": 0})
        self.cam.set_controls({"ExposureTime": 120000})
        self.cam.set_controls({"ExposureValue": 1.0})
        self.cam.set_controls({"AnalogueGain": 1.0})
        self.cam.set_controls({"ColourGains": gain})
        time.sleep(0.1)

    def start_preview(self, a_preview: bool = False):
        if a_preview == True:
            self.cam.start_preview(Preview.QTGL)
        else:
            self.cam.start_preview()
        self.cam.start()

    def stop_preview(self):
        self.cam.stop_preview()
        self.cam.stop()

    def close(self):
        self.cam.close()

def draw_menu(camera,i2c_bus,CameraNumber):
    focuser = Focuser(i2c_bus)
    auto_focus = None
    focuser.set(Focuser.OPT_FOCUS,600)
    time.sleep(0.5)
    for i in range(20):
        Meta = camera.getRawFrame('Test_'+str(CameraNumber)+'_'+str(i)+'.jpg', 1)
        time.sleep(0.01)
    return True

def GetCaptureImage(CameraNumber,Gain):
    camera = Camera(gain=Gain,width=2592, height=1944)
    camera.start_preview(False)
    Falge = draw_menu(camera, 10,CameraNumber)
    camera.stop_preview()
    camera.close()

if __name__ == "__main__":
    Gain=(1.8, 0.9)
    GetCaptureImage(0,Gain)
    time.sleep(10)
    GetCaptureImage(1,Gain)
    
    
    
    
    if any problem in set camera pls note to me...

i2rlabs, 2023-10-17:
this is my python code (Focuser.py is only set i2c for motor focus camera):

    import time
    import threading
    from picamera2 import Picamera2, Preview
    import picamera2
    from libcamera import controls
    from Focuser import Focuser
    
    class Camera(object):
        # frame_reader = None
        cam = None
        _value_lock = None
        config = None
    
        def __init__(self, gain,width=640, height=360):
            self.open_cameraRaw(gain,width, height)
    
        def getRawFrame(self, Name, SaveImage, a_wait: bool = True):
            r = self.cam.capture_request(self.config)
            if (SaveImage == 1):
                r.save("main", Name)
            Metadata = r.get_metadata()
            print(Metadata)
            r.release()
            return Metadata
    
    
        def open_cameraRaw(self,gain, width=640, height=360, framerate=30):
            self.cam = Picamera2()
            self.config = self.cam.create_still_configuration(main={"size": (width, height)}, raw={"format": "SRGGB10"},
                                                              buffer_count=1)
            self.cam.configure(self.config)
            ###### COntrol
            self.cam.set_controls({"AeEnable": 0})
            self.cam.set_controls({"AwbEnable": 0})
            self.cam.set_controls({"ExposureTime": 120000})
            self.cam.set_controls({"ExposureValue": 1.0})
            self.cam.set_controls({"AnalogueGain": 1.0})
            self.cam.set_controls({"ColourGains": gain})
            time.sleep(0.1)
    
        def start_preview(self, a_preview: bool = False):
            if a_preview == True:
                self.cam.start_preview(Preview.QTGL)
            else:
                self.cam.start_preview()
            self.cam.start()
    
        def stop_preview(self):
            self.cam.stop_preview()
            self.cam.stop()
    
        def close(self):
            self.cam.close()
    
    def draw_menu(camera,i2c_bus,CameraNumber):
        focuser = Focuser(i2c_bus)
        auto_focus = None
        focuser.set(Focuser.OPT_FOCUS,600)
        time.sleep(0.5)
        for i in range(20):
            Meta = camera.getRawFrame('Test_'+str(CameraNumber)+'_'+str(i)+'.jpg', 1)
            time.sleep(0.01)
        return True
    
    def GetCaptureImage(CameraNumber,Gain):
        camera = Camera(gain=Gain,width=2592, height=1944)
        camera.start_preview(False)
        Falge = draw_menu(camera, 10,CameraNumber)
        camera.stop_preview()
        camera.close()
    
    if __name__ == "__main__":
        Gain=(1.8, 0.9)
        GetCaptureImage(0,Gain)
        time.sleep(10)
        GetCaptureImage(1,Gain)

if have any problem pls inform to me

davidplowman, 2023-10-17:
Hi, I don't really know why things aren't behaving as you want. It's probably worth double checking that you've edited the correct json file - I frequently find myself editing these files and discovering that it's loading a different one.

Beyond that it might be worth capturing some DNG files where the metadata claims they are all the same. Then use a raw converter (dcraw would do) and see if the images really are the same.

i2rlabs, 2023-11-03:
Hi @davidplowman 

And according to the updated json that you say, we started testing. We first tested the tests in a high-light environment, just like the code. The ambient light was high and the variety of colors was high. If you look at wide_1.jpg and wide_2.jpg photos, you will notice that there is no change in the color of the two images. And everything is perfect. 

But the problem appears when the ambient light decreases and a color chart is placed in a low-light environment. Look at the two images Closed_1.jpg and Closed_2.jpg, both images have different ambient color. One of the images has a cooler color and the other has a warmer color. Note that the first few photos are still dropped. 
I think that there is still an algorithm that affects the amount of color and awb or color balancing is not turned off properly and maybe this can be solved by changing a parameter in json. Of course, this is a guess from our side and your more experience can help us. 

[image]
Wide_1

[image]
wide_2

[image]
Closed_1

[image]
Closed_2

davidplowman, 2023-11-06:
Hi, when you capture these images that look different, could you print out the metadata for each so that you can compare them? Maybe post the results here so that we can check what's happening. Thanks.

i2rlabs, 2023-11-06:
Hi,in this current page ,previous 6 post before you can see two different image with meta data. metadata is same in all image but color will be different

davidplowman, 2023-11-06:
Yes, thanks, the metadata does look the same. Hard to understand how things are different then. You're 100% sure you've made the edits that I suggested to the correct tuning file (very easy for that to go wrong)? After that I'm a bit at a loss. Can you capture raw files and compare that the camera really is giving you exactly the same thing? Does it make a difference if you don't close and re-open the camera? Does it make a difference if you sleep for a moment before each capture after starting the camera? (Cameras can sometimes take a frame or two to settle completely after starting.)

tvoverbeek, 2023-11-06:
Just a thought. 
Can it be that the other end (i.e. the color management of the monitor) is playing tricks?
Thinking about i.e. colord on Linuc.
On my Intel Macbook pro I cannot see any dufference between closed_1 and closed-2 from a few days ago.

i2rlabs, 2023-11-06:
First of all, I apologize for the long text:
Yes, I'm sure I made the changes in the right files because even in the load file
I have changed the path of the json file, and when I change the name of the file, the camera gives an error and does not find the file. And it means that the path and file name are correct. In addition to setting the two parameters that you mentioned, I have also tested some other parameters. For example, I even copied the awb section from the ov5647_noir.json file. In this file, the awb section is set with a fixed byias value. But even with this, the colors still change. The problem is why we don't have the problem of changing the light in an environment where there is a lot of light in the background, but when the environment goes to low light with high contrast, this problem occurs? In all cases, I always delete the first 20 photos completely and do the reviews from the 20th photo onwards. For the issue of opening and closing the camera, I have tested both modes. When the camera is open forever, the images taken will change from cold to warm. That is, for example, the camera starts taking cold photos at first (with the size of each photo, for example, 760 kilobytes), and the more photos are taken, the warmer the images become, and the size of the images increases. For example, after 20 photos, the size reaches 780 KB and so on.
Even when we open and close the camera again, the images go from cold to warm at first, but every time we open the camera again, for example, the amount of heat (or size) of the image starts from the previous time. Unless, for example, a photo is not taken for 20 minutes or the Raspberry is restarted, the images will start to warm up again.

We even have this problem in regular Raspberry cameras and it is not related to the focusing model. But the thing that really confused us is that in some cameras this problem is less and in some it is much more. For example, in one camera it never works like this, but in another camera, there are 3 different photos in every 10 photos (only The camera of the device changes and the device itself and its raspberry are completely fixed).
Note that the images are really different and their rgb levels are different from each other.

davidplowman, 2023-11-06:
Are you able to capture and compare some raw (DNG) files (when the JPEGs are different)? It sounds like we need some confirmation what the sensor is actually doing.

i2rlabs, 2023-11-07:
The color difference in the dng file is also quite clear. I uploaded the files

[DNG.zip](https://github.com/raspberrypi/picamera2/files/13277555/DNG.zip)

davidplowman, 2023-11-07:
Thanks for the update. That's very helpful as it shows it must be something the sensor is doing.

The only thing we touch while the sensor is running is the exposure time, the frame blanking (so that we can achieve the requested exposure time) and the analogue gain. I don't think there's anything else. I wonder if the sensor is doing something else behind our back? I'm a bit at a loss, however.

I'm wondering if there are some makes of sensor that behave like this, and some that don't?

balaktp, 2024-04-04:
Exactly, the problem was with the camera sensor 5MP OV5647 , and now when we use Sony Imx708 sensors 12Meg, this problem no longer exists. I close this section