raspberrypi/picamera2 #827: [HOW-TO] Disable on-board defective pixel correction?

cgobat, 2023-10-17:
**Describe what it is that you want to accomplish**
It seems the [legacy camera stack had a way to disable on-sensor defective pixel correction](https://forums.raspberrypi.com/viewtopic.php?f=43&t=277768). I am wondering if there is a way to do the same using the modern `picamera2`/`libcamera`-based stack.

**Describe alternatives you've considered**
I am aware of the `rpi.dpc` algorithm entry in the `libcamera` tuning JSON file, and that I can edit it to have `{"strength": 0}` in order to accomplish this. However, it is difficult and clunky to modify this adaptively/on-the-fly. A control exposed via the API (as there exists for AWB, sharpening, noise reduction, etc.) would be better.

davidplowman, 2023-10-18:
The Linux kernel driver will let you disable this via a kernel module parameter. For the HQ cam, for example,  do the following
```
sudo su # you will have to be root to change it
echo 0 > /sys/module/imx477/parameters/dpc_enable
```
You can check its value with (you don't have to be root for this)
```
cat /sys/module/imx477/parameters/dpc_enable
```
You can't change the value while the camera is running, though it probably gets actioned every time you change the camera configuration and start it. Maybe just do it first after booting.

As regards the DPC in the pipeline, libcamera doesn't have an API for this so there simply is no mechanism we could use to give you control. But it's a conversion we could have with the libcamera folks. For the moment any such discussions would be parked behind a very large volume of Pi 5 work, however. If this were super critical you could obviously edit the DPC algorithm yourself to give you a "back door" control, though this kind of thing is obviously not generally desirable.

cgobat, 2023-10-18:
Thanks for the helpful response as always, David. Does editing `/sys/module/imx477/parameters/dpc_enable` have the same effect as modifying `imx477.json -> algorithms -> rpi.dpc -> strength`? What happens if the two things don't agree?

cgobat, 2023-10-18:
Okay, got it. So in order to completely disable DPC, one must do both?

closed 2023-10-19 by cgobat

cgobat, 2023-10-21:
One more question on this actually: in the Raspberry Pi Camera Algorithm and Tuning guide, it says

> The DPC control algorithm writes a `DpcStatus` object (`controller/dpc_status.h`) into the image metadata

Is this `DpcStatus` metadata accessible at all using the `picamera2` package?

davidplowman, 2023-10-23:
I'm afraid `dpc.status` is only available the metadata that we use to communicate between our own control algorithms (which existed before libcamera). There's no interface for passing this metadata out through libcamera's public API so far as I know.

It's a bit confusing that there's Raspberry Pi metadata (which is where we put `dpc.status`) and libcamera metadata, and they're different things. We do transfer many things from our metadata to libcamera's where it becomes publicly accessible (AGC and AWB information among others), but not everything. Obviously it would be possible to transfer more stuff, but we'd have to get the libcamera team's agreement to create new metadata categories for them.