raspberrypi/picamera2 #830: [BUG] Black pixel border on right side of image when in 2028x1520 / 2028*1080 modes

simonbuehler, 2023-10-20:
**Describe the bug**
when setting above resolutions in the e.g. sample app a black border is added to the image, both video or still

**To Reproduce**
run sample app_complete, set resolution, be amazed

**Expected behaviour**
no black border

**Console Output, Screenshots**
border
[image]

no border
[image]

**Hardware :**
4b and hq cam

davidplowman, 2023-10-20:
Indeed, it's strange. Even this trivial example seems to do it:
```
from PyQt5.QtWidgets import QApplication
from picamera2 import Picamera2
from picamera2.previews.qt import QGlPicamera2

app = QApplication([])
picam2 = Picamera2()
config = picam2.create_preview_configuration({'size': (2028, 1520)})
picam2.configure(config)
qpicamera2 = QGlPicamera2(picam2, width=640, height=480)

picam2.start()
qpicamera2.show()
app.exec()
```
I found that in the file [q_ql_picamera2.py where it calls glViewport](https://github.com/raspberrypi/picamera2/blob/main/picamera2/previews/q_gl_picamera2.py#L354), if I replace `w` by `w + 1` then it seems to work. A bit baffling. If anyone can explain that to me then I would be delighted...

simonbuehler, 2023-10-20:
I originally stumbled over it in an app without any qt dependencies or preview if that helps in any way

davidplowman, 2023-10-20:
Oh you're right, it's a camera mode thing. So I guess the simplest test case is
```
libcamera-still -o test.jpg -r --width 2028 --height 1520
```
Yes, we'll need to investigate that - thanks for reporting it!

davidplowman, 2023-10-23:
There does seem to be a problem in the kernel driver. It'll be "a couple of weeks" or so before a fix will available via rpi-update.