raspberrypi/picamera2 #831: [BUG] jpg save to file broken.

fiepfiep, 2023-10-21:
sorry for bad formatting.

update in PIL seems to affect jpg save to file.

https://pillow.readthedocs.io/en/stable/releasenotes/10.1.0.html#api-changes

is not compatible with

https://github.com/raspberrypi/picamera2/blob/main/picamera2/request.py#L262

`pi@raspberrypi:~/ams_rpi_software/picamera2/examples $ python capture_jpeg.py [0:06:51.925627735] [1475] INFO Camera camera_manager.cpp:299 libcamera v0.0.4+22-923f5d70-dirty (2023-10-20T02:40:23+02:00) [0:06:51.929091642] [1479] WARN CameraSensorProperties camera_sensor_properties.cpp:293 No static properties available for 'mira016' [0:06:51.929141698] [1479] WARN CameraSensorProperties camera_sensor_properties.cpp:295 Please consider updating the camera sensor properties database [0:06:51.937898624] [1479] WARN RPI raspberrypi.cpp:1357 Mismatch between Unicam and CamHelper for embedded data usage! [0:06:51.938776050] [1479] INFO RPI raspberrypi.cpp:1476 Registered camera /base/soc/i2c0mux/i2c@1/mira016@36 to Unicam device /dev/media0 and ISP device /dev/media2 [0:06:51.942780494] [1475] INFO Camera camera.cpp:1028 configuring streams: (0) 800x600-XBGR8888 [0:06:51.943074124] [1479] INFO RPI raspberrypi.cpp:851 Sensor: /base/soc/i2c0mux/i2c@1/mira016@36 - Selected sensor format: 400x400-SGRBG12_1X12 - Selected unicam format: 400x400-pgCC RGBA Traceback (most recent call last): File "/home/pi/ams_rpi_software/picamera2/examples/capture_jpeg.py", line 20, in metadata = picam2.capture_file("test.jpg") File "/usr/local/lib/python3.9/dist-packages/picamera2/picamera2.py", line 1195, in capture_file return self.execute_or_dispatch(partial(self.capture_file, file_output, name, format=format), File "/usr/local/lib/python3.9/dist-packages/picamera2/picamera2.py", line 1182, in _execute_or_dispatch return job.get_result() if wait else job File "/usr/local/lib/python3.9/dist-packages/picamera2/job.py", line 63, in get_result return self._future.result() File "/usr/lib/python3.9/concurrent/futures/_base.py", line 433, in result return self.__get_result() File "/usr/lib/python3.9/concurrent/futures/_base.py", line 389, in __get_result raise self._exception File "/usr/local/lib/python3.9/dist-packages/picamera2/job.py", line 44, in execute done, result = self.functions[0](https://github.com/raspberrypi/picamera2/issues/new?assignees=&labels=&projects=&template=bug_report.md&title=%5BBUG%5D) File "/usr/local/lib/python3.9/dist-packages/picamera2/picamera2.py", line 1165, in capture_file request.save(name, file_output, format=format) File "/usr/local/lib/python3.9/dist-packages/picamera2/request.py", line 163, in save return self.picam2.helpers.save(self.make_image(name), self.get_metadata(), file_output, format) File "/usr/local/lib/python3.9/dist-packages/picamera2/request.py", line 253, in save img.mode = "RGBX" AttributeError: can't set attribute`

davidplowman, 2023-10-23:
I've not noticed this myself, can you say what OS you're using and how you got this incompatible version of PIL? Thanks.

fiepfiep, 2023-10-23:
> I've not noticed this myself, can you say what OS you're using and how you got this incompatible version of PIL? Thanks.

I'm using my own compiled version of Bullseye + self compiled libcamera + cloned picamera2 from git and installed with pip.

I got the incompatible version of PIL by simply installing it using pip install pillow.

a solution could be to specify the pillow version in setup.py

fiepfiep, 2023-10-23:
> Do you know what the fix would be with newer versions of PIL, I'm wondering if it would be better to support the new version too?

replace
`img.mode = "RGBX"`
with
`img = img.convert(mode="RGBX")`

davidplowman, 2023-10-23:
Thanks. It seems like that works in the older version too, so maybe that's the way to go. I worry a bit that it's now traversing and copying the memory buffer where it wasn't previously, but maybe that's "progress".

We're just in the throes of making some new releases to support Pi 5 (and some other improvements), but I'll try and put this in after that.

closed 2026-11-06 by phili-b