raspberrypi/picamera2 #832: [HOW-TO] install in a virtual environment with python 3. 8

Svid23, 2023-10-23:
I have a raspberry pi 4 model b, I am on rasbpian bullseye in the full 64-bit version, and it had several problems, so I installed a virtual environment with python version 3.8.10 because I need it for my project, and installed several libraries with pip without problems, but when I arrive with picmaera2 it always gives me problems with the pyqt5 module, it gives me an error in the metadata and I cannot install picamera2 and the truth is I need it, does anyone know about the subject and can help me? I'm in a virtual environment and the problem is that I cannot use apt-get because everything that is installed with that goes to another version of python, so I need to know what I can do

davidplowman, 2023-10-23:
The problem is that we don't have pip packages for (the Python bindings for) kms++ and libcamera. libcamera is particularly awkward because it's not ABI stable, meaning that with every system libcamera update (and it's in constant development) your Python bindings will simply break. These are not Raspberry Pi owned packages but we are planning to look further into these issues.

lurch, 2023-12-18:
See https://github.com/raspberrypi/picamera2/issues/503#issuecomment-1861034147