raspberrypi/picamera2 #834: [HOW-TO] install picamera2 in pyenv,

Svid23, 2023-10-23:
I have a project that requires python version 3.8.7, I had errors at the beginning, but in the end I was able to put pip install picamera2 and it was imported correctly but when I wanted to use it in a code, I got no module name of libcamera, I was searching but I can't find a way to solve this, most do it with apt-get, however I am in pyenv so lp that is done with apt goes to another version of python

Manuel-Angel-Es, 2023-11-02:
Hi @Svid23,

I don't know why you really need that specific version of Python to run your software.

Several years ago I started programming in Python with version 3.7, now we are in version 3.12 and, in the latest version of Raspberry Pi Os Bookworm in 3.11.

I've never had any problems with Python versions as long as I've run the software on a higher version.

Ultimately the question is, have you tried running your software in Python 3.11? And if so, have there been errors? And if so, what errors has it given?

Regards

Alitzlan, 2023-12-14:
I am facing the same issue. In my case, another dependency in my project, pycoral library, only supports < 3.10. I use pyenv to fallback to 3.9.18 and found that I can no longer import libcamera

lurch, 2023-12-18:
See if https://github.com/raspberrypi/picamera2/issues/503#issuecomment-1861034147 helps?

AlgorithmicAce, 2024-03-06:
Hi @Alitzlan 

Were you able to fix it by any chance, I am using Python 3.7.3 for the same reason and I can't make it work, OpenCV hasn't caught up with picamera2 as stated [here](https://forums.raspberrypi.com/viewtopic.php?t=346684#p2077154). But, if I want to use picamera2, it's not being supported in pyenv either. I have no idea how to fix it, but I have an idea for a workaround, not sure if it will work though. Rather than capturing the webcam using OpenCV or Picamera2, I want to take pictures using libcamera-hello, and then pass it to the processing function, every 0.1sec (10 fps) using automation in Python. Please let me know if you have fixed the problem.

Alitzlan, 2024-03-06:
> Hi @Alitzlan
>
> Were you able to fix it by any chance, I am using Python 3.7.3 for the same reason and I can't make it work, OpenCV hasn't caught up with picamera2 as stated [...]

@AlgorithmicAce I found that migrating all my stuff to Bullseye, which uses Python 3.9, makes life much easier.

AlgorithmicAce, 2024-03-06:
@Alitzlan I see, but how did you migrate it to Bullseye, isn't Bullseye legacy OS, I thought it doesn't support picamera2

Alitzlan, 2024-03-06:
> @Alitzlan I see, but how did you migrate it to Bullseye, isn't Bullseye legacy OS, I thought it doesn't support picamera2

@AlgorithmicAce IIRC Bullseye is the first major version that adopts the new camera stack libcamera. I am using picamera2 in my project. And since my project is in Python only, I was able to just copy over the source code and run on bare metal Bullseye OS. [This is the little project if you are interested](https://youtu.be/Ri_GSQ-WzEg?si=W2H4lN20utkGgsk3)

AlgorithmicAce, 2024-03-06:
> @AlgorithmicAce IIRC Bullseye is the first major version that adopts the new camera stack libcamera. I am using picamera2 in my project. And since my project is in Python only, I was able to just [...]

@Alitzlan that's cool, i tried installing the Bullseye on my Pi 4, it works fine, but I can't install it on pi 5 since pi 5 only supports bookworm, i have to use pi 5 because I gotta use 2 cameras, its for a robotic competition. I honestly have no idea how to do it.

jensjoachim, 2024-09-10:
Hi @AlgorithmicAce and @Alitzlan 

I have exact same problem as you @Alitzlan. 
Need to run Pycoral in a pyenv. But for this case I need it on RPI5 to have two cameras.
Did any of you manage to solve the issue with picamera2 in Bookworm OS?

AlgorithmicAce, 2024-09-10:
Hi @jensjoachim 
I was following the guide from [https://github.com/raspberrypi/picamera2/issues/964] (this) link and it worked for me, I think it will work for you too

jensjoachim, 2024-09-16:
@AlgorithmicAce, the guide didn't work for me. A nee version of libcamera is used in RPI5 which is in front of pylibcamera.
Anmyways, this worked for me: https://github.com/feranick/pycoral

AlgorithmicAce, 2024-09-17:
@jensjoachim Not sure why that guide didn't work for you. But I'm glad you found out an alternative. Good luck on your project!