raspberrypi/picamera2 #841: [HOW-TO] capture a full resolution still during video

marcofariasmx, 2023-11-02:
**Describe what it is that you want to accomplish**
Take a full resolution still picture during a video stream.
Based on this example so far: https://github.com/raspberrypi/picamera2/blob/main/examples/still_during_video.py

**Describe alternatives you've considered**
Saving a DNG file from the raw stream and then converting that DNG to a JPG. The results are not so great colorwise, cpu expensive and prone to sd memory wear. Could get success with the .save_dng() method and io.BytesIO() to save to memory while converting to JPG.

**Additional context**
Not interested in high FPS or anything similar, just a low quality video stream used for live time monitoring and 1 HQ full resolution picture every minute.

davidplowman, 2023-11-06:
Hi, thanks for the question. I haven't quite understood how the "still_during_video.py" example isn't the right answer. It's true that this isn't the full sensor resolution, but you are fundamentally restricted by the size being output by the sensor for the video. You _could_ run the camera at full resolution, and this would give you the full sensor resolution for your stills, but your video would be limited to 10fps (or whatever framerate the system supports for that resolution).

marcofariasmx, 2023-11-06:
Thank you for your reply @davidplowman 

The thing is I am not able to take a full res picture with the available encoders in a Raspberry Pi Zero 2 W.

I've tried with the following code:

`#!/usr/bin/python3

import time

from picamera2 import Picamera2
from picamera2.encoders import H264Encoder

# Encode a VGA stream, and capture a higher resolution still image half way through.

picam2 = Picamera2()
full_resolution = picam2.sensor_resolution
half_resolution = [dim // 2 for dim in full_resolution]

print("Sensor resolution: ")
print(full_resolution)

main_stream = {"size": full_resolution}
lores_stream = {"size": (640, 480)}
video_config = picam2.create_video_configuration(main_stream, lores_stream, encode="lores")
picam2.configure(video_config)

encoder = H264Encoder()

picam2.start_recording(encoder, 'test.h264')
time.sleep(5)

# It's better to capture the still in this thread, not in the one driving the camera.
request = picam2.capture_request()
request.save("main", "test.jpg")
request.release()
print("Still image captured!")

time.sleep(5)
picam2.stop_recording()`

And I get this error:
[image]

Then I replaced the H264Encoder for the JpegEncoder and I got this error:
[image]

Then I replaced the encoder for the MJPEGEncoder and got the same error as the last one:
[image]

Same for the rest of encoders.

Not really sure what other settings to change or which would be the appropiate encoder. I understand that H264 is not an option since it is limited to 1920 x 1280.

Any further advice or guidance would be highly appreciated.

marcofariasmx, 2023-11-07:
I've also tried changing the CMA settings to the following in the config.txt file 
'''
# Enable DRM VC4 V3D driver
dtoverlay=vc4-kms-v3d,cma-400
max_framebuffers=2
'''

but it seems that the max amount that the system would allow is half which is 256. I've also tried increasing and decreasing the max_framebuffers with no success

davidplowman, 2023-11-07:
Hi, thanks for posting some code. Am I right in thinking it's trying to use a video configuration with 12MP buffers? By default it tries to allocate 8 of these buffers, and they'll be 32-bit RGBA each. I don't think that's ever going to work.

I would suggest reducing the `buffer_count`. You'll need at least 2 to avoid frame drops all over the place, maybe 3. Generally higher `buffer_count` means fewer frame drops, but if you're running the sensor at 12MP then you won't get that high a framerate anyway.

Also set the main stream format to `'RGB888'` (or `'BGR888'`), this is a bit less bad than the 32-bit format. You only need the 32-bit version if you want a preview window.

Finally try `cma-384`.

marcofariasmx, 2023-11-08:
Thank you very much, with your suggestions it finally worked! I will proceed to close this issue.

In case anyone is wondering how or faces the same deal, here is the code that worked:

```
#!/usr/bin/python3

import time

from picamera2 import Picamera2
from picamera2.encoders import JpegEncoder, MJPEGEncoder, H264Encoder

# Encode a VGA stream, and capture a higher resolution still image half way through.

picam2 = Picamera2()
full_resolution = picam2.sensor_resolution
half_resolution = [dim // 2 for dim in full_resolution]

print("Sensor resolution: ")
print(full_resolution)

main_stream = {"size": full_resolution, "format": "RGB888"}
lores_stream = {"size": (640, 480)}
video_config = picam2.create_video_configuration(main_stream, lores_stream, encode="lores", buffer_count=2)
picam2.configure(video_config)

encoder = MJPEGEncoder()

picam2.start_recording(encoder, 'test.mjpeg')
time.sleep(5)

# It's better to capture the still in this thread, not in the one driving the camera.
request = picam2.capture_request()
request.save("main", "test.jpg")
request.release()
print("Still image captured!")

time.sleep(5)
picam2.stop_recording()
```

closed 2023-11-08 by marcofariasmx