raspberrypi/picamera2 #846: [HOW-TO] apply a config

Mat24amal, 2023-11-06:
Hi! I seem to have problems applying a custom configuration. Seeing examples it looks like the picamera2.configure() function should take a dictionary as an argument, but in my code it won't.

this is my code: 

```
import datetime
import time
from pathlib import Path

from picamera2 import Picamera2
from picamera2.encoders import H264Encoder

def Capture_1_min():
    date_and_time = datetime.datetime.now().strftime("%d-%m-%Y_%H:%M:%S")
    camera = Picamera2()
    config = camera.create_video_configuration(
        main={"size": (640, 400)},
        encode="main",
        buffer_count=12,
        controls={"FrameDurationLimits": (3952, 3952)},
        sensor="640x400-R10_CSI2P",
    )
    camera.configure("config")

    encoder = H264Encoder(bitrate=None, framerate=253, iperiod=253)
    output_file = Path.cwd() / Path("test_253fps_" + date_and_time + ".h264")

    camera.start_recording(encoder, str(output_file))
    time.sleep(60)
    camera.stop_recording()
```

Form looking at all the examples and the documentation I feel like this should work, but the video is taken with the default configuration and not with my configuration.

I now this is a dum question but I can't understand why it does not work.

davidplowman, 2023-11-06:
Hi, no problem, just use `camera.configure(config)` (without the quotes double-quotes that you had).

Mat24amal, 2023-11-06:
Thanks for the quick response.

Im sorry I didin't see that i left the quotes. Those where there as an attempt to fix the issue. Even if they are not there the configuration still doesn't get applied to the camera.

davidplowman, 2023-11-06:
Ah, I see. You should also remove the `sensor=...` parameter. It looks like you might be using the latest version of Picamera2 from GitHub? I would revert back to the standard installation packages from apt if you can. Actually we've just (today) updated those, but you'll need to do a full system update which includes a new kernel (`sudo apt update` and `sudo apt upgrade`).

Maybe run just this:
```
camera = Picamera2()
config = camera.create_video_configuration(
    main={"size": (640, 400)},
    encode="main",
    buffer_count=12,
    controls={"FrameDurationLimits": (3952, 3952)}
    )
camera.configure(config)
print(camera.camera_configuration())
```
What does the camera configuration at the end report? Also, I don't recognise the camera, what does `libcamera-hello --list-cameras` say?

cpixip, 2023-11-07:
maybe correlated to this [issue](https://github.com/raspberrypi/picamera2/issues/839#issuecomment-1795691747)...

Mat24amal, 2023-11-08:
this is what the output of the code you provided:
```
[22:24:17.463449000] [5007]  INFO Camera camera_manager.cpp:284 libcamera v0.1.0+99-4a23664b
[22:24:17.497543898] [5010]  WARN CameraSensorProperties camera_sensor_properties.cpp:261 No static properties available for 'ov9281'
[22:24:17.497616434] [5010]  WARN CameraSensorProperties camera_sensor_properties.cpp:263 Please consider updating the camera sensor properties database
[22:24:17.505442360] [5010]  WARN V4L2 v4l2_subdevice.cpp:505 'ov9281 10-0060': Unknown subdev format 0x0000, defaulting to RGB encoding
[22:24:17.505525340] [5010] ERROR V4L2 v4l2_subdevice.cpp:244 No information available for format '1280x800-0x0000'
[22:24:17.506194149] [5010]  WARN RPiSdn sdn.cpp:39 Using legacy SDN tuning - please consider moving SDN inside rpi.denoise
[22:24:17.507613821] [5010]  INFO RPI vc4.cpp:444 Registered camera /base/soc/i2c0mux/i2c@1/ov9281@60 to Unicam device /dev/media3 and ISP device /dev/media1
[22:24:17.507741134] [5010]  INFO RPI pipeline_base.cpp:1142 Using configuration file '/usr/share/libcamera/pipeline/rpi/vc4/rpi_apps.yaml'
[22:24:17.510574774] [5007]  INFO Camera camera_manager.cpp:284 libcamera v0.1.0+99-4a23664b
[22:24:17.541868828] [5012]  WARN CameraSensorProperties camera_sensor_properties.cpp:261 No static properties available for 'ov9281'
[22:24:17.541924957] [5012]  WARN CameraSensorProperties camera_sensor_properties.cpp:263 Please consider updating the camera sensor properties database
[22:24:17.548604968] [5012]  WARN V4L2 v4l2_subdevice.cpp:505 'ov9281 10-0060': Unknown subdev format 0x0000, defaulting to RGB encoding
[22:24:17.548679134] [5012] ERROR V4L2 v4l2_subdevice.cpp:244 No information available for format '1280x800-0x0000'
[22:24:17.549262888] [5012]  WARN RPiSdn sdn.cpp:39 Using legacy SDN tuning - please consider moving SDN inside rpi.denoise
[22:24:17.550598301] [5012]  INFO RPI vc4.cpp:444 Registered camera /base/soc/i2c0mux/i2c@1/ov9281@60 to Unicam device /dev/media3 and ISP device /dev/media1
[22:24:17.550673967] [5012]  INFO RPI pipeline_base.cpp:1142 Using configuration file '/usr/share/libcamera/pipeline/rpi/vc4/rpi_apps.yaml'
[22:24:17.557493587] [5007]  INFO Camera camera.cpp:1181 configuring streams: (0) 640x400-XBGR8888 (1) 640x400-R10_CSI2P
[22:24:17.558658298] [5012]  INFO RPI vc4.cpp:608 Sensor: /base/soc/i2c0mux/i2c@1/ov9281@60 - Selected sensor format: 640x400-Y10_1X10 - Selected unicam format: 640x400-Y10P
{'use_case': 'video', 'transform': <libcamera.Transform 'identity'>, 'colour_space': <libcamera.ColorSpace 'SMPTE170M'>, 'buffer_count': 12, 'queue': True, 'main': {'format': 'XBGR8888', 'size': (640, 400), 'stride': 2560, 'framesize': 1024000}, 'lores': None, 'raw': {'format': 'R10_CSI2P', 'size': (640, 400), 'stride': 800, 'framesize': 320000}, 'controls': {'NoiseReductionMode': <NoiseReductionModeEnum.Fast: 1>, 'FrameDurationLimits': (3952, 3952)}, 'sensor': {'bit_depth': 10, 'output_size': (640, 400)}, 'display': 'main', 'encode': 'main'}
[22:24:18.687722697] [5012]  WARN V4L2 v4l2_videodevice.cpp:2007 /dev/video0[15:cap]: Dequeue timer of 1000000.00us has expired!
[22:24:18.687802919] [5012] ERROR RPI pipeline_base.cpp:1374 Camera frontend has timed out!
[22:24:18.687824622] [5012] ERROR RPI pipeline_base.cpp:1375 Please check that your camera sensor connector is attached securely.
[22:24:18.687845807] [5012] ERROR RPI pipeline_base.cpp:1376 Alternatively, try another cable and/or sensor.
```
and this is what `libcamera-hello  --list-cameras` says:
```
Available cameras
-----------------
0 : ov9281 [1280x800 10-bit MONO] (/base/soc/i2c0mux/i2c@1/ov9281@60)
    Modes: 'R8' : 640x400 [325.52 fps - (0, 0)/1280x800 crop]
                  1280x720 [180.51 fps - (0, 40)/1280x720 crop]
                  1280x800 [150.97 fps - (0, 0)/1280x800 crop]
           'R10_CSI2P' : 640x400 [260.42 fps - (0, 0)/1280x800 crop]
                         1280x720 [144.40 fps - (0, 40)/1280x720 crop]
                         1280x800 [120.76 fps - (0, 0)/1280x800 crop]
```
Thanks again for your quick response

davidplowman, 2023-11-09:
Hi, I was actually just trying to ascertain whether the camera is being configured correctly. Could you run the code snippet I sent, including where it prints out the camera configuration, but don't start the camera. Let's check that first, which I think was the original bug report. Once that's correct we can start the camera and see what happens. Thanks.

Mat24amal, 2023-11-13:
Hi, sorry for the late response, here is the output:
```
[118:50:08.949536768] [13728]  INFO Camera camera_manager.cpp:284 libcamera v0.1.0+99-4a23664b
[118:50:08.980861468] [13732]  WARN CameraSensorProperties camera_sensor_properties.cpp:261 No static properties available for 'ov9281'
[118:50:08.980962355] [13732]  WARN CameraSensorProperties camera_sensor_properties.cpp:263 Please consider updating the camera sensor properties database
[118:50:08.989162832] [13732]  WARN RPiSdn sdn.cpp:39 Using legacy SDN tuning - please consider moving SDN inside rpi.denoise
[118:50:08.990596393] [13732]  INFO RPI vc4.cpp:444 Registered camera /base/soc/i2c0mux/i2c@1/ov9281@60 to Unicam device /dev/media1 and ISP device /dev/media2
[118:50:08.990709558] [13732]  INFO RPI pipeline_base.cpp:1142 Using configuration file '/usr/share/libcamera/pipeline/rpi/vc4/rpi_apps.yaml'
[118:50:08.993431606] [13728]  INFO Camera camera_manager.cpp:284 libcamera v0.1.0+99-4a23664b
[118:50:09.023226728] [13734]  WARN CameraSensorProperties camera_sensor_properties.cpp:261 No static properties available for 'ov9281'
[118:50:09.023358986] [13734]  WARN CameraSensorProperties camera_sensor_properties.cpp:263 Please consider updating the camera sensor properties database
[118:50:09.030927951] [13734]  WARN RPiSdn sdn.cpp:39 Using legacy SDN tuning - please consider moving SDN inside rpi.denoise
[118:50:09.032354845] [13734]  INFO RPI vc4.cpp:444 Registered camera /base/soc/i2c0mux/i2c@1/ov9281@60 to Unicam device /dev/media1 and ISP device /dev/media2
[118:50:09.032470436] [13734]  INFO RPI pipeline_base.cpp:1142 Using configuration file '/usr/share/libcamera/pipeline/rpi/vc4/rpi_apps.yaml'
[118:50:09.040273769] [13728]  INFO Camera camera.cpp:1181 configuring streams: (0) 640x400-XBGR8888 (1) 640x400-R10_CSI2P
[118:50:09.041356666] [13734]  INFO RPI vc4.cpp:608 Sensor: /base/soc/i2c0mux/i2c@1/ov9281@60 - Selected sensor format: 640x400-Y10_1X10 - Selected unicam format: 640x400-Y10P
{'use_case': 'video', 'transform': <libcamera.Transform 'identity'>, 'colour_space': <libcamera.ColorSpace 'SMPTE170M'>, 'buffer_count': 12, 'queue': True, 'main': {'format': 'XBGR8888', 'size': (640, 400), 'stride': 2560, 'framesize': 1024000}, 'lores': None, 'raw': {'format': 'R10_CSI2P', 'size': (640, 400), 'stride': 800, 'framesize': 320000}, 'controls': {'NoiseReductionMode': <NoiseReductionModeEnum.Fast: 1>, 'FrameDurationLimits': (3952, 3952)}, 'sensor': {'bit_depth': 10, 'output_size': (640, 400)}, 'display': 'main', 'encode': 'main'}
```

davidplowman, 2023-11-13:
Thanks for posting that. It looks to me as though the camera has been configured as expected, does that look correct to you as well?

Mat24amal, 2023-11-15:
Yes, after playing around a bit, it now works as intended.
Thank you so much for your help