raspberrypi/picamera2 #852: [BUG] Cant use camera at all

dariotarantini, 2023-11-08:
**Please only report one bug per issue!**

**Describe the bug**
I cant use the camera from my rpi. I ve tested RPI3B+, a RPI ZERO 2 W and RPI ZERO. None works, with two different cameras. (V1.1 and HQ one). I ve also tested 3 different ribbon cables. 

**To Reproduce**
Steps to reproduce the behaviour. Wherever possible, please include a small self-contained Python script that shows the bug.

**Expected behaviour**
A clear and concise description of what you expected to happen.

**Console Output, Screenshots**
If applicable, any console output or screenshots that show the problem and associated error messages.
```
$ sudo libcamera-hello 
Preview window unavailable
[0:00:51.027749302] [856]  INFO Camera camera_manager.cpp:284 libcamera v0.1.0+99-4a23664b
[0:00:51.174047844] [860]  WARN RPiSdn sdn.cpp:39 Using legacy SDN tuning - please consider moving SDN inside rpi.denoise
[0:00:51.181043261] [860]  INFO RPI vc4.cpp:444 Registered camera /base/soc/i2c0mux/i2c@1/ov5647@36 to Unicam device /dev/media3 and ISP device /dev/media0
[0:00:51.181192323] [860]  INFO RPI pipeline_base.cpp:1142 Using configuration file '/usr/share/libcamera/pipeline/rpi/vc4/rpi_apps.yaml'
Mode selection for 1296:972:12:P
    SGBRG10_CSI2P,640x480/0 - Score: 3296
    SGBRG10_CSI2P,1296x972/0 - Score: 1000
    SGBRG10_CSI2P,1920x1080/0 - Score: 1349.67
    SGBRG10_CSI2P,2592x1944/0 - Score: 1567
Stream configuration adjusted
[0:00:51.185707323] [856]  INFO Camera camera.cpp:1181 configuring streams: (0) 1296x972-YUV420 (1) 1296x972-SGBRG10_CSI2P
[0:00:51.187436229] [860]  INFO RPI vc4.cpp:608 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected sensor format: 1296x972-SGBRG10_1X10 - Selected unicam format: 1296x972-pGAA
[0:00:52.322399510] [860]  WARN V4L2 v4l2_videodevice.cpp:2007 /dev/video0[13:cap]: Dequeue timer of 1000000.00us has expired!
[0:00:52.322563833] [860] ERROR RPI pipeline_base.cpp:1374 Camera frontend has timed out!
[0:00:52.322623104] [860] ERROR RPI pipeline_base.cpp:1375 Please check that your camera sensor connector is attached securely.
[0:00:52.322682791] [860] ERROR RPI pipeline_base.cpp:1376 Alternatively, try another cable and/or sensor.
ERROR: Device timeout detected, attempting a restart!!!
```

**Hardware :**
 Please state what kind of Pi you are using, and what camera is attached.
Running `Linux raspberrypi 6.1.0-rpi6-rpi-v8`
**Additional context**
Add any other context about the problem here.

davidplowman, 2023-11-08:
The first thing to do is always to download and flash the most recent image available, and to let it perform all its updates when you first boot it. Don't change or reconfigure _anything at all_ and "libcamera-hello" should just work. Can you confirm whether you've done exactly that?

The kinds of errors you have are often typical of long or faulty ribbon cables, power supplies that don't work as well as they claim (very common), or other hardware problems. Where you have spares so that you can rule things out by testing the spares, that's helpful.

If after that you're still having problems, you might do better to file an issue on the [libcamera-apps Github](https://github.com/raspberrypi/libcamera-apps).

dariotarantini, 2023-11-08:
> The first thing to do is always to download and flash the most recent image available, and to let it perform all its updates when you first boot it. Don't change or reconfigure anything at all and "libcamera-hello" should just work. Can you confirm whether you've done exactly that?

Yes, I did it. Now Im testing with a legacy image. I ve tested with a fresh os and different cables, as well different raspberrys

dariotarantini, 2023-11-09:
using legacy image fixed my issue.

closed 2023-11-09 by dariotarantini