raspberrypi/picamera2 #854: Dequeue timer expires / picamera2 crashes in dark environments

matbhe, 2023-11-08:
**Bug description**
It seems that streaming / recording really dark environments causes the following error (after some time):
`
[12:37:57.133376596] [5093]  WARN V4L2 v4l2_videodevice.cpp:2007 /dev/video0[15:cap]: Dequeue timer of 1000000.00us has expired!
`
After this error the camera is no longer working and the program must be restarted.
I've tried to configure the **ExposureTime** / **FrameDurationLimits** but nothing fixed it.

**Steps to Reproduce**
1. `wget https://raw.githubusercontent.com/raspberrypi/picamera2/main/examples/mjpeg_server.py` 
2. Place camera in _really dark room_ (almost no light)
3. Run `python3 mjpeg_server.py`
4. Wait some time (1-5 min) and then the error occurs

**Expected behaviour**
The "dequeue timer expired" - error should not occur even there is no light at all.

**Console Output**
```
[12:37:57.133376596] [5093]  WARN V4L2 v4l2_videodevice.cpp:2007 /dev/video0[15:cap]: Dequeue timer of 1000000.00us has expired!
[12:37:57.133547853] [5093] ERROR RPI pipeline_base.cpp:1374 Camera frontend has timed out!
[12:37:57.133612426] [5093] ERROR RPI pipeline_base.cpp:1375 Please check that your camera sensor connector is attached securely.
[12:37:57.133674351] [5093] ERROR RPI pipeline_base.cpp:1376 Alternatively, try another cable and/or sensor.
```

**Hardware / Software**
- Raspberry Pi Compute Module 4 Rev 1.1
- Cam ov5647 [2592x1944 10-bit GBRG] (/base/soc/i2c0mux/i2c@0/ov5647@36)
- Debian GNU/Linux 12 /  Kernel: Linux 6.1.0-rpi6-rpi-v8 / arm64
- python3-picamera2/stable,stable 0.3.14-1

**More informations**
I've tested the behavior multiple times:
- If the environment is "bright" enough then everything is working as expected.
- If the environment the error occurs always (after some time,  1 - 5 minutes)

davidplowman, 2023-11-09:
@naushir Any ideas on this one?

naushir, 2023-11-09:
The watchdog timer should get adjusted based on the framerate the sensor is currently running, so it should not trigger in dark environments where the exposure may be longer.  It is also a 1s timeout, and our frametime is 100ms, so there's plenty of margin.  Let me see if I can reproduce this.

matbhe, 2023-11-09:
I have another RaspberryPI4 (not CM4) with the same camera (not the same software).
I have now removed the camera from this RaspberryPI and connected it to my CM4.
Tests are in progress...

matbhe, 2023-11-09:
Result: The **problem persists** even with a different camera (same model) and a different cable.

naushir, 2023-11-10:
My board ran all night without failures.  This was on a Pi 4, not CM4 though.  Did your setup fail with the Pi4 board as well?
What kind of environment are you running the camera in?  Is there any chance of electrical interference causing this?

matbhe, 2023-11-10:
I would say it is a quite normal environment (basement in the house, wifi, nothing special).
I cannot test it at the moment on the Pi4 because I have to prepare an SD card first (with the same os/packages). I will test it.
However, I have noticed before that the camera pipeline also stalls on the PI4 every few days/weeks (much less than on the CM4). Unfortunately, the problem can then only be solved with a complete reboot. Either there is a problem with this camera type / batch or it is some kind of software problem....

matbhe, 2023-11-11:
I think there could be another cause of the problem:
The camera has two IR LEDs attached and these LEDs are automatically turned on when it gets dark (photo resistor).
Perhaps this is causing a problem with the power supply of the camera.
I'll disconnect the LEDs and see what happens...

matbhe, 2023-11-11:
I've spent some time and tried to analyze the behavior. The problem seems to be related to the IR LED's:

**IR LED's detached**
The camera is working fine and stable

**IR LED's attached to camera (as designed)**
In a really dark environment each IR-LED will provide the max output power of 3w (according to the spec).
In this situation the camera crashes after some time (as already describen in the issue).
Because of this behavior I've initially tought that the increased current is the culprit (voltage drop etc.).
But this seems not the case...

**IR LED's attached to 3.3v rail of PI**
Same situation as above (dark, 100% LED power) but the camera does not crash.
My current assumption is now that the heat generated by the LEDs is causing the problems.

What are your toughts?

naushir, 2023-11-13:
Ah, ok, to this is likely because of the power limit that can be drawn over the camera connector cable.  It would definitely be recommended to use an external power source, not the rails going up the camera corrector.

naushir, 2023-11-13:
I'll close this down now as I'm pretty sure we've identified the culprit.  Please reopen if you are still experiencing issues.

closed 2023-11-13 by naushir

MSaviorTR, 2023-11-16:
Hello everyone! I just encountred same issue too. I was using my Official Raspberry Pi Camera v2.1 (IMX219) on Rasp Pi 4 and it started to not respond. I have tried it with another Rasp Pi 4 and it couldn't solve the problem. I just wonder that is about hardware or something else like shortcircuit etc. I also changed the CSI cable. What could be the issue if it is about hardware? However, I don't think that is about hardware or components. Because, Rasp Pi recognize the sensor but cannot capture anything.

`from picamera2 import Picamera2

picam2 = Picamera2()
picam2.start()`

For this code, output is below.

`[5:55:58.812302496] [2455]  INFO Camera camera_manager.cpp:297 libcamera v0.0.5+82-2783c8d8
[5:55:58.847497400] [2456]  WARN RPI vc4.cpp:383 Mismatch between Unicam and CamHelper for embedded data usage!
[5:55:58.848479950] [2456]  INFO RPI vc4.cpp:437 Registered camera /base/soc/i2c0mux/i2c@1/imx219@10 to Unicam device /dev/media0 and ISP device /dev/media1
[5:55:58.848594838] [2456]  INFO RPI pipeline_base.cpp:1101 Using configuration file '/usr/share/libcamera/pipeline/rpi/vc4/rpi_apps.yaml'
[5:55:58.855309539] [2455]  INFO Camera camera.cpp:1033 configuring streams: (0) 640x480-XBGR8888 (1) 640x480-SBGGR10_CSI2P
[5:55:58.856078313] [2456]  INFO RPI vc4.cpp:565 Sensor: /base/soc/i2c0mux/i2c@1/imx219@10 - Selected sensor format: 640x480-SBGGR10_1X10 - Selected unicam format: 640x480-pBAA
`
However, when I try to capture an image with code that below:

`from picamera2 import Picamera2

picam2 = Picamera2()
picam2.start()
picam2.capture_file("img.png")`

It responds like that:

`[5:59:18.904755857] [2472]  INFO Camera camera_manager.cpp:297 libcamera v0.0.5+82-2783c8d8
[5:59:18.937802416] [2473]  WARN RPI vc4.cpp:383 Mismatch between Unicam and CamHelper for embedded data usage!
[5:59:18.938728133] [2473]  INFO RPI vc4.cpp:437 Registered camera /base/soc/i2c0mux/i2c@1/imx219@10 to Unicam device /dev/media0 and ISP device /dev/media1
[5:59:18.938813632] [2473]  INFO RPI pipeline_base.cpp:1101 Using configuration file '/usr/share/libcamera/pipeline/rpi/vc4/rpi_apps.yaml'
[5:59:18.948974941] [2472]  INFO Camera camera.cpp:1033 configuring streams: (0) 640x480-XBGR8888 (1) 640x480-SBGGR10_CSI2P
[5:59:18.949949917] [2473]  INFO RPI vc4.cpp:565 Sensor: /base/soc/i2c0mux/i2c@1/imx219@10 - Selected sensor format: 640x480-SBGGR10_1X10 - Selected unicam format: 640x480-pBAA
[5:59:20.036045749] [2473]  WARN V4L2 v4l2_videodevice.cpp:2007 /dev/video0[15:cap]: Dequeue timer of 1000000.00us has expired!
[5:59:20.036174081] [2473] ERROR RPI pipeline_base.cpp:1333 Camera frontend has timed out!
[5:59:20.036214877] [2473] ERROR RPI pipeline_base.cpp:1334 Please check that your camera sensor connector is attached securely.
[5:59:20.036254173] [2473] ERROR RPI pipeline_base.cpp:1335 Alternatively, try another cable and/or sensor.
^CTraceback (most recent call last):
  File "/home/mehdi/seda_rpi/single_n.py", line 5, in <module>
    picam2.capture_file("img.png")
  File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 1230, in capture_file
    return self.dispatch_functions(functions, wait, signal_function)
  File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 1203, in dispatch_functions
    return job.get_result() if wait else job
  File "/usr/lib/python3/dist-packages/picamera2/job.py", line 77, in get_result
    return self._future.result()
  File "/usr/lib/python3.9/concurrent/futures/_base.py", line 435, in result
    self._condition.wait(timeout)
  File "/usr/lib/python3.9/threading.py", line 312, in wait
    waiter.acquire()
KeyboardInterrupt
`
I hope you guys can suggest any solutions. Thank you.

davidplowman, 2023-11-16:
Is this also a problem that occurs only in dark environments, as per the issue title? Thanks.

davidplowman, 2023-11-16:
Yes please. Please also report:

* What kind of Pi you are using.
* What OS you are running. Please report the output of `cat /etc/os-release`, `uname -a`, `apt list python3-picamera2` and `libcamera-hello --version`.
* How you are logged in to the Pi. Are you connected remotely (e.g. via ssh) or directly with a screen and keyboard?
* Are you running a desktop environment?
* Is there anything you've changed since the camera last worked? Such as configuration settings, or other versions of camera software.
* Do `libcamera-hello` and `libcamera-still` work?
* How long your ribbon cables are.
* What kind of power supply are you using?
* You say you've tried a different Pi and a different cable, do you happen to have a different camera to try?
* If you have a spare SD card, are you able to try a clean OS installation where absolutely nothing is changed and all you do is start the camera?

Thanks!

MSaviorTR, 2023-11-16:
Thanks a lot!

oknoote, 2024-01-23:
> Thanks a lot!

Could you provide a link to the issue created as I'm experiencing exactly the same issues. Thank you!

MSaviorTR, 2024-01-24:
> > Thanks a lot!
> 
> Could you provide a link to the issue created as I'm experiencing exactly the same issues. Thank you!

[https://github.com/raspberrypi/picamera2/issues/873](Link)

oknoote, 2024-01-25:
> > > Thanks a lot!
> > 
> > 
> > Could you provide a link to the issue created as I'm experiencing exactly the same issues. Thank you!
> 
> [https://github.com/raspberrypi/picamera2/issues/873](Link)

Awesome, thanks.

Hugo-Huang, 2024-02-21:
> > > > Thanks a lot!
> > > 
> > > 
> > > Could you provide a link to the issue created as I'm experiencing exactly the same issues. Thank you!
> > 
> > 
> > [https://github.com/raspberrypi/picamera2/issues/873](Link)
> 
> Awesome, thanks.

Cant tell me how to fix this problem? The link is not exist now!

MSaviorTR, 2024-02-21:
> > > > > Thanks a lot!
> > > > 
> > > > 
> > > > Could you provide a link to the issue created as I'm experiencing exactly the same issues. Thank you!
> > > 
> > > 
> > > [https://github.com/raspberrypi/picamera2/issues/873](Link)
> > 
> > Awesome, thanks.
> 
> Cant tell me how to fix this problem? The link is not exist now!

I have changed the sensor on the board. Its issue was about the sensor or the tiny ribbon cable of it. I couldn't figure out clearly. Do not implement any procedure on the board that causes any other problems.