raspberrypi/picamera2 #857: [BUG] AttributeError: module 'libcamera' has no attribute 'SensorConfiguration'

Odoo-Tangerine, 2023-11-10:
**AttributeError: module 'libcamera' has no attribute 'SensorConfiguration'**

**My code: **
#!/usr/bin/python3
import cv2
from picamera2 import Picamera2
# Grab images as numpy arrays and leave everything else to OpenCV.
face_detector = cv2.CascadeClassifier("/usr/share/opencv4/haarcascades/haarcascade_frontalface_default.xml")
cv2.startWindowThread()
picam2 = Picamera2()
picam2.configure(picam2.create_preview_configuration(main={"format": 'XRGB8888', "size": (640, 480)}))
picam2.start()

while True:
    im = picam2.capture_array()
    cv2.imshow("Camera", im)
    cv2.waitKey(1)

**Console Output, Screenshots**
ITraceback (most recent call last):
  File "/home/longdn/face-and-license-plate-recognition-on-raspberry-pi-in/camera-recognition/test_camera.py", line 13, in <module>
    picam2.configure(picam2.create_preview_configuration(main={"format": 'XRGB8888', "size": (640, 480)}))
  File "/home/longdn/.local/lib/python3.9/site-packages/picamera2/picamera2.py", line 1102, in configure
    self.configure_(camera_config)
  File "/home/longdn/.local/lib/python3.9/site-packages/picamera2/picamera2.py", line 1034, in configure_
    libcamera_config = self._make_libcamera_config(camera_config)
  File "/home/longdn/.local/lib/python3.9/site-packages/picamera2/picamera2.py", line 915, in _make_libcamera_config
    libcamera_config.sensor_config = libcamera.SensorConfiguration()
AttributeError: module 'libcamera' has no attribute 'SensorConfiguration'

**Hardware : Raspberry Pi 4B**
PRETTY_NAME="Debian GNU/Linux 11 (bullseye)"
NAME="Debian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"

davidplowman, 2023-11-10:
Hi, the latest Picamera2 requires the latest version of libcamera which in turn requires an updated Linux kernel. Can you confirm whether you've updated everything? Please check the Pi camera forum [here](https://forums.raspberrypi.com/viewtopic.php?t=359080). (As always, we recommend doing a backup first or using a spare SD card.)

Odoo-Tangerine, 2023-11-10:
hi @davidplowman, Thanks for your answer. I have installed the latest package for the library. but everything is still not resolved.
sudo apt install python3-libcamera -y && sudo apt show python3-libcamera

Package: python3-libcamera
Version: 0~git20230720+bde9b04f-1
Priority: optional
Section: libs
Source: libcamera
Maintainer: Debian Multimedia Maintainers <debian-multimedia@lists.debian.org>
Installed-Size: 894 kB
Depends: libc6 (>= 2.17), libcamera0 (>= 0~git20230720+bde9b04f), libgcc-s1 (>= 3.0), libstdc++6 (>= 7), python3 (<< 3.10), python3 (>= 3.9~), python3:any
Homepage: https://libcamera.org/
Download-Size: 190 kB
APT-Sources: http://archive.raspberrypi.org/debian bullseye/main arm64 Packages
Description: complex camera support library (Python 3 bindings)
 libcamera is a complex camera support library which handles low-level
 control of the camera devices, providing a unified higher-level
 .
 This package contains the Python 3 bindings.

davidplowman, 2023-11-10:
Have you updated libcamera? (`sudo apt install libcamera0.1`)

Odoo-Tangerine, 2023-11-10:
Everything is the latest version. But I solved my problems. picamera2 encountered an error when installing with pip. I tried installing with sudo apt install python3-picamera2 and it worked fine. And saw your answer [](https://github.com/raspberrypi/picamera2/issues/300)

Thank you very much!

closed 2023-11-10 by Odoo-Tangerine

zoldaten, 2023-11-23:
i have the same issue on Bullseye Raspbian raspberry pi 4.
picamera2==3.14 (installed with pip) returns `AttributeError: module 'libcamera' has no attribute 'SensorConfiguration' `
so to fix that i have to:
```
pip3 uninstall picamera2
sudo apt install python3-picamera2
```
it installed picamera2==3.12

davidplowman, 2023-11-23:
Yes, libcamera does not have a stable binary API, and Bullseye is now "legacy" and therefore does not get updates (except for critical security fixes). This means that installing the latest version of Picamera2 from pip will give you versions that can't work with Bullseye libcamera , unless you build an updated libcamera for yourself. If not, I recommend always fetching updates with apt, as this guarantees you compatible versions.

quitmeyer, 2023-12-11:
> i have the same issue on Bullseye Raspbian raspberry pi 4. picamera2==0.3.14 (installed with pip) returns `AttributeError: module 'libcamera' has no attribute 'SensorConfiguration' ` so to fix that i have to:
> 
> ```
> pip3 uninstall picamera2
> sudo apt install python3-picamera2
> ```
> 
> it installed picamera2==0.3.12

this solved my prob