raspberrypi/picamera2 #860: [BUG] qt.qpa.xcb : execution of example yuv_to_rgb.py with bookworm

alain-lnpcrd, 2023-11-11:
`login@raspberrypi:~ $ python3
Python 3.11.2 (main, Mar 13 2023, 12:18:29) [GCC 12.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> #!/usr/bin/python3
>>> 
>>> import cv2
>>> 
>>> from picamera2 import Picamera2

>>> 
>>> cv2.startWindowThread()
0
>>> 
>>> picam2 = Picamera2()
[3:08:41.504097520] [3885]  INFO Camera camera_manager.cpp:284 libcamera v0.1.0+99-4a23664b
[3:08:41.550022502] [3889]  WARN RPiSdn sdn.cpp:39 Using legacy SDN tuning - please consider moving SDN inside rpi.denoise
[3:08:41.552841720] [3889]  INFO RPI vc4.cpp:444 Registered camera /base/soc/i2c0mux/i2c@1/ov5647@36 to Unicam device /dev/media3 and ISP device /dev/media1
[3:08:41.552929257] [3889]  INFO RPI pipeline_base.cpp:1142 Using configuration file '/usr/share/libcamera/pipeline/rpi/vc4/rpi_apps.yaml'
[3:08:41.556522413] [3885]  INFO Camera camera_manager.cpp:284 libcamera v0.1.0+99-4a23664b
[3:08:41.597647434] [3892]  WARN RPiSdn sdn.cpp:39 Using legacy SDN tuning - please consider moving SDN inside rpi.denoise
[3:08:41.599989971] [3892]  INFO RPI vc4.cpp:444 Registered camera /base/soc/i2c0mux/i2c@1/ov5647@36 to Unicam device /dev/media3 and ISP device /dev/media1
[3:08:41.600150543] [3892]  INFO RPI pipeline_base.cpp:1142 Using configuration file '/usr/share/libcamera/pipeline/rpi/vc4/rpi_apps.yaml'
>>> config = picam2.create_preview_configuration(lores={"size": (640, 480)})
>>> picam2.configure(config)
[3:08:41.609306673] [3885]  INFO Camera camera.cpp:1181 configuring streams: (0) 640x480-XBGR8888 (1) 640x480-YUV420 (2) 640x480-SGBRG10_CSI2P
[3:08:41.609869169] [3892]  INFO RPI vc4.cpp:608 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected sensor format: 640x480-SGBRG10_1X10 - Selected unicam format: 640x480-pGAA
>>> picam2.start()
>>> 
>>> while True:
...     yuv420 = picam2.capture_array("lores")
...     rgb = cv2.cvtColor(yuv420, cv2.COLOR_YUV420p2RGB)
...     cv2.imshow("Camera", rgb)
... 
qt.qpa.xcb: QXcbConnection: XCB error: 148 (Unknown), sequence: 192, resource id: 0, major code: 140 (Unknown), minor code: 20

`
A frame qt appears but without the video

RPi 4B 2GB
Linux raspberrypi 6.1.0-rpi6-rpi-v8 #1 SMP PREEMPT Debian 1:6.1.58-1+rpt2 (2023-10-27) aarch64 GNU/Linux
Debian GNU/Linux 12 (bookworm) Release:	12
Python 3.11.2
python3-picamera2/stable,stable,now 0.3.14-1
xcb/stable,now 2.4-8 arm64

davidplowman, 2023-11-13:
Hi, thanks for the report. I've tried running
```
import cv2

from picamera2 import Picamera2

cv2.startWindowThread()

picam2 = Picamera2()
config = picam2.create_preview_configuration(lores={'size': (640, 480)})
picam2.configure(config)
picam2.start()

while True:
    yuv420 = picam2.capture_array('lores')
    rgb = cv2.cvtColor(yuv420, cv2.COLOR_YUV420p2RGB)
    cv2.imshow('Camera', rgb)
    cv2.waitKey(1)
```
on the latest Raspberry Pi OS image on a Pi 4 and it seems to work as expected. Can you say if yours is a completely clean installation or whether anything has been changed? Are you running locally on your Pi (with a screen plugged in), or remotely? The error seems to suggest a problem with your Qt installation or version - can you say how you have installed PyQt5 and OpenCV?

It might be worth trying the same experiment I did - install a new image, let it do its updates, install OpenCV using `sudo apt install -y python3-opencv`, and then, without touching anything else at all, run the above script.

Thanks!

alain-lnpcrd, 2023-11-13:
Hi,
Thanks for your interest.
I encountered this problem by adapting the software of a toy robot equipped with a raspberrypi 4B 2Go to Raspberryi-OS Bookworm. An installation on an Sd card, the recovery of the robot software. the installation of the python3--xyz modules, if they exist, and for others, installations with pip install in the python virtual space.
Pour les packages Qt5 j'ai cherché à mettre ceux installés sous bullseye sous bookworm. Je vérifierai demain si toujt y est.
For the Qt5 packages I tried to put the ones installed under bullsey under bookworm. I'll check tomorrow if everything is there.

davidplowman, 2023-11-14:
Thanks for the update. The usual culprits are PyQt5 and OpenCV. If you can install both of them from apt (`sudo install -y python3-qt5 python3-opencv`) then they will work together. People often find that installing from pip often gives rise to incompatible versions, so I would certainly consider removing any pip versions that you have and going with apt, if that's possible for you.

alain-lnpcrd, 2023-11-14:
Hi

python3-qt5 et pyrhon3-opencv sont  installés et il n'y a pas de modules pip installés en concurence;

login@raspberrypi:~ $ apt list | grep python3* | grep installé

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

```
libpython3-dev/stable,now 3.11.2-1+b1 arm64  [installé, automatique]
libpython3-stdlib/stable,now 3.11.2-1+b1 arm64  [installé, automatique]
libpython3.11-dev/stable,now 3.11.2-6 arm64  [installé, automatique]
libpython3.11-minimal/stable,now 3.11.2-6 arm64  [installé, automatique]
libpython3.11-stdlib/stable,now 3.11.2-6 arm64  [installé, automatique]
libpython3.11/stable,now 3.11.2-6 arm64  [installé, automatique]
python-apt-common/stable,stable,now 2.6.0 all  [installé, automatique]
python-babel-localedata/stable,stable,now 2.10.3-1 all  [installé, automatique]
python-is-python3/stable,stable,now 3.11.1-3 all  [installé]
python3-apt/stable,now 2.6.0 arm64  [installé, automatique]
python3-asgiref/stable,stable,now 3.6.0-1 all  [installé, automatique]
python3-astroid/stable,stable,now 2.14.2-1 all  [installé, automatique]
python3-asttokens/stable,stable,now 2.2.1-1 all  [installé, automatique]
python3-av/stable,now 10.0.0-1+rpt2 arm64  [installé, automatique]
python3-babel/stable,stable,now 2.10.3-1 all  [installé, automatique]
python3-blinker/stable,stable,now 1.5-1 all  [installé, automatique]
python3-bs4/stable,stable,now 4.11.2-2 all  [installé, automatique]
python3-cairo/stable,now 1.20.1-5+b1 arm64  [installé, automatique]
python3-certifi/stable,stable,now 2022.9.24-1 all  [installé, automatique]
python3-cffi-backend/stable,now 1.15.1-5+b1 arm64  [installé, automatique]
python3-chardet/stable,stable,now 5.1.0+dfsg-2 all  [installé, automatique]
python3-charset-normalizer/stable,stable,now 3.0.1-2 all  [installé, automatique]
python3-click/stable,stable,now 8.1.3-2 all  [installé, automatique]
python3-colorama/stable,stable,now 0.4.6-2 all  [installé, automatique]
python3-colorzero/stable,stable,now 2.0-2 all  [installé, automatique]
python3-cryptography/stable,now 38.0.4-3 arm64  [installé, automatique]
python3-cups/stable,now 2.0.1-5+b4 arm64  [installé, automatique]
python3-cupshelpers/stable,stable,now 1.5.18-1 all  [installé, automatique]
python3-dbus/stable,now 1.3.2-4+b1 arm64  [installé, automatique]
python3-debconf/stable,stable,now 1.5.82 all  [installé, automatique]
python3-dev/stable,now 3.11.2-1+b1 arm64  [installé]
python3-dill/stable,stable,now 0.3.6-1 all  [installé, automatique]
python3-distlib/stable,stable,now 0.3.6-1 all  [installé, automatique]
python3-distro/stable,stable,now 1.8.0-1 all  [installé, automatique]
python3-distutils/stable,stable,now 3.11.2-3 all  [installé, automatique]
python3-docutils/stable,stable,now 0.19+dfsg-6 all  [installé, automatique]
python3-dotenv/stable,stable,now 0.21.0-1 all  [installé, automatique]
python3-filelock/stable,stable,now 3.9.0-1 all  [installé, automatique]
python3-flask-cors/stable,stable,now 3.0.10-2 all  [installé]
python3-flask/stable,stable,now 2.2.2-3 all  [installé]
python3-gi-cairo/stable,now 3.42.2-3+b1 arm64  [installé, automatique]
python3-gi/stable,now 3.42.2-3+b1 arm64  [installé, automatique]
python3-gpiozero/stable,stable,now 2.0-1 all  [installé]
python3-html5lib/stable,stable,now 1.1-3 all  [installé, automatique]
python3-ibus-1.0/stable,stable,now 1.5.27-5 all  [installé, automatique]
python3-idna/stable,stable,now 3.3-1 all  [installé, automatique]
python3-importlib-metadata/stable,stable,now 4.12.0-1 all  [installé, automatique]
python3-isort/stable,stable,now 5.6.4-1 all  [installé, automatique]
python3-itsdangerous/stable,stable,now 2.1.2-3 all  [installé, automatique]
python3-jedi/stable,stable,now 0.18.2-1 all  [installé, automatique]
python3-jinja2/stable,stable,now 3.1.2-1 all  [installé, automatique]
python3-jwt/stable,stable,now 2.6.0-1 all  [installé, automatique]
python3-kms++/stable,now 0~git20230930~162059+b91affd-1 arm64  [installé, automatique]
python3-lazy-object-proxy/stable,now 1.9.0-1+b1 arm64  [installé, automatique]
python3-lgpio/stable,now 0.2.2-1~rpt1 arm64  [installé, automatique]
python3-lib2to3/stable,stable,now 3.11.2-3 all  [installé, automatique]
python3-libcamera/stable,now 0.1.0+rpt20231020-1 arm64  [installé, automatique]
python3-libevdev/stable,stable,now 0.5-3 all  [installé, automatique]
python3-libgpiod/stable,now 1.6.3-1+b3 arm64  [installé]
python3-logilab-common/stable,stable,now 1.9.8-1 all  [installé, automatique]
python3-lxml/stable,now 4.9.2-1+b1 arm64  [installé, automatique]
[... 72 lines cut ...]
python3-wrapt/stable,now 1.14.1-2+b2 arm64  [installé, automatique]
python3-zipp/stable,stable,now 1.0.0-6 all  [installé, automatique]
python3-zmq/stable,now 24.0.1-4+b1 arm64  [installé]
python3.11-dev/stable,now 3.11.2-6 arm64  [installé, automatique]
python3.11-minimal/stable,now 3.11.2-6 arm64  [installé, automatique]
python3.11-venv/stable,now 3.11.2-6 arm64  [installé, automatique]
python3.11/stable,now 3.11.2-6 arm64  [installé, automatique]
python3/stable,now 3.11.2-1+b1 arm64  [installé, automatique]

```
```
login@raspberrypi:~ $ pip freeze -l
login@raspberrypi:~ $ 

(venv) login@raspberrypi:~/Adeept_AWR $ pip freeze -l
Adafruit-GPIO==1.0.3
Adafruit-PCA9685==1.0.1
Adafruit-PureIO==1.1.11
imutils==0.5.4
mpu6050-raspberrypi==1.2
pybase64==1.3.1
rpi-ws281x==5.0.0

```
L

closed 2023-11-14 by alain-lnpcrd