raspberrypi/picamera2 #863: [HOW-TO] Upgrading to 0.3.14 on Raspberry Pi OS lite 64-bit

keith-keycafe, 2023-11-14:
**Describe what it is that you want to accomplish**
I've been having an issue with video recording not stopping when using `FfmpegOutput` with audio, which I believe is fixed in the 0.3.14 release. However I'm struggling to upgrade to this version.
I've tried the usual `sudo apt update && sudo apt upgrade`, as well as `sudo apt full-upgrade` and `sudo apt install python3-picamera2`, but it doesn't find the new version:
```
$ sudo apt install python3-picamera2
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
python3-picamera2 is already the newest version (0.3.12-2).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
```

**Describe alternatives you've considered**
I was able to upgrade to 0.3.14 using pip, but then I was getting the following error:
```
Traceback (most recent call last):
  [...]
  File "/usr/local/lib/python3.9/dist-packages/picamera2/picamera2.py", line 1102, in configure
    self.configure_(camera_config)
  File "/usr/local/lib/python3.9/dist-packages/picamera2/picamera2.py", line 1034, in configure_
    libcamera_config = self._make_libcamera_config(camera_config)
  File "/usr/local/lib/python3.9/dist-packages/picamera2/picamera2.py", line 915, in _make_libcamera_config
    libcamera_config.sensor_config = libcamera.SensorConfiguration()
AttributeError: module 'libcamera' has no attribute 'SensorConfiguration'
```
I think the libcamera and picamera2 libraries were on conflicting versions.

I also tried `sudo rpi-update` to update the kernel, but it's still telling me 0.3.12-2 is the most recent version.

Re-imaging with a different OS isn't an option, so I'm hoping that isn't required.

**Additional context**
We're using 64-bit lite Raspberry Pi OS (Bullseye) on compute module 4
```
$ cat /etc/os-release 
PRETTY_NAME="Debian GNU/Linux 11 (bullseye)"
NAME="Debian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
```

Before `sudo rpi-update`:
```
$ uname -a
Linux raspberrypi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux
```

After `sudo rpi-update`:
```
$ uname -a
Linux raspberrypi 6.1.61-v8+ #1696 SMP PREEMPT Thu Nov  2 16:44:46 GMT 2023 aarch64 GNU/Linux
```

Python on version 3.9.2

Any help would be appreciated!

davidplowman, 2023-11-15:
I would avoid upgrading with pip, because you can get a version that's incompatible with other system libraries (libcamera). This is exactly why things are failing. If you use apt, everything gets upgraded together.

So I think you need to uninstall the pip version, and get the proper apt update process to work. What do the following output?
```
apt list python3-picamera2
sudo apt update
apt list --upgradable
```

davidplowman, 2023-11-15:
@XECDesign Hi, I don't suppose you could comment on what's wrong here? The new packages don't seem to be showing up. Thanks!

XECDesign, 2023-11-15:
Because bullseye is a legacy release and is only going to be getting critical security fixes.

davidplowman, 2023-11-15:
Ah, thanks for that. It sounds like upgrading to Bookworm is best, if that's possible. Alternatively, I suppose you could check out the 0.3.12 tag directly and cherry-pick [this fix](https://github.com/raspberrypi/picamera2/commit/6b2464c0dc8ec65fc8cd65e33663283b23703923). Would that help?

keith-keycafe, 2023-11-15:
Thanks for the assistance!

Unfortunately we have devices in remote locations so doing fresh installs with Bookworm isn't possible, and it seems that attempting an in-place upgrade from Bullseye is unsupported and very likely to cause problems.
I will try cherry-picking the FfmpegOutput fix.

keith-keycafe, 2023-11-16:
The following is how I installed 0.3.12 with the FfmpegOutput fix on Bullseye, in case anyone else is having the same issue:
```
sudo apt remove -y python3-picamera2
git clone --branch v0.3.12 -- https://github.com/raspberrypi/picamera2.git
git -C picamera2 cherry-pick 6b2464c
pip install -e picamera2
```
Thanks again for the help.

closed 2023-11-16 by keith-keycafe