raspberrypi/picamera2 #866: [BUG] Picamera2() can not start ... missing Attributes

Cassadin, 2023-11-16:
**Describe the bug**
A working script stopped working after an pi-reboot.
The Error appears in picamera2.py itself.
The Instance can not be created, because the attributes, it searches, dont exist.

**To Reproduce**
i cant reproduce it either ... 

**Expected behaviour**
To Start the Camera and the Webserver properly

**Console Output, Screenshots**
```python
 Traceback (most recent call last):
   File "/var/www/html/cam.py", line 134, in <module>
     picam2 = Picamera2()
              ^^^^^^^^^^^
   File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 242, in __init__
     camera_num = self.global_camera_info()[camera_num]['Num']
                  ~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^
 IndexError: list index out of range
 Exception ignored in: <function Picamera2.__del__ at 0x7fff2dd2ba60>
 Traceback (most recent call last):
   File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 404, in __del__
     self.close()
   File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 604, in close
     if self._preview:
        ^^^^^^^^^^^^^
 AttributeError: 'Picamera2' object has no attribute '_preview'
``` 

**Hardware :**
Rapsberry Pi 5 8GB
pi cam rev 1.3

**Additional context**
The script i use (modified), can be found here: [https://talutah.de/cam.py](https://talutah.de/cam.py)

Cassadin, 2023-11-16:
Error appeared due to a wrong wired camera

closed 2023-11-16 by Cassadin

yalcins, 2024-04-10:
I have same problem.
I use raspberry pi 4 and Pi camera 3 wide. After reboot didn't work. and 
I tried: 
#!/usr/bin/python3

# The QtPreview uses software rendering and thus makes more use of the
# CPU, but it does work with X forwarding, unlike the QtGlPreview.

import time

from picamera2 import Picamera2, Preview

picam2 = Picamera2()
picam2.start_preview(Preview.QT)

preview_config = picam2.create_preview_configuration()
picam2.configure(preview_config)

picam2.start()
time.sleep(5)

And error:
pi@raspberrypi:~/Documents/ToGo $ python3 picam2_Test.py
[0:23:33.615204968] [2909]  INFO Camera camera_manager.cpp:284 libcamera v0.2.0+46-075b54d5
Traceback (most recent call last):
  File "/home/pi/Documents/ToGo/picam2_Test.py", line 10, in <module>
    picam2 = Picamera2()
             ^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 242, in __init__
    camera_num = self.global_camera_info()[camera_num]['Num']
                 ~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^
IndexError: list index out of range
Exception ignored in: <function Picamera2.__del__ at 0xe7340348>
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 404, in __del__
    self.close()
  File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 604, in close
    if self._preview:
       ^^^^^^^^^^^^^
AttributeError: 'Picamera2' object has no attribute '_preview'