raspberrypi/picamera2 #871: [HOW-TO] force h264 encoding to include b frames?

maholli, 2023-11-17:
Using picamera2, I am manually sending an h264 stream in ~200 byte chunks and reconstructing / displaying the stream.

I have things nominally working, but since there is inherently packet loss as part of my application, it would work a lot better if the encoded h264 could include B-Frames. 

@davidplowman is it possible to specify b frames to the v4l2 encoder? Maybe by setting [V4L2_CID_MPEG_VIDEOENC_NUM_BFRAMES](https://docs.nvidia.com/jetson/l4t-multimedia/group__V4L2Enc.html#ga1f78b5427edc57977bedfe4d631c7d62)?

For demo purposes, below is the video and encoder configs I've been using to baseline:
```python
video_config = picam2.create_video_configuration(
    raw=None,
    lores=None,
    main={"size": (320,240), 'format': 'YUV420'},
    buffer_count=8,
    encode="main",
    )
picam2.configure(video_config)
encoder = H264Encoder(bitrate=62000,framerate=25.0,iperiod=5,enable_sps_framerate=True,repeat=True)
```
And to observe the frame types, I've been using `ffprobe -show_frames -i test.h264 | grep "pict_type"` on a picamera2 output file. 

Any insight would be much appreciated! Thanks :)

davidplowman, 2023-11-20:
I'm afraid the hardware encoder block on the Pi (Pi 4 and earlier models) simply does not support B frames, so it can't be done.

If you're encoding small frames, and are using maybe a Pi 4, you do encode in software using libav (ffmpeg) which would obviously make it possible (at the expense of CPU load).

maholli, 2023-11-20:
Makes sense. Thank you!