raspberrypi/picamera2 #875: Control AfWindows is not advertised by libcamera / How to set AfWindows

LeifSec, 2023-11-21:
How to set `AfWindows `?

```
from picamera2.picamera2 import Picamera2

camera = Picamera2()
camera.camera_controls["AfWindows"]
```
-> `KeyError: 'AfWindows' `

```
camera.set_controls({"AfWindows": [(0, 0, 0, 0, 0)]})
```
->
```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 1177, in set_controls
    self.controls.set_controls(controls)
  File "/usr/lib/python3/dist-packages/picamera2/controls.py", line 58, in set_controls
    self.__setattr__(k, v)
  File "/usr/lib/python3/dist-packages/picamera2/controls.py", line 33, in __setattr__
    raise RuntimeError(f"Control {name} is not advertised by libcamera")
RuntimeError: Control AfWindows is not advertised by libcam
```

libcamera v0.1.0+52-a858d20b
picamera2: 0.3.12
sensor: imx477

davidplowman, 2023-11-21:
There is no autofocus on our HQ cam (imx477), so the autofocus related controls don't appear. You would need to use the Camera Module 3 based on the imx708.

LeifSec, 2023-11-21:
Thanks a lot.  

I should have seen it [here](https://www.raspberrypi.com/documentation/accessories/camera.html#libcamera-and-libcamera-apps) .

I should have watched for `Focus` `Adjustable`/`Motorized`instead of autofocus there

closed 2023-11-21 by LeifSec