raspberrypi/picamera2 #877: [HOW-TO] Picamera2 + Camera Module 3 / HQ Camera + Raspberry Pi Zero

mmoollllee, 2023-11-22:
How can I use Picamer2 with the Camera Module 3 / HQ Camera on a Raspberry Pi Zero?

I'm trying to get full resolution jpeg but the process get's killed:
```
python3 test.py 
[0:52:31.771804662] [5929]  INFO Camera camera_manager.cpp:297 libcamera v0.0.5+83-bde9b04f
[0:52:31.978128955] [5944]  INFO RPI vc4.cpp:437 Registered camera /base/soc/i2c0mux/i2c@1/imx477@1a to Unicam device /dev/media3 and ISP device /dev/media0
[0:52:31.979631943] [5944]  INFO RPI pipeline_base.cpp:1101 Using configuration file '/usr/share/libcamera/pipeline/rpi/vc4/rpi_apps.yaml'
[0:52:32.052462340] [5929]  INFO Camera camera.cpp:1033 configuring streams: (0) 640x480-XBGR8888 (1) 2028x1520-SBGGR12_CSI2P
[0:52:32.055966311] [5944]  INFO RPI vc4.cpp:565 Sensor: /base/soc/i2c0mux/i2c@1/imx477@1a - Selected sensor format: 2028x1520-SBGGR12_1X12 - Selected unicam format: 2028x1520-pBCC
commit failed
commit failed
[0:52:34.817113486] [5948]  INFO Camera camera.cpp:1033 configuring streams: (0) 4056x3040-BGR888 (1) 4056x3040-SBGGR12_CSI2P
[0:52:34.836979322] [5944]  INFO RPI vc4.cpp:565 Sensor: /base/soc/i2c0mux/i2c@1/imx477@1a - Selected sensor format: 4056x3040-SBGGR12_1X12 - Selected unicam format: 4056x3040-pBCC
Killed
```

My test.py:
```
from picamera2 import Picamera2
picam2 = Picamera2()
picam2.start(show_preview=True)
picam2.start_and_capture_file("test.jpg")
picam2.stop_preview()
picam2.stop()
```

mmoollllee, 2023-11-22:
Thanks for the fast reply!!

> ```
> from picamera2 import Picamera2
> picam2 = Picamera2()
> picam2.start(show_preview=True)
> ```
> 
> Does this work and show you a preview window?
Yes!

>     * Are you using the latest OS? (and which one)
Yes!
```
cat /etc/os-release 
PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
[...]
```
>     * Is all your software up to date?
All updates installed.
>     * Are you using a GUI? Or are you logged in at the console? Or remotely?
Using GUI via VNC
>     * How much CMA do you have? (output of `grep Cma /proc/meminfo`)
```
grep Cma /proc/meminfo
CmaTotal:         262144 kB
CmaFree:           75704 kB
```

davidplowman, 2023-11-22:
OK, so you're on Bullseye (not the latest), but should work. Camera preview performance is usually terrible on a Pi Zero when using a GUI, but if it's working for you then try this:
````
import time
from picamera2 import Picamera2
picam2 = Picamera2()
picam2.start(show_preview=True)
capture_config = picam2.create_still_configuration()
time.sleep(2)
picam2.switch_mode_and_capture_file(capture_config, "test.jpg")
````
Presumably this will fail. I think the reason is probably that you don't have enough CMA memory. The first thing to try would be to increase it by following the instructions in section 8.3 of the [manual](https://datasheets.raspberrypi.com/camera/picamera2-manual.pdf).

The reason `libcamera-still` works is because it's a C++ application and can send YUV420 images, which are much smaller, straight to the JPEG encoder. Python mostly doesn't handle YUV420 very well, so you're stuck with 24-bit (or even 32-bit) RGB images. These images are quite challenging for a Pi Zero.

mmoollllee, 2023-11-22:
So Iincreased CMA memory. `/boot/config.txt` looks like this now:
```
# For more options and information see
# http://rpf.io/configtxt
# Some settings may impact device functionality. See link above for details

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
hdmi_group=2
hdmi_mode=82

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800

# Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on

# Uncomment this to enable infrared communication.
#dtoverlay=gpio-ir,gpio_pin=17
#dtoverlay=gpio-ir-tx,gpio_pin=18

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

# Automatically load overlays for detected cameras
camera_auto_detect=1

# Automatically load overlays for detected DSI displays
display_auto_detect=1

# Enable DRM VC4 V3D driver
dtoverlay=vc4-kms-v3d,cma-320
max_framebuffers=2

# Disable compensation for displays with overscan
disable_overscan=1

[cm4]
# Enable host mode on the 2711 built-in XHCI USB controller.
# This line should be removed if the legacy DWC2 controller is required
# (e.g. for USB device mode) or if USB support is not required.
otg_mode=1

[all]

[pi4]
# Run as fast as firmware / board allows
arm_boost=1
```

Running your code:
```
[0:05:23.679264154] [1953]  INFO Camera camera_manager.cpp:297 libcamera v0.0.5+83-bde9b04f
[0:05:24.119240588] [1967]  INFO RPI vc4.cpp:437 Registered camera /base/soc/i2c0mux/i2c@1/imx477@1a to Unicam device /dev/media2 and ISP device /dev/media0
[0:05:24.127956909] [1967]  INFO RPI pipeline_base.cpp:1101 Using configuration file '/usr/share/libcamera/pipeline/rpi/vc4/rpi_apps.yaml'
[0:05:24.306726488] [1953]  INFO Camera camera.cpp:1033 configuring streams: (0) 640x480-XBGR8888 (1) 2028x1520-SBGGR12_CSI2P
[0:05:24.313332657] [1967]  INFO RPI vc4.cpp:565 Sensor: /base/soc/i2c0mux/i2c@1/imx477@1a - Selected sensor format: 2028x1520-SBGGR12_1X12 - Selected unicam format: 2028x1520-pBCC
[0:05:27.419942797] [1975]  INFO Camera camera.cpp:1033 configuring streams: (0) 4056x3040-BGR888 (1) 4056x3040-SBGGR12_CSI2P
[0:05:27.469272480] [1967]  INFO RPI vc4.cpp:565 Sensor: /base/soc/i2c0mux/i2c@1/imx477@1a - Selected sensor format: 4056x3040-SBGGR12_1X12 - Selected unicam format: 4056x3040-pBCC
[0:05:45.126840965] [1975]  INFO Camera camera.cpp:1033 configuring streams: (0) 640x480-XBGR8888 (1) 2028x1520-SBGGR12_CSI2P
[0:05:45.145282086] [1967]  INFO RPI vc4.cpp:565 Sensor: /base/soc/i2c0mux/i2c@1/imx477@1a - Selected sensor format: 2028x1520-SBGGR12_1X12 - Selected unicam format: 2028x1520-pBCC
```

Works!!!! 

Thank you sooo much for your fast help!! 🥇

closed 2023-11-22 by mmoollllee