raspberrypi/picamera2 #884: [SUGGESTION]

siranshen, 2023-12-03:
**Is your feature request related to a problem? Please describe.**
If you call Picamera2#autofocus_cycle with Camera V3 in certain positions (facing down on a table or facing up to an empty ceiling), the autofocus process will take a really long time, about 25 seconds in my case.

Reproducible by the following code:
```
import time
from picamera2 import Picamera2

picam2 = Picamera2()
capture_config = picam2.create_still_configuration()
picam2.start(config=capture_config)
print('Started')

print('Starting autofocus')
start_time = time.time()
success = picam2.autofocus_cycle()
print(f'Autofocused: {success}, {time.time() - start_time}s')
print('Taking photo')
picam2.capture_file("test.jpg")
print('Photo taken')
```

**Describe the solution you'd like**
Be able to set a timeout for autofocus.

**Describe alternatives you've considered**
Check if the job has finished after a period of time and cancel the job by code if it is still not done. There currently seems no good way to check if job has finished or not, either.

njhollinghurst, 2023-12-04:
I'll investigate to check that it's actually taking that long and is not some kind of race condition (but then I'd expect it to take forever). But I suspect it's doing the AF sweep at full-frame size at 10fps, which will be slower than at 30fps.

siranshen, 2023-12-05:
Hey guys, thanks for taking a look.

25s is reproducible on my Zero 2 W. It's probably less on other models.

Also, I tried modifying the script to use preview first, and yes, it did slash the time down from ~25s to ~8.5s:
```
picam2 = Picamera2()
preview_config = picam2.create_preview_configuration()
capture_config = picam2.create_still_configuration()
picam2.start(config=preview_config)  # Starting with preview instead
print('Started')

print('Starting autofocus')
start_time = time.time()
success = picam2.autofocus_cycle()
print(f'Autofocused: {success}, {time.time() - start_time}s')
print('Taking photo')
picam2.switch_mode_and_capture_file(capture_config, "test.jpg")  # Switch to still and shoot
print('Photo taken')
```

davidplowman, 2023-12-05:
Ah, thanks for the info - maybe it's related to running on a Zero 2W. I wonder if we're experiencing more frame drops there. I'll have a look at that, though it'll have to wait just a bit as I don't have one available to me just at the moment.

davidplowman, 2023-12-05:
I've had a go on a Pi 3 which is very similar to a Pi Zero 2. Here's what I found:

* Original script: I managed to get this to take as long as 17.5s
* With buffer_count=2: down to 9.7s
* With buffer_count=3 at 30fps (use `create_still_configuration(controls={'FrameRate': 30}, buffer_count=3)`: 5.9s
* Using preview (no framerate given): 6.2s
* Using preview at 30fps (use `create_preview_configuration(controls={'FrameRate': 30})`): 2.6s

If you can use that preview mode at 30fps then I think you'll be OK. I am planning to add a timeout as well to all asynchronous camera commands because it's not difficult, but I'm hoping the above will work OK for you.

siranshen, 2023-12-05:
Looks great to me. Thanks a lot!