raspberrypi/picamera2 #885: [HOW-TO]  Improve frame sequential 'locking' between V2.1 cameras on Pi5

sandyol55, 2023-12-05:
I'm trying some experiments to create a frame sequential stereoscopic preview display from a Pi5 4GB with a pair of V2.1 cameras, being fed to a '3D' TV capable of accepting L/R frame sequential images at 60Hz.

I understand that there is no synching/locking between the two free running cameras and processes, but the example below gets very close to working - (better that I expected!)  - and wondered if there is any tweak to the code, or different approach, that might improve things.

At the moment the sequential frames seem to come out 'correctly' for about 10 seconds, then the left/right images flip over to right/left and continue with that pattern with a flip every 10 seconds or so.

```
#!/usr/bin/python3

from picamera2 import Picamera2, Preview
from picamera2 import MappedArray
from libcamera import Transform

leftcam = Picamera2(0)
rightcam = Picamera2(1)

leftcam.start_preview(Preview.QTGL, x=0, y=0, width=1920, height=1080)
leftcam.title_fields =["ExposureTime", "AnalogueGain", "FrameDuration"]

config_l = leftcam.create_video_configuration(raw={"size":(640,480)}, main={"size":(1280,960)},transform=Transform(hflip=True,vflip=True),controls = {"FrameDurationLimits":(16666,16666)})
config_r = rightcam.create_video_configuration(raw={"size":(640,480)}, main={"size":(1280,960)},transform=Transform(hflip=True,vflip=True),controls = {"FrameDurationLimits":(16666,16666)})

leftcam.configure(config_l)
rightcam.configure(config_r)

toggle = True

def sequentialviews(request):
    with MappedArray(request, "main") as leftview:
        if toggle == False:
            leftview.array[:] = rightview[:]
        
leftcam.start()
rightcam.start()
rightview=rightcam.capture_array()
leftcam.pre_callback = sequentialviews

try:
    while True:
        rightview = rightcam.capture_array()
        toggle = not toggle
except KeyboardInterrupt:
    pass

```

Appreciate any thoughts even if it's a 'not possible'
Thanks
Sandy

davidplowman, 2023-12-06:
That's an interesting problem. I would have hoped that if you've got identical cameras running with the same duration then they would stay completely synced. It might be worth investigating that first, may with something like this:
```
while True:
    left_timestamp = leftcam.capture_metadata()['SensorTimestamp']
    right_timestamp = rightcam.capture_metadata()['SensorTimestamp']
    print(right_timestamp - left_timestamp)
````
and see if that drifts over time or not. If there is drift then maybe that's the problem, otherwise perhaps it's the vagaries of when Python gets to execute bits of code that creates the variability? But it would certainly be good to get a bit more to the bottom of this.

As regards the code you have, an alternative strategy might be to flip the `toggle` inside `sequentialviews`? That would presumably stop the phase from flipping, though if things are getting out of sync you could get 2 duplicated right images in a row, or perhaps skip a right image. But it sounds like that might upset the TV less.

davidplowman, 2023-12-06:
The 'SensorTimestamp' value comes from the kernel so it should really be very reliable. From what you say, do I interpret that as about 25 - 8 = 17us drift in 600 frames. Did that happen gradually (i.e. cameras really running at different rates), or was there a sudden jump (i.e. a frame actually got dropped somewhere).

With the `toggle` outside `sequentialviews` I can certainly see that both gradual drift or a frame drop could cause the left/right phase to flip. But with the `toggle` inside I don't really see how that could be possible. How are these images finding their way to the display?

sandyol55, 2023-12-06:
This was the result of that initial trial, with a slightly noisy but gradually reducing difference with an isolated drop to ~8microseconds after ~550 frames.  As I said I'm not sure how relevant that was though !

[image]

and showing the single change to 8 microseconds.
[image]

Will have a closer look at the impact of moving the toggling into the `sequentialviews`, and in any difference between QTGL and DRM modes 

The display (an LG 47LW650T) is being driven from the Pi5 via a short micro HDMI to HDMI adaptor and a 2m HDMI lead.
It's switched to its framesequential '3D' mode. 

Thanks again for the interest, help and pointers.

davidplowman, 2023-12-06:
Indeed, the first graph seems to imply a slight but steady drift (still a bit disappointing for identical cameras), and the 2nd with the big jump is presumably a frame that got lost?

I assume you're sending the left image (possibly after overwriting with the right) to the TV? Presumably the TV is fixed to an incoming framerate, and if a left frame arrives slighty late then things can get flipped? Sending images to the DRM output also waits for a frame sync, and with the vagaries of when Python wakes up and does stuff, that all sounds quite hard to control if all your 60Hz clocks are marginally different!

It would be nice if one could do side-by-side kind of stereo instead as the timing issues would hopefully be less critical. But I can't really see a way that doesn't involve horrid amounts of pixel copying, and getting the various buffers to the optimal size is probably difficult as things stand.

sandyol55, 2023-12-06:
Success!
The code  below does the left/right merging into a single frame without all the timing issues of the frame sequential method.
Despite its problems I might still pursue a frame sequential method but for now this counts as result.
Once again , many thanks for the advice!

```
#!/usr/bin/python3

from picamera2 import Picamera2, Preview
from picamera2 import MappedArray
from libcamera import Transform

leftcam = Picamera2(0)
rightcam = Picamera2(1)

leftcam.start_preview(Preview.QTGL, x=0, y=0, width=1920, height=1080)
leftcam.title_fields =["ExposureTime", "AnalogueGain", "FrameDuration"]

config_l = leftcam.create_video_configuration(main={"size":(1920,1080)},transform=Transform(hflip=True,vflip=True),controls = {"FrameDurationLimits":(16666,16666)})
config_r = rightcam.create_video_configuration(main={"size":(1920,1080)},transform=Transform(hflip=True,vflip=True),controls = {"FrameDurationLimits":(16666,16666)})

leftcam.configure(config_l)
rightcam.configure(config_r)

def mergeviews(request):
    
    with MappedArray(request, "main") as leftview:
        tmp=leftview.array
        #fill left half of original array with half-width sampled left camera view
        leftview.array[:,960:] = tmp[:,::2]
        #fill right half of original array with half width sampled right camera view
        leftview.array[:,:960] = rightview[:,::2]
        

leftcam.start()
rightcam.start()

rightview=rightcam.capture_array()
leftcam.pre_callback = mergeviews

try:
    while True:
        rightview=rightcam.capture_array()
except KeyboardInterrupt:
    pass
```

davidplowman, 2023-12-07:
One more thought.

You could probably ask the right cam to give you 960x1080 output but then set the 'ScalerCrop' so that you get the full field of view squished down into half the width. That would also reduce the number of pixels you need to copy around.

It would be great if we could do that with the left cam too, but we'd have to be able to force the buffers to be allocated with double the width so that you could paste in the right cam image. Actually the code is not too far from being able to do that, and in fact it might even be possible now with some hackery.

Do close the ticket if you want to - you can leave it open if there's more to discuss, or open a new one!

sandyol55, 2023-12-07:
Like the idea to get a pre-compressed half width image, which I expect will be have a slightly higher subjective quality than simply throwing away every second pixel?  Will have a tinker with that soon.

Also intrigued by the potential for a double width buffer ready for a frame to be dropped into..........

I'll close this now as you've guided me towards a working solution, but might re-open a new one if anything else crops up during the ongoing tinkering.
Thanks again for the support.

closed 2023-12-07 by sandyol55

sandyol55, 2023-12-07:
_You could probably ask the right cam to give you 960x1080 output but then set the 'ScalerCrop' so that you get the full field of view squished down into half the width. That would also reduce the number of pixels you need to copy around._

Just had to jump back in and say 'Great suggestion'.

Using the inherent ability to rescale works a treat and as a bonus using the ScalerCrop X/Y offsets allows the image to be panned around, making the task of aligning the two images much easier.   Just had to get the mechanical alignment roughly right then 'walk-in'  the right camera image with X/Y offset changes.

sandyol55, 2024-01-05:
Re-opening just to update on progress and check whether any other suggestions might arise.
I've tried implementing some of the earlier suggestions eg moving the toggle into the callback, getting the 'rightview' as a pre_call back, also reducing the array sizes , increasing buffer_count and lowering nice value when called (as per a suggestion in the forum for another topic - perhaps related?)    
In summary none of the changes seem to have much effect. There is still a flip (Frame drop?) about once every 10 seconds.
Would putting each camera process in it's own thread be likely to help, or is that already the default method of operation?
Example of my stab at a version with the pre_call back getting the 'rightview' below.
```
#!/usr/bin/python3

from picamera2 import Picamera2, Preview
from picamera2 import MappedArray
from libcamera import Transform
import time

leftcam = Picamera2(0)
rightcam = Picamera2(1)

leftcam.start_preview(Preview.DRM, x=0, y=0, width=1920, height=1080)

config_l = leftcam.create_video_configuration(raw={"size":(640,480)}, main={"size":(640,480)},transform=Transform(hflip=True,vflip=True),buffer_count=8,controls = {"FrameDurationLimits":(16666,16666)})
config_r = rightcam.create_video_configuration(raw={"size":(640,480)}, main={"size":(640,480)},transform=Transform(hflip=True,vflip=True),buffer_count=8,controls = {"FrameDurationLimits":(16666,16666)})

leftcam.configure(config_l)
rightcam.configure(config_r)

toggle =True

def sequentialviews(request):
    global toggle
    toggle = not toggle
    with MappedArray(request, "main") as leftview:
        if toggle :
            leftview.array[:] = rightviewarray[:]

def rightview(request):
    global rightviewarray
    with MappedArray(request,"main") as right:
        rightviewarray=right.array

rightcam.start()
rightcam.pre_callback = rightview

leftcam.start()
leftcam.pre_callback = sequentialviews

time.sleep(30)
```