raspberrypi/picamera2 #887: Cannot allocate memory after several times create camera and close camera

LeifSec, 2023-12-06:
**Describe the bug**
If creating a camera object, closing it / destroying it (when going out of scope) there is a `Cannot allocate memory` error after some iterations.

**To Reproduce**
```
from picamera2 import Picamera2
from libcamera import ColorSpace, Transform

camera = Picamera2()
configuration = {
    'use_case': 'preview',
    'transform': Transform(),
    'colour_space': ColorSpace.Sycc(),
    'buffer_count': 4,
    'queue': True,
    'main': {'format': 'XBGR8888', 'size': (640, 480)},
    'lores': None, 'raw': {'format': 'SRGGB12_CSI2P', 'size': (640, 480)},
    'controls': {'NoiseReductionMode': 3,
                 'FrameDurationLimits': (100, 83333),
                 'Saturation': 0,
                 'Sharpness': 30,
                 'Brightness': 45, 'AnalogueGain': 5,
                 'ColourGains': (0, 0)},
    'sensor': {},
    'display': 'main',
    'encode': 'main'}

for i in range(0, 20):
    camera.configure(configuration)

camera.close()

for i in range(0, 20):
    print("\n\n -------------------------------", i)
    camera = Picamera2()
    camera.configure(configuration)   # error here after 7 iterations
    camera.close()
```

**Console Output, Screenshots**
```
 ------------------------------- 7
[6:41:18.507074914] [32605]  INFO Camera camera.cpp:1183 configuring streams: (0) 640x480-XBGR8888 (1) 2028x1520-SBGGR12_CSI2P
[6:41:18.508306508] [32611]  INFO RPI vc4.cpp:608 Sensor: /base/soc/i2c0mux/i2c@1/imx477@1a - Selected sensor format: 2028x1520-SBGGR12_1X12 - Selected unicam format: 2028x1520-pBCC
Traceback (most recent call last):
  File "/home/pi/./memory.py", line 37, in <module>
    camera.configure(configuration)
  File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 1104, in configure
    self.configure_(camera_config)
  File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 1087, in configure_
    self.allocator.allocate(libcamera_config)
  File "/usr/lib/python3/dist-packages/picamera2/allocators/dmaallocator.py", line 43, in allocate
    fd = self.dmaHeap.alloc(f"picamera2-{i}", stream_config.frame_size)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/picamera2/dma_heap.py", line 98, in alloc
    ret = fcntl.ioctl(self.__dmaHeapHandle.get(), DMA_HEAP_IOCTL_ALLOC, alloc)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
OSError: [Errno 12] Cannot allocate memory
```

**Hardware :**
* imx477
* Raspberry Pi 2 Model B Rev 1.1

**Additional context**
Raspberian 12
```
rpicam-apps build: f74361ee6a56 23-11-2023 (17:01:12)
libcamera build: v0.1.0+118-563cd78e
```

davidplowman, 2023-12-06:
Can you check what version of Picamera2 you have (`apt list python3-picamera2`)? Thanks.

LeifSec, 2023-12-06:
`python3-picamera2/stable,stable,now 0.3.16-1 all [installed`

davidplowman, 2023-12-06:
@will-v-pi Hi William, could I maybe ask you to have a look at this one? I think we may be leaking stuff in the DmaAllocator when the camera is configured and closed without ever being started. Possibly the DMA heap handle too, not sure. The simplest test case appears to be this
```
from picamera2 import Picamera2

for _ in range(200):
    picam2 = Picamera2()
    picam2.configure()
    picam2.close()
```
Thanks!

will-v-pi, 2023-12-07:
Yep, I've found the leak and a fix - I'll make a pull request to next

davidplowman, 2023-12-07:
> Yep, I've found the leak and a fix - I'll make a pull request to next

Brilliant, thanks!

tim-oe, 2024-12-09:
getting what looks like the same error and reboot is the only thing that seems to clear it

lib: python3-picamera2/stable,stable,now 0.3.23-1 all [installed]
OS: Debian GNU/Linux 12 (bookworm)

stack:

2024-12-08 17:53:02,670|CameraSvc.py:50|ERROR- failed to take picture
Traceback (most recent call last):
  File "/home/tcronin/src/WeatherWatch/weatherwatch/camera/Camera.py", line 80, in process
    picam2.switch_mode_and_capture_file(capture_config, imgFile, wait=True)
  File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 1467, in switch_mode_and_capture_file
    return self.dispatch_functions(functions, wait, signal_function, immediate=True)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 1363, in dispatch_functions
    return job.get_result(timeout=timeout) if wait else job
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/picamera2/job.py", line 79, in get_result
    return self._future.result(timeout=timeout)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/concurrent/futures/_base.py", line 456, in result
    return self.__get_result()
           ^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/concurrent/futures/_base.py", line 401, in __get_result
    raise self._exception
  File "/usr/lib/python3/dist-packages/picamera2/job.py", line 48, in execute
    done, result = self._functions[0]()
                   ^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 1430, in switch_mode_
    self.configure_(camera_config)
  File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 1110, in configure_
    self.allocator.allocate(libcamera_config, camera_config.get("use_case"))
  File "/usr/lib/python3/dist-packages/picamera2/allocators/persistent_allocator.py", line 29, in allocate
    super().allocate(libcamera_config, use_case)
  File "/usr/lib/python3/dist-packages/picamera2/allocators/dmaallocator.py", line 43, in allocate
    fd = self.dmaHeap.alloc(f"picamera2-{i}", stream_config.frame_size)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/picamera2/dma_heap.py", line 98, in alloc
    ret = fcntl.ioctl(self.__dmaHeapHandle.get(), DMA_HEAP_IOCTL_ALLOC, alloc)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
OSError: [Errno 12] Cannot allocate memory

davidplowman, 2024-12-09:
@tim-oe Thanks for the report. However, it is almost certainly not the same problem because that one was fixed. Please do file a new report. Be sure to:

* Confirm that you are running fully up-to-date software. If necessary, download a fresh install, run `sudo apt update` and `sudo apt full-upgrade` and confirm that you can still reproduce the problem.
* Please report the output of `apt list python3-picamera2`, `rpicam-hello --version` and `uname -a`.
* Please state what kind of a Pi you are using.
* Please state what kind of camera you are using.
* Please state whether you have made any alterations in your `/boot/firmware/config.txt` file.
* Please include a _short_ script (no more than 20 lines if possible) that we can use to reproduce the problem.

Thanks!

PR #888 (2023-12-11): Fix DmaAllocator memory leak when closing camera
When closing the camera, the allocator was not closing the open file descriptors, causing a memory leak. Fixed by closing the file descriptors and deleting the allocator when closing the camera.

Fixes #887