raspberrypi/picamera2 #896: [SUGGESTION] Add a WiKi

Boaztheostrich, 2023-12-15:
I am working on a project involving the hq camera module and this project, and am having some difficulty finding information on settings. 

Is there a wiki other than the one on GitHub I could look at?

I'm specifically looking to adjust basic settings like iso, white balance, and shutter speed. 

Ultimately I want to build a gui similar to a canon or Sony mirrorless camera so any information regarding gui would be great as well. 

Right now I am having to look through old issue threads to get answers to questions, having answers in a wiki format would save me a lot of time. And I would hope reduce the amount of repeat issues.

davidplowman, 2023-12-18:
Have you discovered the [manual](https://datasheets.raspberrypi.com/camera/picamera2-manual.pdf)? There are also many example in the [examples](https://github.com/raspberrypi/picamera2/tree/main/examples) folder here.

davidplowman, 2023-12-18:
I guess documentation is always written by folks who are fairly expert in various ways, so it can be hard to appreciate what level of knowledge one can assume of the reader. Perhaps there needs to be a "Basic Examples" chapter in there, or something, that teaches some of the basic camera techniques with Picamera2.

But to answer your immediate question, for a 15ms exposure time you would use
```
picam2.set_controls({'ExposureTime': 15000})
```

The Raspberry Pi [camera forum](https://forums.raspberrypi.com/viewforum.php?f=43) is also quite a good place to ask, there are lots of folks around with a good level of camera knowledge.

davidplowman, 2023-12-18:
Touch controls isn't something we've done ourselves. There are a number of Qt applications which give you an application window with controls on it (e.g. https://github.com/raspberrypi/picamera2/tree/main/apps), and I guess you could use something like that. If you wanted to make your controls "float" over the camera image then you'd have to use an overlay. Again, there are some examples there.

There may be folks on the forum who have done stuff like this and would share - I don't really know I'm afraid!

Boaztheostrich, 2023-12-18:
Thanks for the help! I’m thinking more in regards to an overlay, I’ll make
sure to check out some examples!