raspberrypi/picamera2 #897: [HOW-TO] Correct green color in video recording

jobjansweijer, 2023-12-18:
I'm trying to take a video with a Raspberry Pi 5 and a Pi Camera v2 using the code below. I have two issues:

1) **The 5th sensor mode leads to a green image.** I've tested with all sensor modes. They all look normal, except for sensor mode number 5. How can I fix the color?
2) **h264 file has no length:** When I save a h264 file and open it in VLC, I see a flash of the shot, but then the video is finished. So it is not the full 5 seconds. When I look at the metadata in Windows explorer I can also see that the h264 file has no length. 

Observations:
* Sensor mode 6 starts out blue shortly, but corrects to normal colors quickly, which has to do with AWB, this behavior disappears when I disable AWB. So my guess is the green color for sensor mode 5 does not have to do with AWB.
* FfmpegOutput and FileOutput yield the same results.

Thanks for helping me out in advance!

### Code
```
import time
from picamera2 import Picamera2
from picamera2.encoders import H264Encoder
from picamera2.outputs import FfmpegOutput, FileOutput

#set variables
hour = int(time.strftime('%H'))
now = time.strftime("%y_%m_%d_%H_%M")

basename = "test"+now
h264_output = basename+".h264"
mp4_output = basename+".mp4"

#set video configuration
picam2 = Picamera2()
mode = picam2.sensor_modes[5]
config = picam2.create_preview_configuration(sensor={'output_size': mode['size'], 'bit_depth': mode['bit_depth']})
picam2.configure(config)
#picam2.set_controls({"AwbEnable": False})
encoder = H264Encoder(bitrate=17000000)

output = FfmpegOutput(mp4_output)
# output = FfmpegOutput(h264_output)
# output = FileOutput(h264_output)

#set length of footage
picam2.start_recording(encoder, output)
time.sleep(5)
picam2.stop_recording()

```

### Sensor mode 5
```
 {'bit_depth': 8,
  'crop_limits': (0, 0, 3280, 2464),
  'exposure_limits': (37, 5883414, None),
  'format': SRGGB8,
  'fps': 83.7,
  'size': (1640, 1232),
  'unpacked': 'SRGGB8'},
```

### All sensor modes
```
[{'bit_depth': 10,
  'crop_limits': (1000, 752, 1280, 960),
  'exposure_limits': (37, None),
  'format': SRGGB10_CSI2P,
  'fps': 206.65,
  'size': (640, 480),
  'unpacked': 'SRGGB10'},

 {'bit_depth': 10,
  'crop_limits': (0, 0, 3280, 2464),
  'exposure_limits': (75, 5883419, None),
  'format': SRGGB10_CSI2P,
  'fps': 41.85,
  'size': (1640, 1232),
  'unpacked': 'SRGGB10'},

 {'bit_depth': 10,
  'crop_limits': (680, 692, 1920, 1080),
  'exposure_limits': (75, 11766829, None),
  'format': SRGGB10_CSI2P,
  'fps': 47.57,
  'size': (1920, 1080),
  'unpacked': 'SRGGB10'},

 {'bit_depth': 10,
  'crop_limits': (0, 0, 3280, 2464),
  'exposure_limits': (75, 11766829, None),
  'format': SRGGB10_CSI2P,
  'fps': 21.19,
  'size': (3280, 2464),
  'unpacked': 'SRGGB10'},

 {'bit_depth': 8,
  'crop_limits': (1000, 752, 1280, 960),
  'exposure_limits': (37, 5883414, None),
  'format': SRGGB8,
  'fps': 206.65,
  'size': (640, 480),
  'unpacked': 'SRGGB8'},
 
 {'bit_depth': 8,
  'crop_limits': (0, 0, 3280, 2464),
  'exposure_limits': (37, 5883414, None),
  'format': SRGGB8,
  'fps': 83.7,
  'size': (1640, 1232),
  'unpacked': 'SRGGB8'},

 {'bit_depth': 8,
  'crop_limits': (680, 692, 1920, 1080),
  'exposure_limits': (75, 5883419, None),
  'format': SRGGB8,
  'fps': 47.57,
  'size': (1920, 1080),
  'unpacked': 'SRGGB8'},
  
 {'bit_depth': 8,
  'crop_limits': (0, 0, 3280, 2464),
  'exposure_limits': (75, 11766829, None),
  'format': SRGGB8,
  'fps': 21.19,
  'size': (3280, 2464),
  'unpacked': 'SRGGB8'}]
```

### Normal color and green color
Green color
[image]

Normal color
[image]

jobjansweijer, 2023-12-18:
Additional information:

When I split the channnels for the different sensor modes, it's as if only the green channel contains information. See examples below:

### SensorMode 1
#### Channel 0

#### Channel 0

#### Channel 0

davidplowman, 2023-12-18:
Hi, thanks for the report.

So I think the "green" issue is actually the same as this one https://github.com/raspberrypi/libcamera/issues/87. We're a bit baffled why we don't see this on a Pi 4. And with a v1 camera it seems to occur randomly when you start the camera, yet on the v2 camera it seems to happen every time in mode 5 and none of the others. So we're a bit confused by it. Nonetheless there's a fix which will be in the next release.

In the meantime, you could build the fix for yourself, or you could even just verify that it's the same problem by disabling the lens shading block (find the imx219.json file on your system and change the string "rpi.alsc" ro "x.rpi.alsc"). But this will give rise to some colour shading problems, so probably not something you want to do permanently.

Also, for most (all?) purposes, the 1640x1232 10-bit mode is better so I would recommend trying to use that. If the higher framerate is important, you can probably get the mode to work by starting the camera in a different mode, stopping it immediately, and then starting again in mode 5. So this would incur a short delay, at least until the fix is released.

As regards the other problem, do you mean that the .mp4 file is OK but the .h264 file is not? If so, the cause may be the "problem" with .h264 files in that this is not a proper container format. It's a pure h.264 bitstream so there are no timestamps in it. We have found that recent versions of vlc just won't play files without timestamps properly. ffplay works better for such files - it just plays them at 30fps.

jobjansweijer, 2023-12-18:
Thanks for your swift and comprehensive reaction.

Changing the imx219.json file solved the issue for me.

I'm indeed choosing for mode 5 because of the frame rate.

Also thanks for your clarification on h264.

closed 2023-12-18 by jobjansweijer