raspberrypi/picamera2 #898: [SUGGESTION] Make CompletedRequest support context manager protocol.

rosesyrett, 2023-12-19:
I've just been reading the picamera2 manual, and really enjoying it so far; thanks for all the effort put into this! I've recently started writing abstraction layers for my camera, and thought it'd be really neat if the `capture_request` method of a `Picamera2` object could be written as a context manager. That is, at the moment to do anything with a request, one has to:

```python
#set up camera.
picam2 = Picamera2()
picam2.configure(...)
picam2.start()

#get a request
request = picam2.capture_request(**kwargs)
# do stuff with request... e.g. save or get an array from "main" stream.
request.release()
```

When I first came across this it was a bit cumbersome; to me it seems natural that `picam2.capture_request` should be treated in a `with` block, so that I don't need to manually type out `request.release()` and can immediately assign a value to its output. I've found that if I just type;

```python
picam2.capture_request(**kwargs)
```

then because I've never saved a reference to that request, I can never capture another request; if I try it hangs. Also, it is perhaps tedious to have to remember that you need to `.release` the request, instead of `.close` or something similar. So a with block could improve these minor usability issues quite reasonably:

```python
with camera.capture_request(**kwargs) as request:
    #do stuff with request... e.g. save or get an array from "main" stream.
```

I've tried looking at the code for the capture_request() method to suggest the easiest way to implement this, and at the moment it's a little bit obfuscating to me. But if it's not too complex to add, I think this would be a nice feature.

Thanks!

davidplowman, 2023-12-19:
Yes, that sounds like a good idea. The slightly weird thing is that requests are handed out from the camera thread with a "use_count" of 1, so it feels like the "enter" method would do nothing, and the "exit" would do a release. That would kinda work, but it doesn't feel quite right...

davidplowman, 2023-12-20:
Maybe there wants to be a CapturedRequest class with a context manager so that you could write:
```
with CapturedRequest(picam2) as request:
    # do stuff
```
where the CaptureRequest enter would actually do `picam2.capture_request()`, and exit would do the release. Does that seem like a reasonable thing to do?