raspberrypi/picamera2 #900: [HOW-TO] Recreate capture_continuous function with PiCamera2

Ara-O, 2023-12-20:
I am trying to migrate from picamera to picamera2 and I have been struggling to translate this code to picamera2

```
with picamera.PiCamera(resolution=(640,480), framerate=30) as camera:
     camera.start_preview(fullscreen=False, window=(100, 20, 640, 480))
     try:
        stream = io.BytesIO()

        # I'm having problems specifically with the line below
        for _ in camera.capture_continuous(stream, format='jpeg', use_video_port=True):
            stream.seek(0)
            image = Image.open(stream).convert('RGB')
``` using picamera

I want to update that to use picamera2's functions.

puterboy, 2024-01-25:
I have the same question - did you figure this out?
I'm surprised that such a basic feature is not replicated in picamera2 and barring that at least a workaround should be documented.
How else are people supposed to update picamera code?

i1Cps, 2024-03-27:
Any luck guys?

LedRami, 2024-04-11:
Up