raspberrypi/picamera2 #902: [BUG] Recording video with audio=True results in "Non-monotonous DTS in output stream"

guighub, 2023-12-21:
**Describe the bug**
Trying to record video with audio from a USB microphone results in a lot of "Non-monotonous DTS in output stream" errors from ffmpeg and preventing my script from running, I also tried using `FfmpegOutput` but got the same result.

**To Reproduce**
```python
config = cam.create_video_configuration(main={"size": video_resolution}, lores={"size": (640, 480)}, display="lores") # Video resolution is 1440x1080 in this case
cam.configure(config)
cam.set_controls({"AfMode": controls.AfModeEnum.Manual, "LensPosition": focus})
cam.set_controls({"FrameRate": video_fps}) # video_fps was set to 24, but I also tried 30

time_now = datetime.today().strftime('%Y-%m-%d_%H-%M-%S')

file_path = f"/home/guiguig/Picamera2_Photos/RPI_{time_now}.mp4"

cam.start_and_record_video(file_path, duration=video_length, audio=True) # video_length was 6, but recorded for much longer than six seconds
```

**Expected behaviour**
Audio to be recorded to the video with no errors or freezing up.

**Console Output, Screenshots**
```
[h264 @ 0x559c44f440] Thread message queue blocking; consider raising the thread_queue_size option (current value: 32)
[aac @ 0x559c46f670] Queue input is backward in time
[mp4 @ 0x559c48b610] Non-monotonous DTS in output stream 0:1; previous: 204682, current: 203062; changing to 204683. This may result in incorrect timestamps in the output file.
[mp4 @ 0x559c48b610] Non-monotonous DTS in output stream 0:1; previous: 204683, current: 204086; changing to 204684. This may result in incorrect timestamps in the output file.
[aac @ 0x559c46f670] Queue input is backward in time
[mp4 @ 0x559c48b610] Non-monotonous DTS in output stream 0:1; previous: 218422, current: 184244; changing to 218423. This may result in incorrect timestamps in the output file.
[mp4 @ 0x559c48b610] Non-monotonous DTS in output stream 0:1; previous: 218423, current: 185268; changing to 218424. This may result in incorrect timestamps in the output file.
[mp4 @ 0x559c48b610] Non-monotonous DTS in output stream 0:1; previous: 218424, current: 186292; changing to 218425. This may result in incorrect timestamps in the output file.
[mp4 @ 0x559c48b610] Non-monotonous DTS in output stream 0:1; previous: 218425, current: 187316; changing to 218426. This may result in incorrect timestamps in the output file.
```

**Hardware :**
Raspberry Pi Zero 2 W, Camera Module 3, Mini USB Microphone from Digikey/Adafruit

**Additional context**
The exact microphone I used: https://www.adafruit.com/product/3367

goldenrubb3rs, 2023-12-28:
experiencing similar issue, also getting:

"[out_0_1 @ 0x9fdb80] 100 buffers queued in out_0_1, something may be wrong."

using a Zero 2 W with recommended Legacy 32-bit, and IQaudIO Codec Zero.

p.s. sorry I'm new to using github and all of this, just been tinkering around with this stuff and trying to teach myself so I don't really know the proper way of capturing and uploading console output. but I've just been working off the examples in the documentation for picamera2 and it seems ffmpeg is completely broken in picamera2. works fine using libcamera-vid from the terminal though.

davidplowman, 2023-12-28:
Hi, thanks for the reports. There certainly was a problem at one point with the FfmpegOutput not stopping when there was an audio stream. Can you check what version of Picamera2 you have? Use `apt list python3-picamera2` - I expect it will be 0.3.12?

If that's the case you could try checking out the 0_3_12 branch from this repository. You can either clone it directly and point your PYTHONPATH at it (this might be easier as it would let us experiment with some of the Ffmpeg parameters later), or you could get pip to install that branch directly from git. At least things shouldn't hang after that, so lets see what errors we still have then.

guighub, 2024-01-03:
Hi, sorry for the late reply.

I don't think I'm able to reinstall the microphone onto my Pi to test the code, as it's already mounted to a wooden post outdoors since I eventually decided to ditch audio recording as it wasn't entirely necessary for my project. I may purchase another camera module for my Pi 5 though to test if I have the same problem.

davidplowman, 2024-01-03:
No problem. Well, I'll probably make that small update to the queue sizes, and also push that onto the 0_3_12 branch for Bullseye users.

sarvasvkulpati, 2024-02-01:
Hi, just wanted to follow up on this issue. 

I'm also running a Raspi Zero 2W, Bullseye, using a raspi cam v3 and a sph0645 I2s mic. 

When I run audio_video_capture.py, but also the code at https://github.com/raspberrypi/picamera2/issues/902#issuecomment-1874095110, it returns the same issues: 
```
Guessed Channel Layout for Input Stream #0.0 : stereo
[mp4 @ 0x122c370] Timestamps are unset in a packet for stream 0. This is deprecated and will stop working in the future. Fix your code to set the timestamps properly
[out_0_1 @ 0x1255520] 100 buffers queued in out_0_1, something may be wrong.
[pulse @ 0x1221e70] Thread message queue blocking; consider raising the thread_queue_size option (current value: 2048)

```
and remains frozen. I also tried recording a video and simultaneously recording with pyaudio, and then combining the two with ffmpeg, but that always returns no audio output. The camera and mic work independently. 

Do you have any suggestions on how to fix this or what I could do?

davidplowman, 2024-02-02:
Hi, could you try the 0_3_12 branch from this repository, there was a known issue with audio not stopping. Thanks.

luqtas, 2025-08-10:
rpi zero 2w, pi global shutter, INMP441 mic. (works with arecord), bullseye, picamera2 0.3.12-2

```
[pulse @ 0x16d63b0] Thread message queue blocking; consider raisin the thread_queue_size option (current value: 512)
[out_0_1 @ 0x16dd270] Thread message queue blocking' consider raising the thread_queue_size option (current value: 32)
default: Generic error in an external library
```

i tried the first solution of https://github.com/raspberrypi/picamera2/issues/467 with no avail. then i tried to modify the .py to 'alsa' and it worked. but. audio records only on one channel (it doesn't detect we are mono) and after some seconds it stops. video record is also laggy (which isn't if we are just recording video)

davidplowman, 2025-08-10:
Have you tried the `PyavOutput` instead? That's a more recent (and better) way to including audio. Note that you have to set `audio` to `True` in the encoder, like [this](https://github.com/raspberrypi/picamera2/blob/main/examples/pyav_stream2.py#L15). Also check section 7.2.4 of the [manual](https://datasheets.raspberrypi.com/camera/picamera2-manual.pdf).

nzottmann, 2025-08-10:
I do not use audio, but I get this error message also sporadically using the `PyavOutput`:

```
2025-08-08 19:58:54,266 DEBUG    - Discarded non monotonically increasing dts: 19484976901 <= 19485018555
2025-08-09 11:17:37,433 DEBUG    - Discarded non monotonically increasing dts: 74608151300 <= 74608192965
2025-08-09 15:38:40,979 DEBUG    - Discarded non monotonically increasing dts: 90271705804 <= 90271747441
2025-08-09 16:47:59,670 DEBUG    - Discarded non monotonically increasing dts: 94430383346 <= 94430425001
2025-08-09 17:15:41,891 DEBUG    - Discarded non monotonically increasing dts: 96092613450 <= 96092655105
```

How often this happens varies from device to device, all of them are CM4.

The only solution I found was a workaround. I created a custom encoder based on `PyavOutput`:
```python
class ContinousPyavOutput(PyavOutput):
    def __init__(self, *args, **kwargs):
        self.last_timestamp = 0
        super().__init__(*args, **kwargs)

    def outputframe(self, frame, keyframe=True, timestamp=None, packet=None, audio=False):
        if timestamp <= self.last_timestamp:
            logger.debug(f"Discarded non monotonically increasing dts: {timestamp} <= {self.last_timestamp}")
            return
        
        super().outputframe(frame, keyframe, timestamp, packet, audio)
        self.last_timestamp = timestamp
```

Seems like there is a problem which is triggered more or less often depending on different factors like audio and something else.

Unfortunately I cannot help here with a minimal example as this happens in a bigger application and I did not have time for testing a minimal example (which would probably take a long time to wait), as I currently have this workaround. As I don't loose many frames this causes no problem currently.  

I just wanted to report this to give a hint that the actual problem is possibly not only audio related.

luqtas, 2025-08-10:
> Have you tried the `PyavOutput` instead? That's a more recent (and better) way to including audio. Note that you have to set `audio` to `True` in the encoder, like [...]

i installed Pyav with pip install av, because i though the error;

`ImportError: cannot import name 'PyavOutpu' from 'picamera2.outputs' (/usr/lib/python3/dist-package/picamera/outputs.__init__.py)`

but i still have the same error after installing (even tried to reboot)

davidplowman, 2025-08-11:
Do not install using pip. Pip packages are generally much newer that the Debian/Bookworm system ones, and are often incompatible.

Uninstall the pip version and use `python3-av` from apt. If this does not work:

* Please confirm what OS you are using and that it's update.
* Please confirm that Picamera2 is up-to-date. I think 0.3.30 is the latest.
* Please report the actual error that stops it from being imported.
* Check that you can `import av` on its own, without Picamera2. Check the value of `av.__file__` to be sure you're importing what you think you are importing. You should have version 12.3.0,

luqtas, 2025-08-11:
@tvoverbeek i'm using this one: http://www.lcdwiki.com/2.4inch_RPi_Display_For_RPi_3A+#How_to_use_with_Raspberry_Pi_OS but i also tried this PR from this lib: https://github.com/juj/fbcp-ili9341/pull/261 and both didn't worked on Bookworm

i don't mind losing the touch capabilities of the display tho