raspberrypi/picamera2 #908: [OTHER] PiZero2W IMX219 AutoExposure never settles

Hartleby, 2024-01-01:
On raspberryOs 12 (bookworm) lite, using the IMX219 camera, auto exposure never settles down.

This code:
```
from picamera2 import Picamera2
import time

tuning = Picamera2.load_tuning_file("imx219.json")
picam2 = Picamera2(tuning=tuning)

config = picam2.create_still_configuration()
picam2.configure(config)
# should be already enabled
picam2.set_controls({"AeEnable":True}) 

picam2.start()
locked = False
while locked != True:
       locked = picam2.capture_metadata()["AeLocked"]
       time.sleep(.1)
       print("waiting for lock")
```
hangs in the loop forever.
Is autoexposure simply broken or am I missing something?

davidplowman, 2024-01-02:
Hi, yes there was a bug to this effect. It was fixed in our libcamera repo here https://github.com/raspberrypi/libcamera/commit/2905eabad9ffa3a83c498d5a4a98a110fc0317af. Unfortunately I see that's only in our "next" branch currently, not in "main" (which corresponds to the latest release). I'm not quite sure, but I expect we'll be making another update quite soon.

dgalland, 2024-01-02:
Hi,
I also have discussion #795 on the AGC/AGE algorithm
Yes we should be able to rely on AeLocked to know when it converged.
Other problem it seems that convergence is currently quite slow.
So let's wait for these improvements !