raspberrypi/picamera2 #910: [HOW-TO] Add anullsrc audio

MazrimT, 2024-01-02:
**Describe what it is that you want to accomplish**
With ffmpeg you can add a null-source for audio (ie. there's audio but there's no sound) by adding the following to the ffmpeg command:
`-f lavfi -i anullsrc=sample_rate=48000:channel_layout=mono -c:a aac b:a 128k`
I want to be able to add to the output (FfmpegOutput) an audio source with no sound.

**Describe alternatives you've considered**
I've tried adding things to the FfmpegOutput string but get errors "Unknown decoder 'copy'" and not sure what that means.
I've also tried adding different things to FfmpegOutput audio_device with no luck. I just get the error message like:  
`anullsrc: No such process`

I've also tried adding a mp3 file as audio but that did not work either.

**Additional context**
This might seem like a strange request, but the reason is that when live streaming to youtube it requires an audio source in the stream and I'm trying to build a pi that can stream directly from it's camera to youtube but only video.

davidplowman, 2024-01-02:
I thought you could just set `audio=True` and run it with no microphone. Is that giving you the same `No such process` error? Can you say what kind of Pi you are using, and confirm that your OS is up to date? Thanks.

MazrimT, 2024-01-02:
just using `audo=True` gives the following error: `default: No such process`.

Raspberry Pi 4 Model B Rev 1.2
uname -a: ```Linux raspberrypi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux```
os version 11.8

I also just spent a few hours trying to set up a "snd-dummy" "fake soundcard" but can't get it to work either.

davidplowman, 2024-01-02:
I'm seeing the same errors on some Pis here, but not on others, so I'm a bit confused. I'm guessing they may have different software versions. In fact, on the ones that don't work, it doesn't even work when a microphone is plugged in. So I think the problem isn't that you need "anullsrc", the audio system is just broken.

It's easy to test this in isolation with commands like
```
ffmpeg -f pulse -i default test.mp4
```

I don't know too much about the way audio works, but the date on your kernel looks rather old - are you running Bullseye? There is only critical support for that now, are you able to try Bookworm instead? I've verified that the above command fails with `No such process` on my "broken" Pis, but works happily on an out-of-the-box clean install of Bookworm (with all updates applied) on a Pi 4.

davidplowman, 2024-01-02:
OK, I discovered rather by accident that I was getting this `No such process` error when I was logged into my Pi with X forwarding (`ssh -X ...`). When I didn't enable X forwarding, in fact it all works fine. Could that be the same thing you're seeing? I wonder if there are some issues with exactly where the audio is supposed to be coming from, and whether you are permitted to access it. So it sounds like there may be something that needs investigation there - possibly one of the more general forums would reach some folks who know more about this?

MazrimT, 2024-01-05:
updating everything (and then reinstalling because I messed stuff up) yes it does work now.

closed 2024-01-05 by MazrimT